john30/ebusd #5: calormatic 470 ebus interface

jekkos, 2015-10-24:
Hello. nice project you have here.

I'm bulding a hardware interface for ebus (Vaillant ecoTec plus + calormatic 470, which communicate through ebus interface). I found a rj-11 like jack on the latter on the right side, [which looks like the cable shown in this picture.](http://www.haustechnik-express.de/imagemagic.php?img=0NHHzZWsYcKvo2hll5SWmWlxZ5rY458%3D&w=630&h=501&page=popup) Any idea if this could be an eBus connection and what the pinout could be?

thanks

john30, 2015-10-25:
Usually you have to find the eBUS connector on the heating system, which should be done by your electrician or heating engineer. On your heating system, the connector should be labelled "BUS" with a "+" and a "-" clamp. The RJ11 might work as well, but you need to check at least the voltage level. afaik the two PINs in the middle are the right ones.

jekkos, 2015-10-25:
Yes that's why I first want to see whether there's a signal on the rj jack first. I'll check the voltage levels think one will be between 9-12 and the other one 12-20V. Just doing the reading part for now to see if I can read out some values. Thanks.

closed 2015-11-08 by john30

bayc4d, 2022-12-11:
any succes this ebus adapter?

vr39
vr32
ebt2