john30/ebusd #11: FreeBSD

beam, 2016-01-19:
Hi, can you add FreeBSD support?

john30, 2016-01-20:
ebusd should already be compilable under FreeBSD.
AFAIK you need to use the "--with-argp-lib" and "--with-argp-include" options of configure pointing to argp-standalone directories.

beam, 2016-01-20:
Hi, i installed https://www.freshports.org/devel/argp-standalone/ and no extra options needed for argp.
But i have to make some patches for code compilation:
```
diff --git a/src/ebusd/network.cpp b/src/ebusd/network.cpp
index 7d586c6..13e9505 100644
--- a/src/ebusd/network.cpp
+++ b/src/ebusd/network.cpp
@@ -28,6 +28,10 @@
 #include <poll.h>
 #endif

+#ifndef POLLRDHUP
+#define POLLRDHUP 0x2000
+#endif
+
 using namespace std;

 int Connection::m_ids = 0;
diff --git a/src/lib/ebus/device.h b/src/lib/ebus/device.h
index 3a8892e..71b0047 100644
--- a/src/lib/ebus/device.h
+++ b/src/lib/ebus/device.h
@@ -23,6 +23,8 @@
 #include <iostream>
 #include <fstream>
 #include <arpa/inet.h>
+#include <netinet/in.h>
+#include <sys/socket.h>
 #include <netdb.h>
 #include "result.h"

diff --git a/src/lib/utils/tcpsocket.cpp b/src/lib/utils/tcpsocket.cpp
index 27ed4a8..81ef810 100644
--- a/src/lib/utils/tcpsocket.cpp
+++ b/src/lib/utils/tcpsocket.cpp
@@ -20,6 +20,7 @@
 #include <cstdlib>
 #include <fcntl.h>
 #include <arpa/inet.h>
+#include <netinet/in.h>
 #include <netdb.h>
 #include <string.h>
```

now i can run ebusd, but can't read from /dev/ttyU2, when i start ebusd TX and RX LEDs on adapter are on, but nothing in shown i ebusd. No idea what to do now.. :/

beam, 2016-01-21:
Any idea?

john30, 2016-01-23:
which interface are you using?

beam, 2016-01-23:
I bought this https://www.mikrocontroller.net/topic/346833#4405093 . 
Scheme is at the begging of threat. It's eBUS<->USB (FT232R) adapter.

john30, 2016-01-23:
well, the interface works, I know that.
I can only assume that you have problems with the serial communication to the device.
I'm not familiar with BSD, so I guess you'll have to find out yourself...

beam, 2016-01-23:
Yes, when i boot on same PC linux it works, but under FreeBSD :(

john30, 2016-01-24:
It'll probably be a difference in the handling of terminal devices in BSD compared to Linux.

john30, 2017-06-08:
closed due to inactivity

closed 2017-06-08 by john30