john30/ebusd #12: Synology Diskstation Support

dsebusd, 2016-01-20:
Hi, 

do you also plan to support NAS systems (eg. Synology Diskstation)?
I really want to monitor my Air Heat Pump (Vaillant VWL Geotherm Plus) over Network w/ my Diskstation. All necessary Hardware (vrDIALOG Interface, eBus Koppler Ethernet, Synology DS213+) is available. To run the eBus Deamon on my FritzBox 3390 and/or on a separate Rasperry Pi is not applicable for me becuase these platforms are too inperformant. 

Thanks for your reply!

Rgds, 
M.

john30, 2016-02-09:
not yet, as the number of users with DS is too low at the moment compared to the effort. If you'd provide a virtual box image with a tool chain I'd create releases for DS as well.

chri2, 2017-01-25:
@dsebusd I'm running ebusd on synology diskstation inside a docker container on a diskstaion with x86_64. The 213+ is PowerPC, but it should be possible to get a container with debian for x86 or armhf running. If so, it would be quite easy to start ebusd.
@john30 I didn't look really deep into docker hub, but it offers a option to automatically build images from a repository. If there would be a genuine ebusd image anybody with a system offering docker could quite easily start a container and try it or use it. What do you think?

john30, 2017-01-25:
@chri2 docker support is already on my TODO list, but I didn't have time for that yet.

chri2, 2017-01-25:
If support is needed (e.g. things like how to get /dev/ttyUSBx inside the container or other details) let me know. I'm running ebusd, fhem, brickd in different containers on diskstation.

john30, 2017-01-29:
Dockerfile is created and image available on docker hub: https://hub.docker.com/r/john30/ebusd/

closed 2017-01-29 by john30