john30/ebusd #13: Enhancement: raw-logfile

Pierce0815, 2016-01-26:
Hi,

it would be great if you could create an extra option for a separate logfile for raw data according to dump data.

Could be something like:

Log Options
[...]
    
    --rawdatafile=FILE
    Log received bytes to FILE [/var/log/ebusd.log]

    --rawdatasize=SIZE
    Make rawdata files no larger than SIZE kB [100]

BTW: cool tool :-)

john30, 2016-01-27:
Thanks!
I'll add it to my TODO list :)

john30, 2016-10-25:
Internal note: add verbose logging of received messages including crc status and timing info, e.g.
- "raw bytes" for bytes log
- "raw" for message log
- "raw -v" for verbose message log

john30, 2016-12-04:
solved in first simple version with commit cdbcb46

closed 2016-12-04 by john30