john30/ebusd #18: Lost VR60 connection

modenet, 2016-05-01:
Hi, I used ebusd from first version. in version 2.0 my INFO results was:
08;Vaillant;BAI00;0703;7401
15;Vaillant;UI   ;0501;6201
1c;Vaillant;RC C ;0501;6201
23;Vaillant;SOLSY;0500;6301
25;Vaillant;SOLSY;0500;6301
26;Vaillant;SOLSY;0500;6301
35;Vaillant;RC C ;0501;6201
3c;Vaillant;RC C ;0507;6201
50;Vaillant;SOLSY;0500;6301
52;Vaillant;MC2  ;0500;6301
53;Vaillant;MC2  ;0500;6301

(note last 2 rows)

now after updating both ebusd and ebusd-configuration my INFO command returns:
version: ebusd 2.0.3b6f385
signal: acquired
symbol rate: 142
masters: 7
messages: 729
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0703;HW=7401", loaded "bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=UI   ;SW=0501;HW=6201", loaded "vaillant/15.ui.csv"
address 17: master #17
address 1c: slave #17, scanned "MF=Vaillant;ID=RC C ;SW=0501;HW=6201", loaded "vaillant/1c.rcc.4.csv"
address 23: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/23.solsy.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/25.solsy.hwc.csv"
address 26: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/26.solsy.hc.csv"
address 30: master #3
address 31: master #8, ebusd
address 35: slave #3, scanned "MF=Vaillant;ID=RC C ;SW=0501;HW=6201", loaded "vaillant/35.rcc.1.csv"
address 36: slave #8
address 37: master #18
address 3c: slave #18, scanned "MF=Vaillant;ID=RC C ;SW=0507;HW=6201", loaded "vaillant/3c.rcc.5.csv"
address 50: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/50.solsy.mc.csv"
address 70: master #4
address 75: slave #4, scanned "MF=Vaillant;ID=RC C ;SW=0501;HW=6201", loaded "vaillant/75.rcc.csv"
address ec: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/ec.solsy.sc.csv"

I lost addresses 52 and 53
why?
thanks

john30, 2016-05-05:
try a full scan and then check the info again

modenet, 2016-05-05:
nothing to do with restart or multiple scan

john30, 2016-05-05:
please provide the output of the following commands:
ebusctl scan full && sleep 30 && ebusctl scan result && ebusctl info

modenet, 2016-05-06:
I can't explain: after many reboot my ebusd start starting at boot.

excuse me for loosing your time.

modenet, 2016-05-06:
It was depending from the homemade ebus/serial adapter: with the original eservice I have no problem, with my homemade adapter I can't see VR60 info...

closed 2016-05-06 by modenet

john30, 2016-05-07:
you could issue an individual scan with raw logging enabled, then we could check what is going wrong in the communication:
ebusctl raw && ebusctl scan 52 && ebusctl raw
After that, the log file contains lines with ">XX" and "<XX". These are the raw bytes seen on the bus and with these I can tell you probably more.