john30/ebusd #19: ebusd service don't start on jessie

modenet, 2016-05-05:
hi, I'm using a new raspberry with jessie. I have copied from contrib /etc/init.d/ebusd.
I encounted the problem wih:
-ebusd 2.0 (git, make, make install)
- ebusd 2.1 installed with deb
when I start the ebusd service with  "service ebusd start", ebusd process don't start (I verified with ps -A).
when I launch "ebusd -f" works perfectly, without any other parameters.
no /var/log/ebusd.log is written.

service ebusd status returns:
● ebusd.service - LSB: controls ebusd, the daemon for communication with eBUS heating systems.
   Loaded: loaded (/etc/init.d/ebusd)
   Active: active (exited) since sab 2016-04-30 23:09:37 CEST; 4 days ago

mag 05 14:02:56 vaillant systemd[1]: Started LSB: controls ebusd, the daemon for communication with eBUS heating systems..

with wheezy I have no problem

john30, 2016-05-05:
I don't really understand what your problem is.
ebusd works with wheezy? If so, then what do you want?

modenet, 2016-05-05:
I would you understand why ebusd doesn't start on boot on a raspberry with Jessie.

modenet, 2016-05-06:
after many reboot my ebusd now start at boot. 
excuse me for loosing your time. thanks

closed 2016-05-06 by modenet

john30, 2016-05-07:
ok great.