john30/ebusd #20: ebusctl return me "ERR: element not found"

modenet, 2016-05-11:
Hi, I developed a php page that show some few values read from ebusd, using ebusctl (shell_exec).
if I run the php page on the same ebusd server I have no problem.
If I run the php page on another server and I request values using "ebusctl -s xxx" major times (not always)  I receive the error "ERR: element not found".
the problem seems to be on ebusd and not on ebusctl.
have you any idea?
thanks

john30, 2016-05-14:
I need more data to be able to say anything about that. Please provide debug level logging of ebusd at the time where such a request fails.

modenet, 2016-05-14:
2016-05-14 10:42:36.488 [network info] [00017] client connection opened 10.6.0.1
2016-05-14 10:42:36.555 [network debug] [00017] wait for result
2016-05-14 10:42:36.555 [main debug] >>> read date
2016-05-14 10:42:36.655 [main debug] <<< ERR: command not found
2016-05-14 10:42:36.655 [network info] [00017] connection closed
2016-05-14 10:42:36.813 [network debug] [00018] wait for result
2016-05-14 10:42:36.813 [main debug] started thread 1963979856=connection
2016-05-14 10:42:36.813 [network info] [00018] client connection opened 10.6.0.1
2016-05-14 10:42:36.813 [main debug] >>> read time
2016-05-14 10:42:36.930 [main debug] <<< ERR: command not found
2016-05-14 10:42:36.930 [network info] [00018] connection closed
2016-05-14 10:42:37.064 [main debug] started thread 1919939664=connection
2016-05-14 10:42:37.064 [main debug] performing regular tasks
2016-05-14 10:42:37.064 [main debug] >>> read -c ui OutsideTemp temp
2016-05-14 10:42:37.064 [main info] read ui OutsideTemp cached: 20.25
2016-05-14 10:42:37.064 [main debug] <<< 20.25
2016-05-14 10:42:37.064 [network debug] [00019] wait for result
2016-05-14 10:42:37.064 [network info] [00019] client connection opened 10.6.0.1
2016-05-14 10:42:37.064 [main debug] >>>
2016-05-14 10:42:37.294 [main debug] started thread 1928328272=connection
2016-05-14 10:42:37.294 [network info] [00020] client connection opened 10.6.0.1
2016-05-14 10:42:37.294 [main debug] <<< ERR: element not found
2016-05-14 10:42:37.294 [network debug] [00020] wait for result
2016-05-14 10:42:37.295 [main debug] >>>
2016-05-14 10:42:37.411 [main debug] <<< ERR: command not found
2016-05-14 10:42:37.411 [network info] [00020] connection closed
2016-05-14 10:42:37.543 [main debug] started thread 1953494096=connection
2016-05-14 10:42:37.543 [network info] [00021] client connection opened 10.6.0.1
2016-05-14 10:42:37.543 [main debug] >>> read -c hwc OperatingMode2
2016-05-14 10:42:42.809 [network info] [00022] client connection opened 10.6.0.1
2016-05-14 10:42:42.886 [network debug] [00022] wait for result
2016-05-14 10:42:42.886 [main debug] >>> read date
2016-05-14 10:42:42.887 [bus info] send message: 3115b509030d6100
2016-05-14 10:42:43.936 [network info] [00023] client connection opened 10.6.0.1
2016-05-14 10:42:43.937 [main debug] >>> read time
2016-05-14 10:42:44.301 [network debug] [00024] wait for result
2016-05-14 10:42:44.301 [network info] [00024] client connection opened 10.6.0.1
2016-05-14 10:42:44.301 [main error] read ui OutsideTemp cached: ERR: element not found
2016-05-14 10:42:44.302 [main debug] <<< ERR: element not found
2016-05-14 10:42:44.302 [main debug] >>>
2016-05-14 10:42:44.418 [main debug] <<< ERR: command not found
2016-05-14 10:42:44.418 [network info] [00024] connection closed
2016-05-14 10:42:44.563 [network debug] [00025] wait for result
2016-05-14 10:42:44.564 [main debug] started thread 1953494096=connection
2016-05-14 10:42:44.564 [network info] [00025] client connection opened 10.6.0.1
2016-05-14 10:42:44.564 [main debug] >>> read -c bai currenterror
2016-05-14 10:42:44.674 [main debug] <<< ERR: command not found
2016-05-14 10:42:44.674 [network info] [00025] connection closed
2016-05-14 10:42:44.816 [main debug] started thread 1919939664=connection
2016-05-14 10:42:44.816 [network info] [00026] client connection opened 10.6.0.1
2016-05-14 10:42:44.816 [network debug] [00026] wait for result
2016-05-14 10:42:44.816 [main debug] >>> read -c hwc OperatingMode2
2016-05-14 10:42:44.674 [main debug] <<< ERR: command not found
2016-05-14 10:42:44.674 [network info] [00025] connection closed
2016-05-14 10:42:44.816 [main debug] started thread 1919939664=connection
2016-05-14 10:42:44.816 [network info] [00026] client connection opened 10.6.0.1
2016-05-14 10:42:44.816 [network debug] [00026] wait for result
2016-05-14 10:42:44.816 [main debug] >>> read -c hwc OperatingMode2
2016-05-14 10:42:48.479 [network info] [00028] client connection opened 10.6.0.1
2016-05-14 10:42:48.478 [main debug] >>> read time
2016-05-14 10:42:48.793 [network info] [00029] client connection opened 10.6.0.1
2016-05-14 10:42:48.793 [main error] read ui OutsideTemp cached: ERR: element not found
2016-05-14 10:42:48.793 [main debug] <<< ERR: element not found
2016-05-14 10:42:48.794 [network debug] [00029] wait for result
2016-05-14 10:42:48.794 [main debug] >>>
2016-05-14 10:42:48.894 [main debug] <<< ERR: command not found
2016-05-14 10:42:48.895 [network info] [00029] connection closed
2016-05-14 10:42:49.043 [network debug] [00030] wait for result
2016-05-14 10:42:49.043 [main debug] started thread 1928328272=connection
2016-05-14 10:42:49.043 [network info] [00030] client connection opened 10.6.0.1
2016-05-14 10:42:49.043 [main debug] performing regular tasks
2016-05-14 10:42:49.043 [main debug] >>> read -c bai currenterror
2016-05-14 10:42:49.141 [main debug] <<< ERR: command not found
2016-05-14 10:42:49.141 [network info] [00030] connection closed
2016-05-14 10:42:49.278 [main debug] started thread 1936716880=connection
2016-05-14 10:42:49.278 [network info] [00031] client connection opened 10.6.0.1
2016-05-14 10:42:49.278 [main debug] >>> read -c hwc OperatingMode2

modenet, 2016-05-14:
as you can see some times requests are satisfied, major times I received "ERR: element not found"

john30, 2016-05-14:
I can only assume that you call ebusctl in the wrong way passing some binary codes that are not visible but lead to not being able to parse the input.
Why do you use ebusctl anyway if you're in PHP? You could just directly connect to the TCP socket...

modenet, 2016-05-14:
I don't know how to connect in php via tcp socket...I would happy to do so.
I changed from apache2 to lighttpd and now my php code works (I don't know that it changes from apache2 to lighttpd).
thanks again for your availability

closed 2016-05-14 by modenet