john30/ebusd #21: Build fails on Arch for RPI1

Crazyachmed, 2016-05-15:
I'm trying to compile for on my RPI1:
Linux ebus 4.4.10-1-ARCH #1 Fri May 13 18:56:03 MDT 2016 armv6l GNU/Linux

The build fails with the following errors (lots and lots of similar output before)

> data.cpp:1632:49: note:   ‘std::basic_istream<char>’ is not derived from ‘const std::unique_ptr<_Tp, _Dp>’
>     else if (getline(input, token, separator) == 0)
>                                                  ^
> In file included from /usr/include/c++/6.1.1/bits/locale_conv.h:41:0,
>                  from /usr/include/c++/6.1.1/locale:43,
>                  from /usr/include/c++/6.1.1/iomanip:43,
>                  from data.cpp:26:
> /usr/include/c++/6.1.1/bits/unique_ptr.h:652:5: note: candidate: template<class _Tp, class _Dp> bool std::operator==(const std::unique_ptr<_Tp, _Dp>&, std::nullptr_t)
>      operator==(const unique_ptr<_Tp, _Dp>& __x, nullptr_t) noexcept
>      ^~~~~~~~
> /usr/include/c++/6.1.1/bits/unique_ptr.h:652:5: note:   template argument deduction/substitution failed:
> data.cpp:1632:49: note:   ‘std::basic_istream<char>’ is not derived from ‘const std::unique_ptr<_Tp, _Dp>’
>     else if (getline(input, token, separator) == 0)
>                                                  ^
> In file included from /usr/include/c++/6.1.1/bits/locale_conv.h:41:0,
>                  from /usr/include/c++/6.1.1/locale:43,
>                  from /usr/include/c++/6.1.1/iomanip:43,
>                  from data.cpp:26:
> /usr/include/c++/6.1.1/bits/unique_ptr.h:657:5: note: candidate: template<class _Tp, class _Dp> bool std::operator==(std::nullptr_t, const std::unique_ptr<_Tp, _Dp>&)
>      operator==(nullptr_t, const unique_ptr<_Tp, _Dp>& __x) noexcept
>      ^~~~~~~~
> /usr/include/c++/6.1.1/bits/unique_ptr.h:657:5: note:   template argument deduction/substitution failed:
> data.cpp:1632:49: note:   mismatched types ‘const std::unique_ptr<_Tp, _Dp>’ and ‘int’
>     else if (getline(input, token, separator) == 0)
>                                                  ^
> Makefile:349: recipe for target 'data.o' failed
> make[2]: *** [data.o] Error 1
> make[2]: Leaving directory '/home/flo/ebusd-2.1/contrib/archlinux/src/ebusd-2.1/src/lib/ebus'
> Makefile:372: recipe for target 'all-recursive' failed
> make[1]: *** [all-recursive] Error 1
> make[1]: Leaving directory '/home/flo/ebusd-2.1/contrib/archlinux/src/ebusd-2.1'
> Makefile:313: recipe for target 'all' failed
> make: *** [all] Error 2
> ==> ERROR: A failure occurred in build().
>     Aborting...
>

john30, 2016-05-15:
could you check again with commit a1a2eca please?

Crazyachmed, 2016-05-20:
I already changed the MD5 in the PKGBUILD to get it to do anything, so this is another error. Or did you change anything else?

john30, 2016-05-20:
just use the PKGBUILD.git instead

Crazyachmed, 2016-05-20:
Still fails in a similar fashion. Here is the full output: http://pastebin.com/umE5bbmG

john30, 2016-05-21:
please check again with commit 6e99890

Crazyachmed, 2016-05-21:
That did it, build succeeded

closed 2016-05-21 by Crazyachmed