john30/ebusd #22: Some help needed with Valliant VCW376 without controller

destans, 2016-08-19:
Hi, i'm in need of some help. First i want to say i'm new to all of this so bare with me ;-)
I have a VCW376 heater and i have connected it to a ebus to usb converter(eservice) and connected everything to a PI2. Installed ebusd and started up service. So far so good i can read values with the 'r' command in ebusctl and response shows up ok.

What i was expecting that updates would come in to ebusd automatically. So if a value changed in de heater and these new values where to be submitted on the bus that i would be able to see these changes with the 'listen' command in ebusctl. But this is not the case. There are no updates. The only way i seem to get updates is when i configure a poll priority on de 'r' command.

So my question is, is this normal behaviour or is this due to the fact that i don't have a controller connected to my setup.

john30, 2016-08-19:
the cause may be the missing controller on the one hand and on the other missing configuration file for the heater. please post the result of the following commands:
`ebusctl info && ebusctl scan && sleep 30 && ebusctl scan result`

destans, 2016-08-19:
Hi john,

This is the result, any ideas???

pi@piServer /var/run $ ebusctl info && ebusctl scan && sleep 30 && ebusctl scan result
version: ebusd 2.1.28b50d2
signal: acquired
symbol rate: 22
masters: 2
messages: 209
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "bai.308523.inc", "vaillant/08.bai.csv"
address 31: master #8, ebusd
address 36: slave #8

done

08;Vaillant;BAI00;0518;7401;21;09;07;0010003812;0001;005437;N5

pi@piServer /var/run $

destans, 2016-08-19:
Hi john,

This is what i get when i enable 'raw' in ebusctl

2016-08-19 17:22:40.781 [bus notice] <aa
2016-08-19 17:22:40.824 [bus notice] <aa
2016-08-19 17:22:40.870 [bus notice] <aa
2016-08-19 17:22:40.915 [bus notice] <aa
2016-08-19 17:22:40.960 [bus notice] <aa
2016-08-19 17:22:41.007 [bus notice] <aa
2016-08-19 17:22:41.008 [bus notice] >31
2016-08-19 17:22:41.014 [bus notice] <31
2016-08-19 17:22:41.015 [bus notice] >08
2016-08-19 17:22:41.021 [bus notice] <08
2016-08-19 17:22:41.022 [bus notice] >b5
2016-08-19 17:22:41.028 [bus notice] <b5
2016-08-19 17:22:41.029 [bus notice] >09
2016-08-19 17:22:41.035 [bus notice] <09
2016-08-19 17:22:41.036 [bus notice] >01
2016-08-19 17:22:41.042 [bus notice] <01
2016-08-19 17:22:41.043 [bus notice] >24
2016-08-19 17:22:41.049 [bus notice] <24
2016-08-19 17:22:41.050 [bus notice] >48
2016-08-19 17:22:41.056 [bus notice] <48
2016-08-19 17:22:41.060 [bus notice] <00
2016-08-19 17:22:41.082 [bus notice] <09
2016-08-19 17:22:41.086 [bus notice] <00
2016-08-19 17:22:41.090 [bus notice] <32
2016-08-19 17:22:41.095 [bus notice] <31
2016-08-19 17:22:41.099 [bus notice] <30
2016-08-19 17:22:41.103 [bus notice] <39
2016-08-19 17:22:41.107 [bus notice] <30
2016-08-19 17:22:41.112 [bus notice] <37
2016-08-19 17:22:41.116 [bus notice] <30
2016-08-19 17:22:41.120 [bus notice] <30
2016-08-19 17:22:41.124 [bus notice] <4f
2016-08-19 17:22:41.126 [bus notice] >00
2016-08-19 17:22:41.132 [bus notice] <00
2016-08-19 17:22:41.133 [bus notice] >aa
2016-08-19 17:22:41.139 [bus notice] <aa
2016-08-19 17:22:41.185 [bus notice] <aa
2016-08-19 17:22:41.186 [bus notice] >31
2016-08-19 17:22:41.192 [bus notice] <31
2016-08-19 17:22:41.193 [bus notice] >08
2016-08-19 17:22:41.199 [bus notice] <08
2016-08-19 17:22:41.200 [bus notice] >b5
2016-08-19 17:22:41.206 [bus notice] <b5
2016-08-19 17:22:41.207 [bus notice] >09
2016-08-19 17:22:41.213 [bus notice] <09
2016-08-19 17:22:41.214 [bus notice] >01
2016-08-19 17:22:41.220 [bus notice] <01
2016-08-19 17:22:41.221 [bus notice] >24
2016-08-19 17:22:41.227 [bus notice] <24
2016-08-19 17:22:41.228 [bus notice] >48
2016-08-19 17:22:41.234 [bus notice] <48
2016-08-19 17:22:41.238 [bus notice] <00
2016-08-19 17:22:41.257 [bus notice] <09
2016-08-19 17:22:41.259 [bus notice] <00
2016-08-19 17:22:41.264 [bus notice] <32
2016-08-19 17:22:41.268 [bus notice] <31
2016-08-19 17:22:41.272 [bus notice] <30
2016-08-19 17:22:41.276 [bus notice] <39
2016-08-19 17:22:41.281 [bus notice] <30
2016-08-19 17:22:41.285 [bus notice] <37
2016-08-19 17:22:41.289 [bus notice] <30
2016-08-19 17:22:41.293 [bus notice] <30
2016-08-19 17:22:41.298 [bus notice] <4f
2016-08-19 17:22:41.299 [bus notice] >00
2016-08-19 17:22:41.305 [bus notice] <00
2016-08-19 17:22:41.306 [bus notice] >aa
2016-08-19 17:22:41.312 [bus notice] <aa
2016-08-19 17:22:41.356 [bus notice] <aa
2016-08-19 17:22:41.357 [bus notice] >31
2016-08-19 17:22:41.363 [bus notice] <31
2016-08-19 17:22:41.365 [bus notice] >08
2016-08-19 17:22:41.370 [bus notice] <08
2016-08-19 17:22:41.372 [bus notice] >b5
2016-08-19 17:22:41.377 [bus notice] <b5
2016-08-19 17:22:41.379 [bus notice] >09
2016-08-19 17:22:41.384 [bus notice] <09
2016-08-19 17:22:41.386 [bus notice] >01
2016-08-19 17:22:41.391 [bus notice] <01
2016-08-19 17:22:41.393 [bus notice] >24
2016-08-19 17:22:41.398 [bus notice] <24
2016-08-19 17:22:41.400 [bus notice] >48
2016-08-19 17:22:41.405 [bus notice] <48
2016-08-19 17:22:41.410 [bus notice] <00
2016-08-19 17:22:41.431 [bus notice] <09
2016-08-19 17:22:41.435 [bus notice] <00
2016-08-19 17:22:41.439 [bus notice] <32
2016-08-19 17:22:41.443 [bus notice] <31
2016-08-19 17:22:41.447 [bus notice] <30
2016-08-19 17:22:41.452 [bus notice] <39
2016-08-19 17:22:41.456 [bus notice] <30
2016-08-19 17:22:41.460 [bus notice] <37
2016-08-19 17:22:41.464 [bus notice] <30
2016-08-19 17:22:41.468 [bus notice] <30
2016-08-19 17:22:41.473 [bus notice] <4f
2016-08-19 17:22:41.474 [bus notice] >00
2016-08-19 17:22:41.480 [bus notice] <00
2016-08-19 17:22:41.481 [bus notice] >aa
2016-08-19 17:22:41.487 [bus notice] <aa
2016-08-19 17:22:41.532 [bus notice] <aa
2016-08-19 17:22:41.533 [bus notice] >31
2016-08-19 17:22:41.539 [bus notice] <31
2016-08-19 17:22:41.540 [bus notice] >08
2016-08-19 17:22:41.546 [bus notice] <08
2016-08-19 17:22:41.547 [bus notice] >b5
2016-08-19 17:22:41.553 [bus notice] <b5
2016-08-19 17:22:41.555 [bus notice] >09
2016-08-19 17:22:41.560 [bus notice] <09
2016-08-19 17:22:41.562 [bus notice] >01
2016-08-19 17:22:41.567 [bus notice] <01
2016-08-19 17:22:41.569 [bus notice] >24
2016-08-19 17:22:41.575 [bus notice] <24
2016-08-19 17:22:41.576 [bus notice] >48
2016-08-19 17:22:41.582 [bus notice] <48
2016-08-19 17:22:41.586 [bus notice] <00
2016-08-19 17:22:41.607 [bus notice] <09
2016-08-19 17:22:41.612 [bus notice] <00
2016-08-19 17:22:41.616 [bus notice] <32
2016-08-19 17:22:41.620 [bus notice] <31
2016-08-19 17:22:41.624 [bus notice] <30
2016-08-19 17:22:41.629 [bus notice] <39
2016-08-19 17:22:41.632 [bus notice] <30
2016-08-19 17:22:41.636 [bus notice] <37
2016-08-19 17:22:41.640 [bus notice] <30
2016-08-19 17:22:41.644 [bus notice] <30
2016-08-19 17:22:41.649 [bus notice] <4f
2016-08-19 17:22:41.651 [bus notice] >00
2016-08-19 17:22:41.657 [bus notice] <00
2016-08-19 17:22:41.658 [bus notice] poll scan.08 id: 
2016-08-19 17:22:41.659 [bus notice] >aa
2016-08-19 17:22:41.664 [bus notice] <aa
2016-08-19 17:22:41.712 [bus notice] <aa
2016-08-19 17:22:41.756 [bus notice] <aa
2016-08-19 17:22:41.803 [bus notice] <aa

john30, 2016-08-21:
this is just the raw output from the scan, so there is no extra relevant information in that one.
anyway, i think the major issue is the absence of a controller which would regularly read data from the heater. since that is missing, you need to poll for the values you're interested on your own.
ebusd can do this for you automatically, if you change the "r" in the first field of the bai.308523.inc to e.g. "r1" in order to have that row polled with priority 1.
After that, restart ebusd and you should see "update" lines in the log file accordingly.

closed 2016-10-25 by john30