john30/ebusd #23: Chaffoteaux ebus port

tikismoke, 2016-10-01:
A chaffoteaux serelia green that as motherboard replace and then have 3 ebus port.
1 directly on the new motherboard (24VDC ebus2 from chaffoteaux docs) ans still the 2 other old ebus por from origin (19VDC)
John30 told me than without other device connect to the bus i can not intercept any thing.
I only saw datetime:
```
2016-10-01 11:02:14.708 [update notice] unknown BC cmd: 70fe20100419190001
2016-10-01 11:02:19.158 [update error] unable to parse broadcast datetime from 10fe070009800fffffffffffffff / : ERR: argument value out of valid range

```
So i found an ambiant thermostat for this boiler and wire it to the ebus2 port (self power via ebus).
unfortunalyy no more things on the old ebus port.

I wired the eservice usb adapter to the ebus2 port 2, now i saw more things:
```

ebusctl info && ebusctl scan && sleep 30 && ebusctl scan result
version: ebusd 2.1.e408c94
signal: acquired
symbol rate: 16
masters: 3
messages: 13
address 10: master #2
address 15: slave #2, scanned "MF=0;ID=" error: ERR: invalid position
address 31: master #8, ebusd
address 36: slave #8
address 70: master #4

```

But in ebusd i got those sort of things:
```

2016-10-01 11:04:18.972 [update error] unable to parse broadcast datetime from 10fe070009800fffffffffffffff / : ERR: argument value out of valid range
2016-10-01 11:04:44.425 [update notice] unknown BC cmd: 70fe20100419190001
2016-10-01 11:04:44.458 [update notice] unknown BC cmd: 70fe203b0175
2016-10-01 11:04:44.509 [update notice] unknown BC cmd: 70fe2010057118270101

```

I am sure chaffoteaux/aristion/etc ebus do not respect 100% of the protocole and send personnals things

How can i hope to get more information from.

tikismoke, 2016-10-01:
`address 70: master #4`

Was not exist before i installed the thermostat.

john30, 2016-10-02:
According to the messages, the original manufacturer seems to be "RAWE", for which I don't have any configuration files yet. So this is going to by a lot like try-and-error to find out what is possible via the bus.

First of all, you need to get your coupler working correctly. Please send me the first 500 lines of your log file (/var/log/ebusd.log), then I can tell you more if this is the case.

From the 10fe7000 message, your outside temperature seems to be 0.06 degrees (800f). Does this sound right?

The other "unknown BC" message come from the room thermostat and contain RAWE specific data. You should watch those for a period of time and check if the values are somehow aligned to real values like the room temperature.

Another thing you can do is to use the room controller to initiate messages on the eBus. Do you have any user manual available for your heating system and/or the room thermostat?

tikismoke, 2016-10-02:
> So this is going to by a lot like try-and-error to find out what is possible via the bus.

Ready for challlenge 👍 Note that on all docs i've found they call the bus BUS BridgeNet.

> First of all, you need to get your coupler working correctly. Please send me the first 500 lines of your log file (/var/log/ebusd.log), then I can tell you more if this is the case.

Going to Pastbin it i'll saw lots of:
`2016-10-02 14:42:55.619 [bus error] prepare poll message: ERR: invalid address`
http://pastebin.com/73k5F06x

> From the 10fe7000 message, your outside temperature seems to be 0.06 degrees (800f). Does this sound right?

Not so cold here :) It could be the inside temp as the boiler as no inside temp (value is to 0.00°C in internal boiler advanced menu).

> Another thing you can do is to use the room controller to initiate messages on the eBus. Do you have any user manual available for your heating system and/or the room thermostat?

Once i could understand how it works yes. The thermostat seems to not even sync date with the boiler.
I'm pretty sure it only send boil or not on the ebus. It looks like a simple thermostat but not with dry contact. docs here: http://www.chaffoteaux.be/upload/docuteque/file_52dfb79a0fe6b.pdf

It looks like exactly like this one in the vidéo: https://www.youtube.com/watch?v=I6fjAITnhAU

For boiler the fr docs is here:
http://www.enrdd.com/documents/documents/Documentation-Constructeurs/Chaffoteaux-Ariston-Styx-Thermor/CX%20SERELIA%20Green%20notice%20installation%20.pdf

Sorry for my poor english :(

john30, 2016-10-02:
okay, your coupler seems to work fine.
But it seems you've started ebusd in the wrong manner. Your /etc/default/ebusd file should contain a "--scanconfig" in the EBUSD_OPTS1 (that should not be preceeded with a "#").
Actually, in your case it would be better to disable the --scanconfig and start with a blank configuration directory /etc/ebusd.
Then, only add the 3 basic CSV files "_templates.csv", "broadcast.csv", and "memory.csv" from here: [ebusd-configuration/ebusd-2.1.x/en](https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.1.x/en).

After that, we should start with a manual scan since the regular one does not seem to work. For each circuit appearing in the "info" command result, issue the following hex command (you will have to add "--enablehex" to the ebusd start parameters in /etc/default/ebusd or when manually starting ebusd in order to use that command at all):

ebusctl hex 15070400

Do this for each slave address shown in the "info" result (i.e. replace "15" by each slave address).

tikismoke, 2016-10-02:
```

pi@raspberrybplus:~ $ ebusctl hex 15070400
00

```

```
2016-10-02 15:51:50.313 [main notice] hex cmd: 3115070400
2016-10-02 15:51:50.593 [update notice] update broadcast ident QQ=10: TEM;11111;0100;0100

```

Fo other slave or master i got: ERR: read timeout

`[bus error] send to 70: ERR: read timeout, retry`

tikismoke, 2016-10-02:
ebusctl hex 10070400
done

2016-10-02 15:54:54.850 [main notice] hex cmd: 3110070400
2016-10-02 15:54:55.120 [update notice] update broadcast ident QQ=10: TEM;11111;0100;0100

tikismoke, 2016-10-02:
When i play with temp target on the thermostat i got:
```
2016-10-02 16:14:36.303 [update notice] unknown BC cmd: 70fe20100419190101
2016-10-02 16:14:49.476 [update notice] unknown BC cmd: 70fe20100419190001
2016-10-02 16:14:54.441 [update notice] unknown BC cmd: 70fe20100419190101
2016-10-02 16:14:58.446 [update notice] unknown BC cmd: 70fe20100419190001

```

0101 seems to be for boil ON and 0001 boi OFF

john30, 2016-10-03:
hm, okay then your heater seems to be from TEM, interesting. Is the room controller from the same manufacturer?

tikismoke, 2016-10-03:
Hi the controller is the same manufacter (chaffoteaux) but it looks like generic nearly like this honeywell:
https://products.ecc.emea.honeywell.com/france/pdf/cm707-fr0p0280-fr03r0206.pdf

With buttons off auto manual not hide.

I put it on the bus but after 5 day of use (still not heating in this period) i'm even not sure it really works to command the boiler.

Chaffoteaux seems to have two protocols (ebus and ebus2) one is also known as bridgenet but not sure it works and are retro compatible.
My motherboard as been changed in the past. The new one is compatible (with same eeprom) with all and new "serelia green" generation. The old one was using an RJ11 to additional electronic card containing 2 ebus port in 19V=. The new one still have this connector and still send datat to the 19V ebus and have another ebus port in 24V= (this iw also used to power ebus device such as the thermosstat). I've tried both port with the adapter. I get thermostat data only when connecting it to the 24V port in // of the thermostat. It do not seems that frame pass from one port to the other.

For chat maybe in the evening.

john30, 2016-10-08:
well that's my impression, too. the room controller does not seem to fit to your heating system.

tikismoke, 2016-10-08:
Will try to find a new one full compatible.
Will take time. I keep you in touch.

philb29, 2016-10-09:
Hello,
This is interesting because I just started playing with ebusd and my Chaffoteaux Pharos Zelios (Year 2012).
1st, Chaffoteaux has 3 types of buses on their hardware : 
- old system with RJ11. ( I have one port like this on my Pharos Zelios motherboard, connected to a EBT2 - eBus Interface, which from I get 2 eBus ports. But none is used.
- eBus 1 : I have one port on my motherboard, on which is connected a mixer for 2 zones, and a room controller for Zone 2.
- eBus 2 : I dont't have this one. only newer Chaffoteaux Pharos Zelios have this bus. maybe starting year 2013 or 2015, I don't remember. eBus2 is compatible with their connected solution Chaffolink.

@tikismoke : are you sure you have a ebus 2 motherboard? from the French documentation you provided, It looks like you have the same control panel as my pharos zelios (orange display)... newer pharos zelios have a gray LCD display. I would be surprised that newer motherboard with ebus 2 can use the old orange display.

Chaffoteaux hardware is badged "Thermowatt", but in fact it's TEM hardware.

So... I started sniffing the bus with ebusd : 

The protocol used by Chaffoteaux looks like very customized :'(

I have trouble to get hardware recognized by ebusd; the slave seems to take too much time to answer and ebusd seems to take the last received frame as answer while the identification answer will come after... too late...

```
$ ebusctl hex 15070400
0ab4830000020000000000

```
```
2016-10-09 19:48:54.086 [update notice] unknown MS cmd: 301506210407b42040 / 0ab4830000020000000000
2016-10-09 19:48:58.547 [network info] [00033] client connection opened 127.0.0.1
2016-10-09 19:48:58.547 [main notice] hex cmd: 3315070400
2016-10-09 19:48:58.547 [bus info] send message: 3315070400
2016-10-09 19:48:58.715 [network info] [00033] connection closed
2016-10-09 19:48:58.913 [update info] update BC cmd: 10fe07040a10313131313101000100
2016-10-09 19:48:58.913 [update notice] update broadcast ident QQ=10: TEM;11111;0100;0100

```
and the ebusctl info command show wrong thing....
```
$ ebusctl info
version: ebusd 2.1.6311149
signal: acquired
symbol rate: 60
masters: 5
messages: 55
address 10: master #2
address 11: master #7
address 15: slave #2, scanned "**MF=Vaillant**;ID=     ;SW=" error: ERR: argument value out of valid range
address 30: master #3
address 31: master #8
address 33: master #13, ebusd
address 35: slave #3, scanned
address 38: slave #13

```
I use a eBUS Koppler Ethernet  from E-Service GmbH; and here is how I launch ebusd:
`EBUSD_OPTS="--scanconfig -d /dev/virtualcom0 -n --loglevel=info --latency=10000 --enablehex --address 33 --receivetimeout 100000"
`
Note that when I only have the boiler connected to the eBus Koppler, the slave at address 15 is correctly recognized, because the bus is not loaded.

How can I solve this ?? the receivetimeout is already at maximum...

Next I didn't find any doc or configuration files for TEM.
Here is my deductions for this message sent every 10 seconds by the room controller: 
`2016-10-09 19:49:12.433 [update info] update MS cmd: 11151011058014e10006 / 09010280080000760002
`
Address 01 is for Zone 1 room controller
Address 11 is for Zone 2 room controller
Address 31 is for Zone 3 room controller
Address 15 is the boiler
So from the above telegram : 
0x11 : source address (Zone 2 room controller)
0x15 : dst address (boiler)
0x101105 : ????
0x8014 : setpoint temperature ( 20.5°C )
0xe1 : actual temperature ( E1 => 225 => 22.5°C )
0x00 : ???
0x06 : mode : 0x00 : manual OFF; 0x01 : auto comfort mode (sun); 0x05 : manual comfort mode (sun); 0x06 : manual Eco mode (moon)

For the response:
0x09 : ???
0x01 :  01 (setpoint ECO), or 00 (setpoint is OFF) or 02 (setpoint comfort) or 41 (boiler ON in ECO mode) or 42 (boiler ON in comfort mode) 
0x02 : ?? can be 00 when Frost mode (??) or 02 or 03 or 04
0x80 : ?????
0x08 : ??? look like a pre-circulation or circulation (44), or post-circulation state; seen values are 04, 08, 40, 44)
0x0000 : ???
0x76 :  looks like a counter, value increment every 30 seconds. ??
0x00 : ???
0x02 : ?? seen values are 0x02 or 0x03.

Any help appreciated for constructing the csv file. Does someone already have used ebusd with Chaffoteaux ( or TEM ) hardware ?

And how to deal with the datetime broadcasting because all date and time fields are FF, and ebusd complains about argument value out of valid range? How can I get the ext temp decoded and datetime ignored?

tikismoke, 2016-10-09:
Nice to hear i'm not alone to try.

Same as you the motherboard as an rj11 port connected to a small card providing 2 ebus port at 19v. Yes it s orange screen like yours.
The new motherboard also have another ebus port directly on the board providing 24v.

My boiler is a serelia green from 2008.
I'm using the eservice USB adapter.

Let me know if you find something usefull I can try to help both of us.

I'm happy to not have buy the adapter useless now that we are at least both.

tikismoke, 2016-10-09:
Pour info j'ai le même afficheur que dans cette doc là page 23:
http://www.enrdd.com/documents/documents/Documentation-Constructeurs/Chaffoteaux-Ariston-Styx-Thermor/pharos%20zelios%20notice%20installation.pdf

Et le connecteur ebus de base en RJ11 comme page 20 avec la carte additionnelle.

tikismoke, 2016-10-10:
I removed the thermostat from the ebus.

now when sending
```

ebusctl hex 15070400
00
2016-10-10 22:51:53.546 [main notice] hex cmd: 3115070400
2016-10-10 22:51:53.824 [update notice] unknown BC cmd: 10fe07040a10313131313101000100

```

And same with: 

```
ebusctl hex 10070400
done

2016-10-10 22:50:51.994 [bus notice] new master 10, master count 2
2016-10-10 22:50:52.208 [update notice] unknown BC cmd: 10fe07040a10313131313101000100

 ebusctl info
version: ebusd 2.1.e408c94
signal: acquired
symbol rate: 17
masters: 2
messages: 1
address 10: master #2
address 15: slave #2, scanned "MF=0;ID=" error: ERR: invalid position
address 31: master #8, ebusd
address 36: slave #8

```

john30, 2016-10-13:
the device seems to answer with a broadcast identification message instead of answering to the request.
could you run the same command again (ebusctl hex 15070400) with full logging enabled and send me the log please?

tikismoke, 2016-10-13:
```
2016-10-13 18:04:03.480 [bus debug] ERR: read timeout during ready, switching to skip
2016-10-13 18:04:03.539 [bus debug] ERR: read timeout during ready, switching to skip
2016-10-13 18:04:03.599 [bus debug] ERR: read timeout during ready, switching to skip
2016-10-13 18:04:03.659 [bus debug] ERR: read timeout during ready, switching to skip
2016-10-13 18:04:03.719 [bus debug] ERR: read timeout during ready, switching to skip
2016-10-13 18:04:03.738 [network info] [00001] client connection opened 127.0.0.1
2016-10-13 18:04:03.743 [main debug] performing regular tasks
2016-10-13 18:04:03.746 [main debug] >>> hex 15070400
2016-10-13 18:04:03.750 [main notice] hex cmd: 3115070400
2016-10-13 18:04:03.751 [bus info] send message: 3115070400
2016-10-13 18:04:03.753 [network debug] [00001] wait for result
2016-10-13 18:04:03.780 [bus debug] ERR: read timeout during ready, switching to skip
2016-10-13 18:04:03.794 [bus debug] switching from ready to send command
2016-10-13 18:04:03.831 [bus debug] switching from send command to receive command ACK
2016-10-13 18:04:03.835 [bus debug] switching from receive command ACK to receive response
2016-10-13 18:04:03.844 [bus debug] switching from receive response to send response ACK
2016-10-13 18:04:03.854 [bus debug] notify request: done
2016-10-13 18:04:03.855 [bus notice] new master 10, master count 2
2016-10-13 18:04:03.857 [bus debug] read res: 00
2016-10-13 18:04:03.861 [main debug] <<< 00
2016-10-13 18:04:03.877 [network info] [00001] connection closed
2016-10-13 18:04:03.880 [bus debug] switching from send response ACK to send SYN
2016-10-13 18:04:03.999 [bus debug] ERR: read timeout during ready, switching to skip
2016-10-13 18:04:04.076 [update info] update BC cmd: 10fe07040a10313131313101000100
2016-10-13 18:04:04.078 [update notice] unknown BC cmd: 10fe07040a10313131313101000100
2016-10-13 18:04:04.135 [bus debug] ERR: read timeout during ready, switching to skip
2016-10-13 18:04:04.192 [bus debug] ERR: read timeout during ready, switching to skip
2016-10-13 18:04:04.249 [bus debug] ERR: read timeout during ready, switching to skip
2016-10-13 18:04:04.308 [bus debug] ERR: read timeout during ready, switching to skip
2016-10-13 18:04:04.369 [bus debug] ERR: read timeout during ready, switching to skip
2016-10-13 18:04:04.428 [bus debug] ERR: read timeout during ready, switching to skip
2016-10-13 18:04:04.488 [bus debug] ERR: read timeout during ready, switching to skip
2016-10-13 18:04:04.548 [bus debug] ERR: read timeout during ready, switching to skip
2016-10-13 18:04:04.608 [bus debug] ERR: read timeout during ready, switching to skip
2016-10-13 18:04:04.668 [bus debug] ERR: read timeout during ready, switching to skip
2016-10-13 18:04:04.728 [bus debug] ERR: read timeout during ready, switching to skip

```

john30, 2016-10-13:
ok, so the device simple answers with "00" (i.e. a zero length acknowledged answer) instead of the expected values. that's why the scan result is not able to parse that empty answer.
furthermore, it sends a broadcast identification message containing the expected values. weird behaviour...

tikismoke, 2016-10-13:
i'm avaiable on gitter to try more to get help if possible

philb29, 2016-10-14:
Near the same behavior here, but my device doesn't answer with "00" :
In fact it looks like the device does his job before answering, and ebusd take the 1st message coming from the device like a response. Look below : 

11151011050005d10000 / 090002800800002b0002 is something sent every 10 seconds on my bus :
```
2016-10-14 22:24:46.727 [update notice] unknown MS cmd: 11151011050005d10000 / 09000280080000280002
2016-10-14 22:24:56.530 [update notice] unknown MS cmd: 11151011050005d10000 / 09000280080000280002
2016-10-14 22:25:06.497 [update notice] unknown MS cmd: 11151011050005d10000 / 09000280080000280002
2016-10-14 22:25:16.438 [update notice] unknown MS cmd: 11151011050005d10000 / 09000280080000290002
2016-10-14 22:25:26.371 [update notice] unknown MS cmd: 11151011050005d10000 / 09000280080000290002
2016-10-14 22:25:36.331 [update notice] unknown MS cmd: 11151011050005d10000 / 09000280080000290002
2016-10-14 22:25:46.280 [update notice] unknown MS cmd: 11151011050005d10000 / 090002800800002a0002
2016-10-14 22:25:56.264 [update notice] unknown MS cmd: 11151011050005d10000 / 090002800800002a0002
2016-10-14 22:26:06.173 [update notice] unknown MS cmd: 11151011050005d10000 / 090002800800002a0002

```

Log of ebusctl hex 15070400 :

```
2016-10-14 22:26:16.164 [network info] [00031] client connection opened 127.0.0.1
2016-10-14 22:26:16.165 [main debug] >>> hex 15070400
2016-10-14 22:26:16.165 [main notice] hex cmd: 3315070400
2016-10-14 22:26:16.165 [bus info] send message: 3315070400
2016-10-14 22:26:16.165 [network debug] [00031] wait for result
2016-10-14 22:26:16.173 [update info] update MS cmd: 11151011050005d10000 / 090002800800002b0002
2016-10-14 22:26:16.174 [update notice] unknown MS cmd: 11151011050005d10000 / 090002800800002b0002
2016-10-14 22:26:16.192 [bus debug] switching from ready to send command
2016-10-14 22:26:16.247 [bus debug] switching from send command to receive command ACK
2016-10-14 22:26:16.251 [bus debug] switching from receive command ACK to receive response
2016-10-14 22:26:16.299 [bus debug] switching from receive response to send response ACK
2016-10-14 22:26:16.310 [bus debug] notify request: done
2016-10-14 22:26:16.310 [bus debug] read res: 090002800800002b0002
2016-10-14 22:26:16.310 [bus debug] switching from send response ACK to send SYN
2016-10-14 22:26:16.311 [main debug] <<< 090002800800002b0002
2016-10-14 22:26:16.314 [network info] [00031] connection closed
2016-10-14 22:26:16.508 [update info] update BC cmd: 10fe07040a10313131313101000100
2016-10-14 22:26:16.509 [update notice] update broadcast ident QQ=10: TEM;11111;0100;0100
```
log raw of ebusctl hex 15070400 :
```
2016-10-14 22:40:44.361 [network info] [00040] client connection opened 127.0.0.1
2016-10-14 22:40:44.362 [main debug] >>> hex 15070400
2016-10-14 22:40:44.362 [main notice] hex cmd: 3315070400
2016-10-14 22:40:44.362 [bus info] send message: 3315070400
2016-10-14 22:40:44.362 [network debug] [00040] wait for result
2016-10-14 22:40:44.397 [bus notice] <aa
2016-10-14 22:40:44.397 [bus notice] >33
2016-10-14 22:40:44.408 [bus notice] <33
2016-10-14 22:40:44.408 [bus debug] switching from ready to send command
2016-10-14 22:40:44.408 [bus notice] >15
2016-10-14 22:40:44.419 [bus notice] <15
2016-10-14 22:40:44.419 [bus notice] >07
2016-10-14 22:40:44.430 [bus notice] <07
2016-10-14 22:40:44.430 [bus notice] >04
2016-10-14 22:40:44.441 [bus notice] <04
2016-10-14 22:40:44.441 [bus notice] >00
2016-10-14 22:40:44.452 [bus notice] <00
2016-10-14 22:40:44.452 [bus notice] >0e
2016-10-14 22:40:44.464 [bus notice] <0e
2016-10-14 22:40:44.464 [bus debug] switching from send command to receive command ACK
2016-10-14 22:40:44.468 [bus notice] <00
2016-10-14 22:40:44.468 [bus debug] switching from receive command ACK to receive response
2016-10-14 22:40:44.472 [bus notice] <0a
2016-10-14 22:40:44.477 [bus notice] <b4
2016-10-14 22:40:44.481 [bus notice] <83
2016-10-14 22:40:44.485 [bus notice] <00
2016-10-14 22:40:44.490 [bus notice] <00
2016-10-14 22:40:44.494 [bus notice] <02
2016-10-14 22:40:44.498 [bus notice] <00
[... 23 lines cut ...]
2016-10-14 22:40:44.697 [bus notice] <31
2016-10-14 22:40:44.702 [bus notice] <31
2016-10-14 22:40:44.706 [bus notice] <31
2016-10-14 22:40:44.710 [bus notice] <31
2016-10-14 22:40:44.715 [bus notice] <31
2016-10-14 22:40:44.719 [bus notice] <01
2016-10-14 22:40:44.724 [bus notice] <00
2016-10-14 22:40:44.728 [bus notice] <01
2016-10-14 22:40:44.732 [bus notice] <00
2016-10-14 22:40:44.737 [bus notice] <ca
2016-10-14 22:40:44.737 [update info] update BC cmd: 10fe07040a10313131313101000100
2016-10-14 22:40:44.737 [update notice] update broadcast ident QQ=10: TEM;11111;0100;0100
2016-10-14 22:40:44.741 [bus notice] <aa
2016-10-14 22:40:44.796 [bus notice] <aa
2016-10-14 22:40:44.856 [bus notice] <aa
2016-10-14 22:40:44.916 [bus notice] <aa
2016-10-14 22:40:44.976 [bus notice] <aa
2016-10-14 22:40:45.036 [bus notice] <aa
2016-10-14 22:40:45.096 [bus notice] <aa
2016-10-14 22:40:45.156 [bus notice] <aa
2016-10-14 22:40:45.216 [bus notice] <aa
2016-10-14 22:40:45.275 [bus notice] <aa
2016-10-14 22:40:45.335 [bus notice] <aa

```

john30, 2016-10-15:
@philb29 090002800800002b0002 is not the answer to the identification message but to the request 11151011050005d10000. according to the broadcast ident, your device is manufactured by TEM.

philb29, 2016-10-15:
@john30 I don't think so... I removed all devices on bus, so I have only the eservice-online Ethernet koppler and the motherboard of my boiler. (No mixer, no room controller connected). I have exactly the same answer than @tikismoke : 
```
2016-10-15 11:01:28.712 [network info] [00063] client connection opened 127.0.0.1
2016-10-15 11:01:28.713 [main debug] performing regular tasks
2016-10-15 11:01:28.713 [main debug] >>> hex 15070400
2016-10-15 11:01:28.713 [main notice] hex cmd: 3315070400
2016-10-15 11:01:28.713 [bus info] send message: 3315070400
2016-10-15 11:01:28.714 [network debug] [00063] wait for result
2016-10-15 11:01:28.758 [bus debug] switching from ready to send command
2016-10-15 11:01:28.813 [bus debug] switching from send command to receive command ACK
2016-10-15 11:01:28.818 [bus debug] switching from receive command ACK to receive response
2016-10-15 11:01:28.826 [bus debug] switching from receive response to send response ACK
2016-10-15 11:01:28.837 [bus debug] notify request: done
2016-10-15 11:01:28.838 [bus debug] read res: 00
2016-10-15 11:01:28.838 [bus debug] switching from send response ACK to send SYN
2016-10-15 11:01:28.838 [main debug] <<< 00
2016-10-15 11:01:28.841 [network info] [00063] connection closed
2016-10-15 11:01:29.036 [update info] update BC cmd: 10fe07040a10313131313101000100
2016-10-15 11:01:29.037 [update notice] update broadcast ident QQ=10: TEM;11111;0100;0100
2016-10-15 11:01:29.713 [network debug] dead connection removed - 0

```

When adding the room controller, the answer to the identification message become the same than the answer to the room controller; and after that the identification broadcast come: 
It's exactly the same behavior than @tikismoke . But the answer is not 00 because the room controller is present on bus.
```
2016-10-15 11:04:24.535 [update info] update MS cmd: 111506210403b51040 / 0ab5010b02001e000a8014
2016-10-15 11:04:24.535 [update notice] unknown MS cmd: 111506210403b51040 / 0ab5010b02001e000a8014
2016-10-15 11:04:28.035 [network info] [00064] client connection opened 127.0.0.1
2016-10-15 11:04:28.036 [main debug] performing regular tasks
2016-10-15 11:04:28.036 [main debug] >>> hex 15070400
2016-10-15 11:04:28.036 [main notice] hex cmd: 3315070400
2016-10-15 11:04:28.036 [bus info] send message: 3315070400
2016-10-15 11:04:28.037 [network debug] [00064] wait for result
2016-10-15 11:04:28.051 [bus debug] switching from ready to send command
2016-10-15 11:04:28.114 [bus debug] switching from send command to receive command ACK
2016-10-15 11:04:28.118 [bus debug] switching from receive command ACK to receive response
2016-10-15 11:04:28.171 [bus debug] switching from receive response to send response ACK
2016-10-15 11:04:28.184 [bus debug] notify request: done
2016-10-15 11:04:28.184 [bus debug] read res: 0ab5010b02001e000a8014
2016-10-15 11:04:28.184 [bus debug] switching from send response ACK to send SYN
2016-10-15 11:04:28.184 [main debug] <<< 0ab5010b02001e000a8014
2016-10-15 11:04:28.188 [network info] [00064] connection closed
2016-10-15 11:04:28.379 [update info] update BC cmd: 10fe07040a10313131313101000100
2016-10-15 11:04:28.379 [update notice] update broadcast ident QQ=10: TEM;11111;0100;0100
2016-10-15 11:04:29.035 [network debug] dead connection removed - 0
2016-10-15 11:04:32.976 [update info] update MS cmd: 11151011050005c70000 / 09000280081144090002
2016-10-15 11:04:32.976 [update notice] unknown MS cmd: 11151011050005c70000 / 09000280081144090002

```

john30, 2016-10-15:
on which address is the room controller? if the address is already taken, this would explain the mixture of messages.

philb29, 2016-10-15:
OK, let's testing that !

ebusd is launched with --address 33

Between tests, Boiler was electrically shutdown and ebusd restarted.

Boiler only :
```
root@mon2:/var/log# ebusctl scan
empty

root@mon2:/var/log# ebusctl info
version: ebusd 2.1.0c0ea8e
signal: acquired
symbol rate: 17
masters: 2
messages: 8
address 10: master #2
address 33: master #13, ebusd
address 38: slave #13

root@mon2:/var/log# ebusctl hex 15070400
00

root@mon2:/var/log# ebusctl info
version: ebusd 2.1.0c0ea8e
signal: acquired
symbol rate: 40
masters: 2
messages: 9
address 10: master #2
address 15: slave #2, scanned "MF=0;ID=" error: ERR: invalid position
address 33: master #13, ebusd
address 38: slave #13

```

Boiler + Room controller set on area 1:

```
root@mon2:/var/log# ebusctl scan
done

root@mon2:/var/log# ebusctl info
version: ebusd 2.1.0c0ea8e
signal: acquired
symbol rate: 56
masters: 3
messages: 10
address 01: master #6
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=   ;SW=" error: ERR: argument value out of valid range
address 33: master #13, ebusd
address 38: slave #13

root@mon2:/var/log# ebusctl hex 15070400
09000284081144060002

root@mon2:/var/log# ebusctl info
version: ebusd 2.1.0c0ea8e
signal: acquired
symbol rate: 17
masters: 3
messages: 10
address 01: master #6
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=   ;SW=" error: ERR: argument value out of valid range
address 33: master #13, ebusd
address 38: slave #13

```

Boiler + Room controller set on area 2 :
```
root@mon2:/var/log# ebusctl scan
done

root@mon2:/var/log# ebusctl info
version: ebusd 2.1.0c0ea8e
signal: acquired
symbol rate: 76
masters: 3
messages: 9
address 10: master #2
address 11: master #7
address 15: slave #2, scanned "MF=0;ID= ?  D;SW=7000;HW=0291"
address 33: master #13, ebusd
address 38: slave #13

root@mon2:/var/log# ebusctl hex 15070400
09000280081144700002

root@mon2:/var/log# ebusctl info
version: ebusd 2.1.0c0ea8e
signal: acquired
symbol rate: 32
masters: 3
messages: 10
address 10: master #2
address 11: master #7
address 15: slave #2, scanned "MF=0;ID= ?  D;SW=7000;HW=0291"
address 33: master #13, ebusd
address 38: slave #13
```

Boiler + Room controller set on area 3 :
```
root@mon2:/var/log# ebusctl scan
done                                                                                                                                      
                                                                                                                                          
root@mon2:/var/log# ebusctl info
version: ebusd 2.1.0c0ea8e
signal: acquired
symbol rate: 16
masters: 3
messages: 10
address 10: master #2
address 15: slave #2, scanned "MF=0;ID= ?  D;SW=0500;HW=0251"
address 31: master #8
address 33: master #13, ebusd
address 38: slave #13

root@mon2:/var/log# ebusctl hex 15070400
09000284081144050002

root@mon2:/var/log# ebusctl info
version: ebusd 2.1.0c0ea8e
signal: acquired
symbol rate: 17
masters: 3
messages: 10
address 10: master #2
address 15: slave #2, scanned "MF=0;ID= ?  D;SW=0500;HW=0251"
address 31: master #8
address 33: master #13, ebusd
address 38: slave #13

```

Boiler + mixer (NO room controller)

```
root@mon2:/var/log# ebusctl scan
done

root@mon2:/var/log# ebusctl info
version: ebusd 2.1.0c0ea8e
signal: acquired
symbol rate: 17
masters: 3
messages: 10
address 10: master #2
address 15: slave #2, scanned "MF=180;ID=?;SW=0000;HW=0000"
address 30: master #3
address 33: master #13, ebusd
address 38: slave #13

root@mon2:/var/log# ebusctl hex 15070400
0ab4830000020000000000

root@mon2:/var/log# ebusctl info
version: ebusd 2.1.0c0ea8e
signal: acquired
symbol rate: 16
masters: 3
messages: 10
address 10: master #2
address 15: slave #2, scanned "MF=180;ID=?;SW=0000;HW=0000"
address 30: master #3
address 33: master #13, ebusd
address 38: slave #13

and... after a few minutes...
root@mon2:/var/log# ebusctl info
version: ebusd 2.1.0c0ea8e
signal: acquired
symbol rate: 24
masters: 3
messages: 10
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=?;SW=0000;HW=0200"
address 30: master #3
address 33: master #13, ebusd
address 35: slave #3
address 38: slave #13

```

Conclusion : 
it doesn't seems that there is a dup address on bus.
Room controller address depends the Area it is set.
0x01 : room controller Area 1 (Not used in my house)
0x11 : room controller Area 2 ( Used in my house )
0x31 : room controller Area 3 (Not used)

tikismoke, 2016-10-16:
Sending ebusctl hex 1506210407b42040
Results in 0ab4830000020000000000

tikismoke, 2016-10-16:
```
ebusctl hex 1506210407b30000
0ab3830000010000000000
ebusctl hex 1506210407b50000
0ab5830008640000000200

ebusctl hex 1506210407b40000
0aff1f0000020000000000
ebusctl hex 1506210407b00000
0aff1f0000010000000000
ebusctl hex 1506210407010000
0aff1f0000010000000000
ebusctl hex 1507210407010000
0aff1f0000010000000000

```

tikismoke, 2016-10-16:
@john30 should we try with latest version?

You seems to have pushed 'tem' things.

tikismoke, 2016-10-16:
With latest 2.2 version i got this results:
```
pi@raspberrybplus:~ $ ebusctl info
version: ebusd 2.2.65328e5
signal: acquired
symbol rate: 17
masters: 2
messages: 11
address 10: master #2
address 15: slave #2, scanned "MF=TEM;ID=11111;SW=0100;HW=0100"
address 33: master #13, ebusd
address 38: slave #13

pi@raspberrybplus:~ $ ebusctl hex 15070400
067f3f00086400

```

tikismoke, 2016-10-16:
```
ebusctl find
broadcast datetime = 14.000; (ERR: argument value out of valid range for 10fe070009000effffffffffffff / )
broadcast error = no data stored
broadcast id = TEM;11111;0100;0100
broadcast netloss = no data stored
broadcast netresetcfg = no data stored
broadcast netresetstate = no data stored
broadcast signoflife = no data stored
broadcast id = no data stored
scan.15 ident = 127;?; (ERR: invalid position for 3315070400 / 067f3f00086400)

```
14°C is the current output temp.
It could also be the current pressure 1.4bar

tikismoke, 2016-10-16:
Playing a lot wiht hex command, now i got;
```
 ebusctl hex 15070400
0ab4830000020000000000
```

philb29, 2016-10-16:
```
philb@mon2:~$ ebusctl info
version: ebusd 2.2.af6e1c1
signal: acquired
symbol rate: 35
masters: 4
messages: 13
address 10: master #2
address 11: master #7
address 15: slave #2, scanned "MF=131;ID=?;SW=0200;HW=0200"
address 30: master #3
address 33: master #13, ebusd
address 35: slave #3, scanned
address 38: slave #13

```
```
philb@mon2:~$ ebusctl find
broadcast datetime = 18.098; (ERR: argument value out of valid range for 30fe0700091912ffffffffffffff / )
broadcast error = no data stored
broadcast id = TEM;11111;0100;0100
broadcast netloss = no data stored
broadcast netresetcfg = no data stored
broadcast netresetstate = no data stored
broadcast signoflife = no data stored
broadcast id = no data stored
scan.15 ident = 131;?;0200;0200
scan.16 ident = no data stored
scan.35 ident = no data stored

```
```
philb@mon2:~$ ebusctl hex 15070400
0ab5830008640000000200

```

john30, 2016-10-17:
@tikismoke please be careful with writing arbitrary hex messages to your devices. this might damage them!
the ident now looks good with ebusd 2.2 since it used the broadcast id as well.
anyway, CSVs are missing for your devices and if you don't have a controller for manipulating things, it is going to be very hard to find anything out about the communication...

john30, 2016-10-17:
@philb29 it seems that another device is interfering the communication, otherwise I can't explain the weird results you've posted. this could be the mainboard itself in case it is designed to not pay attention to any other master on one of the ebus connections. might be worth a try to test the other one.

tikismoke, 2016-10-17:
Yes i was just replaying philb29 frame to try.
philb29 may have a better chance to grab value for csv creation

Another question. With the rigth CSV file how does ebusd get value from boiler?
Does it poll it or i may keep the controller to intercept frame?

I've maybe found a controller at good price.

john30, 2016-10-18:
whether a message is polled, available for explicit reading, or only listened to by ebusd, entirely depends on the definition in the corresponding CSV.

tikismoke, 2016-10-20:
Ok now i have a working controller with lot of frame ;) I can progress.

Let you now later the full trace.

tikismoke, 2016-10-20:
I saw that the controler broadcast datetime but only the boiler broadcast external T°:

```
01fe070009008011092120100416 = broadcast datetime
01fe070009008015082120100416 = broadcast datetime
01fe070009008019072120100416 = broadcast datetime
10fe0700098009ffffffffffffff = broadcast datetime

```

tikismoke, 2016-10-20:
Somes commands
Activate SRA
2016-10-20 22:07:31.056 [update info] update MM cmd: 011006230609e0004000ef
Desactivate SRA
2016-10-20 22:07:41.665 [update info] update MM cmd: 011006230609e0004001ef
mode off
2016-10-20 22:09:35.553 [update info] update MM cmd: 01100623060487004004ef
mode winter
2016-10-20 22:10:30.539 [update info] update MM cmd: 01100623060487004003ef
mode summer
2016-10-20 22:11:03.424 [update info] update MM cmd: 01100623060487004002ef
mode heating only
2016-10-20 22:11:42.397 [update info] update MM cmd: 01100623060487004001ef
mode standby (frozen protection)
2016-10-20 22:12:23.980 [update info] update MM cmd: 01100623060487004000ef

tikismoke, 2016-10-21:
Select temp from SRA automatique contol 39° then 49 then 43:
```
ebusctl hex 1506210401870040
0a878001027f0080002700
0a878001027f0080003100
0a878001027f0080002b00
```
So this is corresponding to at least to heating water departure temp selected.
Seems circuit pressure ()current value 1.6bar (hexa:10)
```
ebusctl hex 1506210400970040
0a17800d18ff7f00801000
```
Seems hot water selected temperature ()current value 56°C (hexa:38)
```
ebusctl hex 1506210401840040
0a84000102410028003800
```
Seems month next maintenance ()current value 60 (hexa:3c) (confirm by changing it to 15 in the second result
```
ebusctl hex 1506210409de0040
0ade0400003c0000003c00
0ade0400003c0000000f00
```
Average boiler time in min(code 865 in menu) current value87  (hexa:57)
```
ebusctl hex 1506210402d90040
0a5981040cffff00005700
```
Hour boiler (code 860 in menu) current value 11 (hexa:b)
```
ebusctl hex 1506210402d60040
0a56810400ffff00000b00
```
Boiler cycle firering (code 863 in menu) current value 353  (hexa:161)
```
ebusctl hex 1506210402d00040
0a50810400ffff00006101
```

tikismoke, 2016-10-21:
Boiler temp:
current measured start heating (code menu 831) current value 32.5°C (hex:145)
```
ebusctl hex 1506210400870040
0a07800d02ff7f00804501
```
current measured back heating (code menu 832) current value 31.4°C (hex:13a)
```
ebusctl hex 1506210400880040
0a08800d02ff7f00803a01
```
current measured hot water (code menu 833) current value 30.1°C (hex:12d)
```
ebusctl hex 1506210400840040
0a04800d02ff7f00802d01
```
current measured exchanger (code menu 840) current value 56.3°C (hex:233)
```
ebusctl hex 15062104008f0040
0a0f800d02ff7f00803302
```
Fan speed (code menu 822) current value: 34x100rpm (hex:22)
```
ebusctl hex 1506210400a90040
0a29800400ffff00002200
```
Current hot water flow debt (code menu 825) current value 4.9L (hex:31)
```
ebusctl hex 1506210400c30040
0a43800000ff0000093100
```

3 way valve (0=hot water/1=heating)

tikismoke, 2016-10-21:
Boiler mode:
```
ebusctl hex 150621060487004000ef
0a070209001f0000000000
0a070209001f0000000100
0a070209001f0000000200
0a070209001f0000000300
```
0=Frozen protect
1=Heating only
2=Summer
3=Winter

tikismoke, 2016-10-21:
Curent amb temp (code menu 430) current value 26.5°C (hex:109) the thermostat is really near the boiler for my test.
```
ebusctl hex 1506210400810040
0a01800d02ff7f00800901
```
Curent target temp (code menu 431) current value 26.5°C (hex:e2)
```
ebusctl hex 1506210401810040
0a81800d02ff7f0080e200
```

tikismoke, 2016-10-21:
I've start working on a csv file:

> r2,boiler,exchange_temp,,,15,0621,04008f0040,,,,,,
> r2,boiler,target_temp,,,15,0621,0401810040,,,,,,
> r2,boiler,amb_temp,,,15,0621,0400810040,,,,,,
> r2,boiler,hot_water_flow_debit,,,15,0621,0400c30040,,,,,,
> r2,boiler,fan_speed,,,15,0621,0400a90040,,,,,,
> r2,boiler,hot_water_measure,,,15,0621,0400840040,,,,,,
> r2,boiler,hot_water_temp_selected,,,15,0621,0401840040,,,,,,
> r2,boiler,heating_start_temp,,,15,0621,0400870040,,,,,,
> r2,boiler,heating_return_temp,,,15,0621,0400880040,,,,,,
> 

```
pi@raspberrybplus:~ $ ebusctl find
broadcast datetime = 8.500; (ERR: argument value out of valid range for 10fe0700098008ffffffffffffff / )
broadcast error = no data stored
broadcast ident = no data stored
broadcast signoflife = no data stored
boiler amb_temp =  (empty for 33150621050400810040 / 0aff1f0d02ff7f00806901)
boiler exchange_temp =  (empty for 331506210504008f0040 / 0aff1f0d02ff7f00806901)
boiler fan_speed =  (empty for 33150621050400a90040 / 0aff1f0d02ff7f00806401)
boiler heating_return_temp =  (empty for 33150621050400880040 / 0aff1f0d02ff7f00806901)
boiler heating_start_temp = no data stored
boiler hot_water_flow_debit =  (empty for 33150621050400c30040 / 0aff1f0d02ff7f00806401)
boiler hot_water_measure =  (empty for 33150621050400840040 / 0aff1f0d02ff7f00806401)
boiler hot_water_temp_selected =  (empty for 33150621050401840040 / 0aff3f0d02ff7f00806901)
boiler target_temp =  (empty for 33150621050401810040 / 0aff1f0d02ff7f00806901)
broadcast ident = no data stored
memory eeprom = no data stored
memory ram = no data stored
```

All answers seems identical/or mixed even if:

```
ebusctl hex 150621050401840040
0aff1f0d02ff7f00804701
```

Like @philb29 it seems they're is a mistake with answer from somewhere even with starting with this commands:
`ebusd -f -d /dev/ttyUSB0 --loglevel=debug --latency=10000 --enablehex --address 33 --receivetimeout 10000`

philb29, 2016-10-21:
Corrected for heating_start_temp :
```
r2,boiler,heating_start_temp,,,15,0621,00870040,ign,,IGN:8,,,,heating_start_temp,,D2C,,°C,temperature,ign,,IGN:*
```

```
2016-10-21 22:13:06.006 [bus info] poll cmd: 011506210400870040
2016-10-21 22:13:06.188 [bus notice] poll boiler heating_start_temp: 22.56

```

```
philb@mon2:~$ ebusctl read -c boiler heating_start_temp
22.56
```

philb29, 2016-10-21:
SRA:
```
r,boiler,sra,,,15,0621,09e0004000ef,ign,,IGN:8,,,,sra,,UCH,0=off;1=on,,SRA,ign,,IGN:*
w,boiler,sra,,,10,0623,09e00040,sra,,UCH,0x00=off;0x01=on,,SRA
```

tikismoke, 2016-10-21:
Ok like this no more mixing error.

@philb29 @john30 
I think we can close the tickets as all seems ok and continue to ebusd-configuration topics to fill the csv
Work in progress

john30, 2016-10-22:
okay, ticket closed. follow discussion in [ebusd-configuration ticket](https://github.com/john30/ebusd-configuration/issues/27)

closed 2016-10-22 by john30

john30, 2016-10-23:
I've worked on the CSVs, optimized, renamed a bit, and added some tabs for easier finding of data types and units. see attachment