john30/ebusd #25: Windows library

Fabsky, 2016-10-22:
Hello,

do you think it would be possible to refactor your code as a library, and make it run under Windows ?
Because I'm trying to integrate my new vaillant (VC256) in my domotic system, which is currently running on a windows machine. I've dev background under windows so I could make an app once I've a dll I can use to communicate with the Vaillant, but my c++ knowledge are too much old to be any help....

Tx,
Fabrice

john30, 2016-10-23:
yes that should be possible, but I don't have time to do this

john30, 2017-02-11:
using Cygwin, you should be able to do that now. See [wiki](https://github.com/john30/ebusd/wiki/1.-Build-and-install#install-required-packages-on-cygwin)

closed 2017-02-11 by john30