john30/ebusd #26: e-services ethernet to e-bus adapter problem

sorosj, 2016-10-23:
I have bought an ethernet e-bus gateway from esevice-online.de, but I can not get it to work.
When I start ebusd, a TCP connection established immediatly, but no data traffic occurs.
When with ebusctl initiated "scan full" command, in log get "No signal" message and no data traffic in TCP session.

philb29, 2016-10-23:
What king of boiler do you have ?

How do you launch ebusd ?

I use [socat](http://www.dest-unreach.org/socat/) to get a virtual serial port. I launch socat like this :

`/usr/bin/socat -D pty,link=/dev/virtualcom0,raw tcp:192.168.100.12:5000
`

My ebusd is launched like this : 
````
/usr/bin/ebusd --pidfile /var/run/ebusd.pid -d /dev/virtualcom0 -n --loglevel=info --latency=10000 --enablehex --address 01 --receivetimeout 100000 
````

john30, 2016-10-24:
@philb29 You don't have to route it through socat, ebusd can handle TCP and/or UDP devices as well.
@sorosj Did you configure the coupler with the e-service config tool already? If so, which settings did you use?
And how did you configure ebusd?

sorosj, 2016-10-24:
No, I have not configured, because it does not find this device on LAN.
I discovered MAC address and a dhcp server gives address to it.

---------------------------

I tried philb29 advice, and received an error message and no data traveled in TCP session.

2016-10-24 09:44:23.112 [bus error] device /dev/virtualcom0 not available
2016-10-24 09:44:33.112 [bus notice] re-opened /dev/virtualcom0
2016-10-24 09:44:33.112 [main debug] performing regular tasks

While TCP connection exits.

tcp        0      0 10.0.2.15:34525         10.25.2.40:5000         ESTABLISHED 3998/socat

sorosj, 2016-10-24:
Boiler type:
Sanier Duval Semiatek Condens 24

john30, 2016-10-24:
I doubt that this will work when you can't reach the coupler with the configuration tool...

sorosj, 2016-10-24:
Sorry, but I started ebusd with some prameter missing.
With correct parameters I get a following error messages:

2016-10-24 10:01:17.179 [bus error] scan 02 failed (227 slaves left): ERR: no signal
2016-10-24 10:01:17.179 [bus info] scan 04 cmd: 0104070400
2016-10-24 10:01:17.179 [bus error] scan 04 failed (226 slaves left): ERR: no signal
2016-10-24 10:01:17.179 [bus info] scan 05 cmd: 0105070400
........ and so on .......

philb29, 2016-10-24:
@john30 : Thank you for the tip, it's working fine with 
```
-d tcp:192.168.100.12:5000
```

@sorosj : e-service config tool should find the device. Did you check the firewall of your windows host ? Give it a try with firewall disabled. Is your PC on the same LAN segment? If you use WiFi, try with a network cable.

sorosj, 2016-10-24:
Disturbing me the absence the traffic of data between host and device.
TCP connection established  correctly, but no byte sent over to device.

sorosj, 2016-10-24:
I have tried with directly with ip address/port device, I have got the exactly same result.

"No signal" and no traffic.

john30, 2016-10-24:
As said already: check the coupler device settings first, otherwise I can't help you

sorosj, 2016-10-24:
OK.

sorosj, 2016-10-24:
Now I have switched windows firewall off, and with wire connected my pc to LAN instead of wifi, but it did not help. Config tool does not find e-bus gateway.
I will try to contact manufacturer.

closed 2017-02-07 by sorosj