john30/ebusd #30: Support for JSON like DSL instead of CSVs

john30, 2016-11-17:


koen-lee, 2021-11-27:
I think this is a good idea that will make contributing easier. The CSVs have quite a learning curve and are error-prone.
Could you elaborate on the reasoning of removing it from the milestone?

john30, 2021-11-28:
It is still on the road map and actually I was browsing for good JSON libs last week for another thing. Anyway, this is a very time consuming thing to do and it will still take a while. In the meanwhile I'm working on a web UI that is supposed to allow clicking configs together and talk directly to ebusd to update the definitions online.

anticultist, 2024-05-29:
Maybe a intermediate json format could be a good starting point. So you can transform between the json format and the current csv based format back and forth but ebusd would only understand the "old" csv based format. With this approach you could test a new json based configuration without rewriting the existing code.

By the way do you know [json schema](https://json-schema.org/)? It is a powerful json-based meta-description of the structure of a json file. With that you can get error checking or auto-completion in modern code editors like VS Code while you work with a json file.

john30, 2024-06-08:
thats already there and of course I know json schema and there is a schema for what ebusd offers on its http port already in the [openapi spec](https://github.com/john30/ebusd/blob/master/contrib/html/openapi.yaml). the same schema is also produced by ebusd when using [`--dumpconfig=json`](https://github.com/john30/ebusd/wiki/2.-Run#message-configuration-options)

nevertheless I'm already working on that topic, recently by using [typespec](https://typespec.io/)

john30, 2024-07-13:
a lot of work has been put into this in the last weeks in [ebus-typespec](https://github.com/john30/ebus-typespec), [ebusd-configuration](https://github.com/john30/ebusd-configuration/tree/rework-normalize), as well as [ebusd](https://github.com/john30/ebusd/commit/b22a1135f5fa4bb2186ae4d8185b9f76fe521cec) itself and [ebus-notebook](https://github.com/john30/ebus-notebook).
the (intermediate) result of all that ends in https://ebus.github.io/ which is the new webservice to be used from the next version of ebusd.