john30/ebusd #43: add message broker support

john30, 2016-12-12:
support for publishing updates to a message broker via MQTT or AMQP would allow easy integration into lots of other systems including FHEM already supporting MQTT publish/subscribe

andig, 2016-12-29:
Could you add something simple like e.g. an integrated web server or plain web sockets, too? Using something like node-red its dead simple to add MQTT support but it needs some basic "getting the data out" capabilities. 

E.g. https://github.com/volkszaehler/vzlogger uses integrated microhttpd for that purpose.

john30, 2017-01-01:
a web server is already integrated, but still treated as experimental (see http commandline options). When enabled, the messages and fields are available as JSON using the /data/ url suffix.
could you point me to the documentation part of node-red for the data lookup feature you are referring to?

andig, 2017-01-02:
> a web server is already integrated, but still treated as experimental

Great, I'll take a look,.

> could you point me to the documentation part of node-red for the data lookup feature you are referring to?

I would generally refer you to http://nodered.org/. using node-reg you can easily "wire" any IOT application to anything else. It speaks http, websockets, mqtt and you name it out of the box. My idea was that- rather than implementing any specific logging technology- keep ebusd simple by implementing the most stupid of push mechanisms possible (e.g. web sockets) and leave everything else to the user side, e.g. by deploying node-red to convert "simple" to "custom" logging.

john30, 2017-01-02:
no, a HTTP connection is accepted from any port, not only localhost, so I guess you have some firewall issue or something else.
for node-red please check if it supports subsription to MQTT topics. If so, then node-red will be supported automatically once the MQTT support is finished.

andig, 2017-01-02:
> no, a HTTP connection is accepted from any port, not only localhost, so I guess you have some firewall issue or something else.

Right. Used port was blocked by client machine.

> for node-red please check if it supports subsription to MQTT topics

Should definitely be possible. Looking forward to the next version to replace logfile parsing with MQTT/node-red!

john30, 2017-01-07:
if you're only interested in pickung up the data from ebusd, then try the HTTP port and the URL `http://<ebusdhost>:<ebusdhttpport>/data/`

Personally, I use this with fhem and a JSONREADINGS device like this:
```
define jsonebus JSONREADINGS http://127.0.0.1:8080/data/ 60
attr jsonebus delay 60
```

john30, 2017-01-08:
solved with da58804f46c6ebbf858945cf4b1c8edd8a183cff and already documented in the wiki

closed 2017-01-08 by john30