john30/ebusd #44: Ebusd with bulex

koen-serneels, 2016-12-22:
Hi! I'm hacking with ebusd in an attempt to control my bulex heating from my house automation. At the moment I'm trying to see if I can change the requested room temperature. My ebus environment:

`address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0502;HW=3302"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=SBX00;SW=0322;HW=4802"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 70: master #4
address 75: slave #4, scanned "MF=Vaillant;ID=E7C00;SW=0211;HW=7402"
`
Master 70 is the room thermostat used to test: 'E7C' is the type (Bulex Exacontrol E7C). When the requested temperature is changed, the thermostat puts these 3 messages on the bus:

`
23°C
7015b509040e230001 / 00
7015b509050e24007001 / 00
7015b509070e250028014000 / 00
`
`
22°C
7015b509040e230001 / 00
7015b509050e24006001 / 00
7015b509070e250028014000 / 00
`
`
21°C
7015b509040e230001 / 00
7015b509050e24005001 / 00
7015b509070e250028014000 / 00
`

The magic number is '07' for 23°C, '06' for 22°C and '05' for 21°C. Now, I'm clueless on what these commands mean, but the first step would be to simply 'replay' commands to see that the examaster (#15) accepts the values send by ebusd. However, when the commands are written with ebusctl an error is returned:

`
localhost: write -h 15b509040e230001
ERR: element not found
localhost: write -h 15b509050e24007001
ERR: element not found
localhost: write -h 15b509070e250028014000
ERR: element not found
`

The ebusd log shows this:

`
write hex cmd: 3115b509040e230001
write hex cmd: 3115b509050e24007001
write hex cmd: 3115b509070e250028014000
`

As nothing is happening on the examaster I'm  unsure if the commands are properly written. Do you have an idea what this could be? Is it to naive to simply replay these commands or...?

Thanks!

koen-serneels, 2016-12-22:
some extra info: I loaded the vaillant CSV for the 'bai' device (renamed the HW version to match mine). It appaers to be parsing some information

> [update notice] update bai00 Mode QQ=10: standby
[update notice] update bai00 Status01 QQ=10: 34.5;31.0;-;39.5;57.5;on
[update notice] update broadcast vdatetime QQ=10: 22:52:00;22.12.2016
[update notice] update broadcast outsidetemp QQ=10: 6.562

Getting the value with ebusctl 

> localhost: read vdatetime
22:51:00;22.12.2016
localhost: read status01
34.5;31.0;-;39.5;57.5;on
localhost: read outsidetemp
6.562

However, this value is read from cache. If I try to force read (before the value appeared on the bus

koen-serneels, 2016-12-22:
Ok, learned that the "write hex" is apparently coupled to commands loaded from csv. As I did not have any csv loaded it errors out. There is also a hex command that puts the value directly on the bus which works as intended. Closing.

closed 2016-12-22 by koen-serneels

john30, 2016-12-23:
this is a regular Vaillant register write message here:
7015b509050e24007001 / 00
70= source master
15= destination slave
b509= Vaillant register command
05=length
0e=Vaillant register write
2400=register number (0x0024=36)
7001=desired value (0x0170=368)

The two bytes value is in datatype D2C (see here: [data types](https://github.com/john30/ebusd/wiki/4.3.-Builtin-data-types) ) and has to be divided by 16 for the real value: 368/16=23 °C

koen-serneels, 2016-12-23:
Thanks @john30