john30/ebusd #45: OSX

andig, 2016-12-25:
Would be nice if at least `ebusctl` could run on OSX. To compile you'd need:

    brew install automake autoconf argp-standalone 

Might be nice to add this to the readme?

john30, 2016-12-27:
added to the [wiki](https://github.com/john30/ebusd/wiki/1.-Build-and-install/dd44ac238f15f60659a20b12568f44accbc501ac)

closed 2016-12-27 by john30

pagaille, 2020-08-16:
Additionally : 

`device.cpp:322:35: error: use of undeclared identifier 'TCP_KEEPIDLE' `

I substituted  TCP_KEEPIDLE by TCP_KEEPALIVE. It worked. I guess it should be possible to add some conditional compile statements.

Ref :  https://stackoverflow.com/questions/15860127/how-to-configure-tcp-keepalive-under-mac-os-x