john30/ebusd #46: http interface problems

filnko, 2016-12-25:
Hello,

I'd like to use the http interface.

What i did:
- Activated the http port:
  - `EBUSD_OPTS="--scanconfig --httpport=80"`
- Copied the contents from `contrib/html` to `/var/ebusd/html`

Sadly nothing works and the request stays open endlessly.

Did I miss something? How does it work?

Thanks, Christoph

john30, 2016-12-27:
what is your scan result?
and did you ensure that no other daemon is occupying port 80 (such as apache/nginx)?

filnko, 2016-12-27:
my scan result:

  `08;Vaillant;BAI00;0706;7401;21;10;37;0010004111;0006;005312;N2`
  `15;Vaillant;F4300;0120;2102;21;10;27;0020028521;0907;005074;N5`

ebusd itself is working on the command line without any problems - it's just the http part.
I also can't really find any documentation on that part.

no other deamons are running, I tried different ports.

john30, 2017-01-01:
the HTTP port ist still experimental and will probably change a lot in future. That's why it is documented very litte as of today. It also only supports EHP/UIH/MC2 devices today.
For your configuration, everything is missing and so the displayed window should be empty.

andig, 2017-01-02:
> Copied the contents from contrib/html to /var/ebusd/html

Missing this step from the wiki. Otherwise checking in here for updates ;)

john30, 2017-01-08:
duplicate of #33 
documentation started in wiki: https://github.com/john30/ebusd/wiki/3.2.-HTTP-client

closed 2017-01-08 by john30