john30/ebusd #47: bus error CfgHeatSinkType failed

dhaeype, 2016-12-28:
Dear,

I have 1 heatpump (Vaillant geotherm VWS 171/2), 2 floorheating circuits (mc2 and mc4), 1 radiator circuit (mc5), 1 Waterheating, 1 VR60 and 2 VR90 which I try to manage via a Raspberry Pi.

I recieve some errors and notifications in my ebusd.log (e.g. "CfgHeatSinkType failed" or "unknown MS cmd: 7050b5040132 / 0a002c06030301a0020100") and have a lot of "no stored data" in the output of the "ebsctl find" command. I attached 2 files: 1 with the output of several commands and 1 showing a small part the raw log.

I would be gratefull if someone could indicate where these errors come from and what to do to solve it.

Thanks in advance and best regards,

Peter

[Raw log.txt](https://github.com/john30/ebusd/files/675433/Raw.log.txt)
[scan.txt](https://github.com/john30/ebusd/files/675434/scan.txt)

john30, 2017-01-01:
it seems that writing on the bus is not as stable as it should be. furthermore, the UIH unit does not appear in the scan, so you should try to get the signal more stable and/or check your interface.
the unknown messages are a known issue (see https://github.com/john30/ebusd-configuration/issues/3)

dhaeype, 2017-01-02:
Hi John, 

Thanks for your answer. 
If I understand well there is a problem with my ebus adapter? For your info, I made one myself using the components and circuit board as described on the fhem website. I connected my interface directly on the Heatpump with the same result. I also turn the potentiometer but I don't really know which is the best position. 

Do you have any ideas what I could do to find out where the problem is or solve it? 

Thanks in advance and best regards, 

Peter 

the unknown messages are a known issue (see [ https://github.com/john30/ebusd-configuration/issues/3 | john30/ebusd-configuration#3 ] )

john30, 2017-01-07:
adjusting the poti is a little bit tricky, I know. I've just added an extra paragraph for that in the [wiki](https://github.com/john30/ebusd/wiki/6.-Hardware)

closed 2017-02-11 by john30