john30/ebusd #49: High memory consumption and eventual crash of the ebusd process on raspberry pi

m3phisto, 2017-01-03:
I have ebusd running on a raspberry pi b+ with raspbian. Every 5 minutes I read about 200 ebusd values via telnet. This worked flawlessly with version 2.3. Since 2.4 (I'm referring to the released *_armhf.deb packages) I noticed an odd behaviour:
- After about 1 day the virtual memory demand of ebusd is higher than 100MB which was not the case with 2.3
- Eventually the process becomes unresponsive and the communication socket just hangs.
- I am forced to kill ebusd with SIGKILL. Subsequently restarting ebusd just yields "signal: no signal" and it does not see any devices on the bus. I need to reboot to be able to use ebusd again.

I'd like to help by bisecting the source and finding out which commit introduced this regression for me. But I found that e.g. v2.4 (I'm now referring to the tag in this repository) doesn't run but just segfaults. So I can't test the changed that several commits made.

If I should provide any more information, let me know.

Thanks very much for this awesome piece of software!

john30, 2017-01-06:
I'm not able to reproduce this behaviour. Please give me all the details you have like platform, OS, exact ebusd version (see log file), the interface used (IP/USB), the CSVs, the scan result, a full log of one day, the number of clients connecting to ebusd via cmdline port and the number of clients connecting to the optional HTTP port.
You could also start ebusd using gdb and try to reproduce the unresponsiveness. Then gdb should give more details about what's going on.

m3phisto, 2017-01-06:
I'm running on a Raspbian Jessie (8.0) on
`Linux heat1 4.4.38+ #938 Thu Dec 15 15:17:54 GMT 2016 armv6l GNU/Linux`
By ebusd_configuration is ebusd-configuration-2.1.5f0ced9-de_all with one additional change:
vaillant/ed.vms.csv is a symlink to vaillant/06.pms.csv

Here is the info on the ebusd daemon itself:
```
root@heat1:~ # ebusctl info
version: ebusd 2.4.5eacc43
signal: acquired
symbol rate: 22
reconnects: 0
masters: 4
messages: 677
conditional: 2
poll: 1
update: 8
address 00: master #1
address 03: master #11
address 05: slave #1, scanned "MF=Vaillant;ID=VD300;SW=0213;HW=7501", loaded "vaillant/05.vd3.csv"
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0703;HW=7401", loaded "bai.0010007508.inc" ([PROD='0010007508']), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=47000;SW=0128;HW=6002", loaded "vaillant/15.470.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address ed: slave, scanned "MF=Vaillant;ID=VMS00;SW=0107;HW=4302", loaded "vaillant/ed.vms.csv"
```
I have only one client connecting via commandline port at a time (every 5 minutes). I don't use the HTTP port at all.

I'll be on holiday the next week. I'll provide the logs etc. as soon as possible, when I get back.
Thanks for your effort.

john30, 2017-01-07:
okay, looks quite normal to me.

> But I found that e.g. v2.4 (I'm now referring to the tag in this repository) doesn't run but just segfaults.
There was a bug introduced with commit cdbcb46 and solved with 8dce270, so you could just treat everything between as a single commit. I appreciate your findings!!

m3phisto, 2017-02-02:
It was really difficult to reproduce this issue reliably. With the current v3.0pre it seems to be resolved for me. Feel free to close.

Best, Georg

john30, 2017-02-03:
too bad

closed 2017-02-03 by john30