john30/ebusd #50: Switching bus address dynamically

koen-serneels, 2017-01-05:
Hi. Is there a possibility to change the bus address that ebusd uses on the bus at runtime? The ebus controller I'm operating uses the base address of the sender (in this case ebusd) to identify the heating circuit. So, at the moment I'm able to operate a single heating circuit by using the "-a" option. To operate the other heating circuit I have to quit ebusd and start it again using the other base address. This is a bit cumbersome :)

Would it be possible for example to add an (optional) parameter to the 'hex' command to select the base address to use? Like "hex -a<address> 0123" or to prefix the command with the base address "hex <address>0123"...

john30, 2017-01-07:
in principle this is possible, but this might lead to breaking the eBUS protocol and disturb communication of circuits with that address. So I'd rather enable this only if the master address was not yet seen on the bus.

koen-serneels, 2017-01-07:
Thanks John. Can you give an ETA on this feature? I mean is it complex and will it take months or, ...

john30, 2017-01-07:
It's not that complex, but my time is very limited and other features are currently in work that have to be finished first. So, a month or so is realistic.

koen-serneels, 2017-02-18:
Hi John. Did you have a chance to look at this already?

john30, 2017-02-19:
solved with ce85ce75d770d5d0866e080fa8fd4275cc0c9991, enabled globally (independant of seen addresses), so use with care!

closed 2017-02-19 by john30