john30/ebusd #52: Refreshing data for mqtt or httpd json interface- use of polling?

andig, 2017-01-15:
I want to log some ~30 values of my bai- either each minute or when it changes. I've currently solved this by

  - using a cron job
  - and listening to logfile changes especially for the `Status01` notifications

My code is in https://github.com/andig/vaillant.

I'm wondering what other/ better options there are. Using `mqtt` or `/data/` I can get all currently received values from ebusd. Using `mqtt` (but not httpd) I can force updates using `/get` suffix.

Would the polling feature be suitable for my purpose/ could you explain how it works? I'm wondering if I could setup polling once and only use mqtt or json client to look for the updates?

andig, 2017-01-15:
**Update**

I've tried refreshing data/ reading from mqtt but am not successful. The `ebusd/global` topics are working but e.g. `ebusd/bai/Status01` isn't- is this the right notation?

john30, 2017-01-17:
I can't verify your problem with the MQTT interface. Everything works just fine. Which version of mosquitto are you using? When I do a `mosquitto_pub -n -t ebusd/$circuit/$name/get` a couple of seconds later a `mosquitto_sub -v -t "ebusd/#"` receives the result.

For the regular polling of data: why don't you use the contrib/scripts/readall.sh in the cronjob for that?
And why looking into the log file at all if ebusd supports multiple regular ways to retrieve the values?

andig, 2017-01-17:
> I can't verify your problem with the MQTT interface. Everything works just fine. 

Confirmed. I had 2 ebus instances running, reason unknown (sorry- I thought I had closed this issue to clean it up).

> For the regular polling of data: why don't you use the contrib/scripts/readall.sh in the cronjob for that?

Haha ;) Never looked there. `readall` seems to run through every message found and read it? 

> And why looking into the log file at all if ebusd supports multiple regular ways to retrieve the values?

I wanted to get a good and detailed look at the bai operation- subminute resolution and was looking for the bai to *push* messages instead me *polling* it. Looking at the logfile I got that `Status01` whenever the bai sent it (though not much else). I was afraid to have my polling overload the bus and impact bai/vrc operations if I requested too much data.

I'm still wondering what the polling (with priority levels) actually does?

john30, 2017-01-18:
yes, readall.sh reads all elements matching the parameters give, so e.g. with `./readall.sh -c bai temp` it would read all temperature messages of the bai circuit.
the polling does what it says, it polls the messages marked as such in regular intervals (see --pollinterval argument to ebusd). you can also use the "read" command to adjust poll priorities during runtime of ebusd. a higher a poll priority value will lead to less often poll of that message, i.e. prio 1 messages are polled in every cycle, prio 2 messages are polled every second cycle and so on.

closed 2017-02-11 by john30