john30/ebusd #56: Question: how to compile with MQTT support

schka17, 2017-01-30:
Hi John,

can you give me hint to compile with MQTT Support?

Thanks

john30, 2017-01-30:
you need to have libmosquitto-dev available on the build environment

schka17, 2017-01-31:
Hi Johann hn,

Thanks works fine.

I have just an issue assigning the topic:

I have a lot of devices/systems publishing into my MQTT server, all other publishing with a starting / like /owado/devices/esp-sensor-64/#
but i cannot start ebusd with the option --mqtttopic=/owado/ebusd...... i get the error invalid mqtttopic
Without the lading / it works, but my problem ist that i want to subscribe to /owado/# to see all of this topics.

Without analysing the spurce, is there a chance to change that?

Thanks and regards

john30, 2017-01-31:
just remove the slash. the subscription is of course with a starting slash by design.
or maybe I get you wrong here.
With --mqtttopic=blah/... you should be able to subscribe to /blah/#

schka17, 2017-01-31:
Sorry, no that doesnt work in that way,

here is a small snippet of mqtt messages:

`owado/ebusd/bai/StorageTempDesired 55.00
owado/ebusd/bai/ReturnTemp 65.94;64480;ok
owado/ebusd/bai/ModulationTempDesired 16.0
owado/ebusd/bai/PrEnergyCountHc1 266571
owado/ebusd/bai/FlowTempDesired 65.50
owado/ebusd/bai/PrEnergyCountHc2 0
owado/ebusd/scan.52/ Vaillant;VR_70;0109;2903
owado/ebusd/broadcast/vdatetime 16:32:07;31.01.2017
owado/ebusd/broadcast/outsidetemp -1.875
ebusd/global/version ebusd 3.0pre.15d5500
ebusd/global/running false
ebusd/global/signal true
ebusd/global/uptime 48
ebusd/scan.08/ Vaillant;BAI00;0116;9602
ebusd/scan.08/id 21;16;31;0010015600;3100;006988;N1
ebusd/scan.15/ Vaillant;70000;0419;4603
ebusd/scan.52/ Vaillant;VR_70;0109;2903
ebusd/bai/Mode standby
ebusd/bai/Status01 69.5;65.5;-1.875;47.0;48.0;on
ebusd/bai/Status02 auto;60;75.0;70;65.0
/owado/devices/esp-sensor-68/Heizung_FBDGKZ_VL 28.6
/Outdoor_Weather/rain 2479.9
/Outdoor_Weather/wind 1.1
/Outdoor_Weather/humidity 70
/Outdoor_Weather/temperature -3.7
/Outdoor_Weather/israining no
/Outdoor_Weather/rain_calc_tsecs 1485877534.3002
/Outdoor_Weather/rain_calc_now_value 2479.9
/Outdoor_Weather/rain_calc_now_diff 0
/Outdoor_Weather/rain_calc_all cH: 0.0 lH: 0.0 cD: 0.0 lD: 0.0 IR: no Rnow: 0.0 Rdif: 0
/owado/HAL9000/TF_HelgasZimmer/temperature 22.4
/owado/devices/esp-sensor-68/Heizung_FBDGKZ_RL 28.6
`

so if a subscribe to owado/# i just can see the ebusd messages but the other messages of /owado/# and viceversa

To get all owado messages i need 2 sbscriptions, with and without leading slash.

and the scond thing, the paramater --mqtttopic seems not not to apply to ebusd/bai, ebusd/global, and ebusd/scan.

Regards,, Karl

john30, 2017-02-01:
from my understanding, having a leading slash would yield in an empty top level in the topic and that seems to be wrong.

schka17, 2017-02-02:
Hi John,

< from my understanding, having a leading slash would yield in an empty top level in the topic and that seems to be wrong.>

I dont know if its wrong, all my other stuff is using/supporting a starting slash, and so i structured that in this way ie.

/owado/HAL9000/TF_HelgasZimmer/temperature 22.4
/owado/devices/esp-sensor-68/Heizung_FBDGKZ_RL 28.6

so its easy (just for diagnosis) to subscribe to /owado/# or any other topic root and get all the relevant messages. In production its not an issue, because the i subscribed to a dedicated topic, just for diagnosis, i can live with that.

<what do you mean by that exactly? examples?>
what i mean is, when i configure the topic i.e. --mqtttopic=ebusd/%circuit/%name
then this topic is applied to aaahrrgggg -> i see, its just applied circuit and name but not to "global" , "broadcast","scanxx" and so on. ie.

ebusd/global/signal true
ebusd/scan.15/ Vaillant;70000;0419;4603

OK my fault.

andig, 2017-02-02:
It is possible to use / but not recommendet, see e.g. HiveMQ docs. 

Viele Grüße,
Andreas

john30, 2017-02-03:
for the time being I won't change the code. you should rather think about correcting your other devices topics

schka17, 2017-02-03:
Hi John,

no problem, i can do it myself, thanks.

changing the other stuff is for me an impracticable option, because there are not only a lot of sensors and systems but also event based systems involved,  depending on that structure, and this is a historical grown over the last three years (from a proof of concept into production :-) ) i will not change a running system, it works, just for diagnosis it would be fine. But that's not essential

Thanks and Regards, Karl

closed 2017-02-03 by schka17

NemoN, 2017-03-17:
Small patch, that allows a leading / in mqtttopic (I using the / in front of my mqtt topics too)

```
--- a/src/ebusd/mqtthandler.cpp
+++ b/src/ebusd/mqtthandler.cpp
@@ -105,7 +105,8 @@ static error_t mqtt_parse_opt(int key, char *arg, struct argp_state *state) {
     break;

   case 5:  // --mqtttopic=ebusd
-    if (arg == NULL || arg[0] == 0 || arg[0] == '/' || arg[strlen(arg)-1] == '/') {
+    // if (arg == NULL || arg[0] == 0 || arg[0] == '/' || arg[strlen(arg)-1] == '/') {
+    if (arg == NULL || arg[0] == 0 || arg[strlen(arg)-1] == '/') {
       argp_error(state, "invalid mqtttopic");
       return EINVAL;
     }
````

john30, 2017-03-19:
solved with f7f79bb