john30/ebusd #57: Q: In which version can I find MQQT

absalom-muc, 2017-01-30:
Since I'm very curious to use the MQTT implementation I installed [lastbeforec++11](https://github.com/john30/ebusd/releases/tag/lastbeforec%2B%2B11). 
But there seems to be no MQTT support in this version:
```
ebusd -V
ebusd 3.0pre.p20170130

sudo ebusd --scanconfig --mqttport=1883
ebusd: unrecognized option '--mqttport=1883'
Try `ebusd --help' or `ebusd --usage' for more information.
```
But ebusd --help shows me no MQTT option. Could you lead me to the version including MQTT?

Thank you.
absalom

andig, 2017-01-31:
See https://github.com/john30/ebusd/issues/56

absalom-muc, 2017-01-31:
I've seen it but how is it related to my question? I had no problem to build it and ebusd 3.0pre.p20170130 doesn't recognize the MQTT options.
btw: libmosquitto-dev is installed

john30, 2017-01-31:
please attach the output of ./autogen.sh

absalom-muc, 2017-01-31:
sorry, libmosquitto-dev was **not** installed. Now it is and I'm able to subsribe to the ebusd messages.

Thank you!

closed 2017-02-01 by john30