john30/ebusd #58: fyi: using ebusd with mqtt and node-red to log to volkszaehler

andig, 2017-02-05:
Hi everyone,

I've pieced together a small node-red flow that:

  - pulls ebusd via mqtt
  - converts the results (especially Status01 as well as pump and vent state into numbers)
  -  pushes the results- if updated- to volkszaehler

Any interest please let me know to add some details here.

NemoN, 2017-03-17:
> Any interest please let me know to add some details here.

please provide details

andig, 2017-03-17:
> please provide details

My setup consists of:

  - A flow that builds a map of all channels/uuids. Channel name equals ebusd register name which makes it simple to assign automatically.
  - A polling flow.
  - A receiving flow with a but of magic for decoding things like `Status01` und doing value change detection before writing changed values back to VZ

andig, 2017-03-17:
[image]
[image]
[image]

andig, 2017-03-17:
I can share the flows if you want

NemoN, 2017-03-26:
please share the flows :-)

andig, 2017-03-27:
> please share the flows

Any questions/improvements please let me know. Error handling is rudimentary since I obviously know that the required channels exist ;)

https://gist.github.com/andig/9b1d715ba1fcb60b6fb646bf69dfb7f3

NemoN, 2017-04-03:
@andig do you have a screenshot from the final charts in VZ?

andig, 2017-04-21:
@NemoN Etwa so in der Monatsansicht, nur die interessantesten Kanäle ausgewählt:

[image]

john30, 2017-07-02:
this can be achieved easier now by using the new ebusd argument --mqttjson since 259f9b76ef44b4d5873443e24d911a604fe271f4

andig, 2017-07-02:
@john30 couldn't find it in the wiki and don't want to break my prod setup. Could you add json example to the docs?

john30, 2017-07-02:
its just using "--mqttjson" additionally and then you'll get the data in json format just as by retrieving from HTTP port /data

andig, 2017-07-02:
Sure, but what do you mean by "json format"? I'm getting a single value per channel only, what advantage does json have here?

john30, 2017-07-06:
then leave it as it is :-)