john30/ebusd #60: ReadRequest only sends 1 Byte

AndyNeubacher, 2017-02-10:
Hi All,

i've running the ebusd on an FriendlyARM NanoPi NEO [http://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO](url) now for some days. The Rx/Tx lines from the eBusConverter are directly connected to the ttyS1 of the board ... so without the FTDI. (IMHO seems to work fine)

My heat-pump has an TEM-Controller inside (WPC6024) and therefore i had to investigate/sniff to get all the unknow params and values i'm interested in. Now it seems that i can listen and decode all the messages on the bus, but when i try to read from another eBus-Slave with a PollRequest (type "`r1`") i always get a `[bus error] poll  WP_Leistung failed: ERR: read timeout` 

First i thought i do have a hardware-issue, but i can see with the Oscilloscope the the data sent on the eBus are exactly the same coming out of the processor (NanoPi). The weird thing is that my ReadRequest should be at least `QQ ZZ PBSB NN IDIDIDID CRC`  about 10Bytes long. 
I can see on the Scope that the ReadRequest is only about 1Byte long? ... 

By sending the request via the ebusctl i also get only 1Byte on the bus/uart.  IMHO my csv-file looks similar to the vaillant ones ... so can't find any reason for missing bytes?! 

does anybody has ever seen such a behavior, any hints etc. ?

br and thx for your support,
Andy

btw: GREEAAT project :1st_place_medal:  - will definitely donate some beers :)

john30, 2017-02-10:
the onboard serial driver of Pi's have a buffering enabled on a rather low level which the OS is not able to control. At least that's the reason for the onboard serial interface of Pi B not being usable for ebusd, as this breaks the eBUS specification due to running into timeouts and stuff. You could give a try to the [--latency=MICROS](https://github.com/john30/ebusd/wiki/2.-Run#device-options) option though.

> btw: GREEAAT project 🥇 - will definitely donate some beers :)

see here: [donate button in wiki](https://github.com/john30/ebusd/wiki) :-)

s1nn3r, 2017-02-11:
Hey Andy,
as I got the same controller, would you be so kind and upload your csv files here?
Many thanks!
Marc

AndyNeubacher, 2017-02-11:
the latency hint was perfect ... now all the cmd's are working! 

thx and PROST john :)

AndyNeubacher, 2017-02-11:
@s1nn3r only in german but i added the numeric menu-code in the comment
[tem.txt](https://github.com/john30/ebusd/files/768854/tem.txt) (rename to .csv and copy to /home/ebusd/)

closed 2017-04-08 by john30