john30/ebusd #64: Incorrect roomtemp and questions about errors and ebusctl -h commands

Mario-H, 2017-02-21:
Hi,

I've been struggling with my heater combination (Vaillant hrSolide VHR NL 30 - 34/3 C in combination with a calorMATIC 360 room thermostat) for a week now and finally have most things running.
(It took me a lot of Google searches to find that the GPIO UART on the PI shouldn't be used)
I use the readall script to read all the data and store it in a MySQL database.

One of the things that keeps bugging me is that the room temperature on the display in the living room is not the same value as reported bij de output of readall
for example:
when the display shows 16.5 the collected data says the displayed temp is 17.75
when the display shows 17.5 the collected data says the displayed temp is 18.69
According to the collected data the offset = 0 (360 RoomTempOffset = 0.00)

In addition I get several errors that I would like to fix (if possible).
I listed some of the errors below.

To figure out what hex-data is actually received by ebusd I have been trying to send the commands myself but I guess I don't understand how to do this correctly because all variations on commands I have tried so far gave me an error message form ebusctl (ERR: element not found or ebusctl syntax info)
Can someone show my what ebustctl write -h xxxxxxx I should send to read "Roomtemp" and let's say "bai status" and "360 SavingsFunctionTime" ?

Finally I get some errors at the bottom of the readall output (broadcast, memory and scan). I assume the broadcast messages should be available so what could be the cause of this?
As for the scan result errors: It looks like the readall script doesn't send the correct data to ebusdctl read resulting in those errors.
Is this something that can be fixed?

Thanks in advance,

Mario

----- System info and logs snippets ----------

pi@raspberrypi:~ $ ebusctl info
version: ebusd 2.4.79708d2
signal: acquired
symbol rate: 22
reconnects: 0
masters: 4
messages: 428
conditional: 4
poll: 0
update: 8
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "bai.0010006101.inc" ([PROD='']), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=36000;SW=0139;HW=7301", loaded "vaillant/15.360.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 7f: master #24
address 84: slave #24, scanned "MF=Vaillant;ID=OTS00;SW=0205;HW=9101"

-----------------------------------------

pi@raspberrypi:~ $ ebusctl scan result
08;Vaillant;BAI00;0518;7401
15;Vaillant;36000;0139;7301
84;Vaillant;OTS00;0205;9101 (what is this device?)

-----------------------------------------

These are some of the errors I get in de readall output:
360 SavingsFunctionTime = ERR: argument value out of valid range in decode
bai exhaustWayBlockCounter = ERR: invalid position in decode
bai Status = ERR: invalid position in decode
broadcast datetime = ERR: no data stored
broadcast error = ERR: no data stored
broadcast hwcStatus = ERR: no data stored
broadcast load = ERR: no data stored
broadcast signoflife = ERR: no data stored
memory eeprom = ERR: invalid address
memory ram = ERR: invalid address
scan.08 = usage: read [-f] [-m SECONDS] [-c CIRCUIT] [-d ZZ] [-p PRIO] [-v|-V] [-n] [-i VALUE[;VALUE]*] NAME [FIELD[.N]]
scan.08 id = ;;;;;;
scan.15 = usage: read [-f] [-m SECONDS] [-c CIRCUIT] [-d ZZ] [-p PRIO] [-v|-V] [-n] [-i VALUE[;VALUE]*] NAME [FIELD[.N]]
scan.84 = usage: read [-f] [-m SECONDS] [-c CIRCUIT] [-d ZZ] [-p PRIO] [-v|-V] [-n] [-i VALUE[;VALUE]*] NAME [FIELD[.N]]
scan id = ERR: invalid address

john30, 2017-02-22:
okay, added a note to https://github.com/john30/ebusd/wiki/6.-Hardware
the room controllers usually have a register telling the self-warming temperature difference that has to be substracted from the temperature register value.
for the hex messages you can simply use "ebusctl grab result all".
the readall script is relatively stupid as it does not care e.g. whether sending a message needs additional input (such as errorhistory). the scan messages are removed with 7d5dd27.
the decode issues are due to the CSVs not matching exactly your devices. a full scan should reveal the product number of your BAI and I guess this is not part of the list in 08.bai.csv.
you can easily remove the "no data stored" results by using e.g.:
`./readall.sh|egrep -v "ERR: no data stored"`

Mario-H, 2017-02-22:
Thanks for your answers.

Hope the note in hardware wiki helps others.
RoomTempOffsetSelfWarming solved the roomtemp issue.
readall script mod fixed the errors
adding egrep solved the "ERR: no data stored" issue

A full scan doesn't give me any additional info about the BAI product number...
pi@raspberrypi:/etc/ebusd $ ebusctl scan result
08;Vaillant;BAI00;0518;7401;;;;;;;
15;Vaillant;36000;0139;7301;21;08;35;0020045552;0907;006478;N6
84;Vaillant;OTS00;0205;9101
How do I instruct ebusd to use a specific (set of) config file(s) even when no product number is available?

That leaves me with the hex messages:
"ebusctl grab result all" gives me:
7f15070400 / 0ab5333630303001397301 = 35: scan.15
10feb516080017151022020317 = 3: broadcast vdatetime
1008b5040100 / 0a00062815ffffffff0080 = 6: bai DateTime
1008b5100900001478ffff41ff00 / 0101 = 16: bai Mode
1008b5110101 / 092a28008034440000ff = 18: bai Status01
1008b5110102 / 05033c96467c = 7: bai Status02

But the errors in the readall output are not all listed in the scan result.
Errors listed by readall:
360 SavingsFunctionTime = ERR: argument value out of valid range
bai exhaustWayBlockCounter = ERR: invalid position
bai Status = ERR: invalid position
bai Status16 = ERR: invalid position

When I try to read bai Status01 with: "ebusctl write -h 08b5110101" (copied from grab result above but without the master)
I get: "ERR: element not found"
But in the readall output I see: "bai Status01 = 38.0;37.0;-;29.0;36.0;off"
So it must exist.
What am I doing wrong?

john30, 2017-02-24:
> A full scan doesn't give me any additional info about the BAI product number...

interesting. you bai seems to be very special :-)

> How do I instruct ebusd to use a specific (set of) config file(s) even when no product number is available?

In this case you can only use the hardware version fallback (see end of 08.bai.csv) and maybe include thw software version as well.

> But the errors in the readall output are not all listed in the scan result.

please post the complete grab result in hex

> When I try to read bai Status01 with: "ebusctl write -h 08b5110101" (copied from grab result above but without the master) I get: "ERR: element not found"

That's interesting. Need to investigate first.

Mario-H, 2017-02-24:
> please post the complete grab result in hex

Not sure how to do that. I don't see a -h option listed in de ebusctl help for grab.
What command(s) should I use to provide you with the wanted info?

john30, 2017-02-25:
> When I try to read bai Status01 with: "ebusctl write -h 08b5110101" (copied from grab result above but without the master) I get: "ERR: element not found"

This is due to the message bai Status01 being defined as "read" message. Consequently, you need to issue "ebusctl read -h 08b5110101" in order to use the hex retrieval.

Mario-H, 2017-02-25:
> you need to issue "ebusctl read -h 08b5110101" in order to use the hex retrieval.

That worked!

pi@raspberrypi:/etc/ebusd $ ebusctl read -h 08b5110101
093a3a00803c420000ff51

> the output of grab result (without the decode option) is in hex

Isn't that what I posted above? Anyway, I ran a grab and below are the results.
grab resultl:
pi@raspberrypi:/etc/ebusd $ ebusctl grab result
1008b512020064 / 00 = 5
1008b513020507 / 00 = 5

grab result all:
pi@raspberrypi:/etc/ebusd $ ebusctl grab result all
7f15070400 / 0ab5333630303001397301 = 501: scan.15
10feb516080001242225020617 = 46: broadcast vdatetime
1008b5040100 / 0a00470706ffffffff0080 = 93: bai DateTime
1008b5100900001478ffff01ff00 / 0101 = 272: bai Mode
1008b5110101 / 094a4a00805e640000ff = 282: bai Status01
1008b5110102 / 05033c96467c = 101: bai Status02
1008b512020064 / 00 = 5
1008b513020507 / 00 = 5
7f08b503020001 / 0affffffffffffffffffff = 2: bai currenterror

Errors from readall:
360 SavingsFunctionTime = ERR: argument value out of valid range
bai BoilerType2 = ERR: invalid position                         
bai DCFTimeDate = ERR: invalid position                         
bai errorhistory = ERR: end of input reached                    
bai exhaustWayBlockCounter = ERR: invalid position              
bai Status = ERR: invalid position                              
bai Status16 = ERR: invalid position                            
broadcast ident = ERR: invalid address                          
memory eeprom = ERR: invalid address                            
memory ram = ERR: invalid address                               

Decode option is not supported in the version I'm running (2.4.79708d2)

john30, 2017-02-26:
is this really the complete grab result? seems that some messages are missing

Mario-H, 2017-03-01:
At the time it was.

Grab has been running since then and now looks like this:

pi@raspberrypi:/etc/ebusd $ ebusctl grab result all
7f15070400 / 0ab5333630303001397301 = 58393: scan.15
10feb516080006412101030317 = 5037: broadcast vdatetime
7f08b5110100 / 08a802103a040f0089 = 254
1008b5040100 / 0a00111608ffffffff0080 = 10608: bai DateTime
1008b5100900001478ffff01ff00 / 0101 = 32374: bai Mode
1008b5110101 / 095a5400805e5c0000ff = 32422: bai Status01
1008b5110102 / 05033c96467c = 11523: bai Status02
1008b512020000 / 00 = 181
1008b512020064 / 00 = 372
1008b513020507 / 00 = 538
7f08b503020001 / 0affffffffffffffffffff = 540: bai currenterror
7f08b509030d4b00 / 0150 = 480: bai HwcTypes

john30, 2017-03-05:
I don't know where the "360 SavingsFunctionTime" comes from in your readall output. seems you have done something weird with the CSVs.
the undecodable bai messages are due to the fact that your version is not really supported.

closed 2017-04-22 by john30