john30/ebusd #66: grab result decode crashes ebusd

s1nn3r, 2017-02-22:
Most of the time I try the new decode feature, ebusd crashes. Sometimes (with only 1 or 2 entries in the grab result) it shows the decoded output.

Example:
```
root@Linux:~# ebusctl grab result
1003050a00 = 1
1013050a00 = 1
100305020101 = 1
101305020101 = 1
100305010aee59320a000001000000 = 2
101305010a00000a0a000001000100 = 2
10fe050d0a0000800000800080ffff = 1

root@Linux:~# ebusctl grab result decode
<hangs>
```
ebusd crashes, no log entries.

ebusd github build from yesterday.

john30, 2017-02-22:
please post your configuration details (output of ebusctl info) and let me know which ebusd-configuration version you have

s1nn3r, 2017-02-22:
```
root@Linux:~# ebusctl info
version: ebusd 3.0pre.p20170221
signal: acquired
symbol rate: 63
reconnects: 0
masters: 5
messages: 99
conditional: 0
poll: 0
update: 9
address 01: master #6
address 03: master #11
address 10: master #2
address 13: master #12
address 15: slave #2
address 31: master #8, ebusd
address 36: slave #8, ebusd
```
using the ochser csvs (https://github.com/john30/ebusd-configuration/files/245766/ochsner_csv.zip) from sdfim.

legacycode, 2017-02-23:
I can confirm this. After excecuting `ebusctl grab result decode`  the ebusd immediately throws a `Floating point exception` and terminates.

```
version: ebusd 3.0pre.7d5dd27
signal: acquired
symbol rate: 37
reconnects: 0
masters: 3
messages: 557
conditional: 3
poll: 0
update: 11
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0502;HW=9102", loaded "bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0419;HW=4603", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
```
Best regards.

john30, 2017-02-23:
should be fixed with 510df55, please test

s1nn3r, 2017-02-23:
Yes, looks good.
Thanks!

closed 2017-02-23 by s1nn3r