john30/ebusd #69: docker images latest on synology ds 916+

chri2, 2017-02-24:
# docker run -it --rm john30/ebusd
ebusd: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by ebusd)

ldd `which ebusd`
/usr/bin/ebusd: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /usr/bin/ebusd)
	linux-vdso.so.1 (0x00007ffd9b5be000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007feb74c49000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007feb74a41000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007feb74736000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007feb74435000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007feb7421f000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007feb73e74000)
	/lib64/ld-linux-x86-64.so.2 (0x00007feb74e66000)

# apt-cache show libstdc++6
Package: libstdc++6
Source: gcc-4.9
Version: 4.9.2-10
Installed-Size: 1319
Maintainer: Debian GCC Maintainers <debian-gcc@lists.debian.org>
Architecture: amd64
Replaces: libstdc++6-4.9-dbg (<< 4.9.0-3)
Depends: gcc-4.9-base (= 4.9.2-10), libc6 (>= 2.18), libgcc1 (>= 1:4.1.1)
Pre-Depends: multiarch-support
Conflicts: scim (<< 1.4.2-1)
Breaks: gcc-4.3 (<< 4.3.6-1), gcc-4.4 (<< 4.4.6-4), gcc-4.5 (<< 4.5.3-2)
Description: GNU Standard C++ Library v3
Description-md5: 724ab84919e0e220afb960e36463914d
Multi-Arch: same
Homepage: http://gcc.gnu.org/
Tag: implemented-in::c++, role::shared-lib
Section: libs
Priority: important
Filename: pool/main/g/gcc-4.9/libstdc++6_4.9.2-10_amd64.deb
Size: 272946
MD5sum: 391ae724457378c71fb16fb5d8eb232c
SHA1: 234c8593643d0e3bcfd36302f48edeaa4ca01fe5
SHA256: f1509bbabd78e89c861de16931aec5988e1215649688fd4f8dfe1af875a7fbef

john30, 2017-02-24:
should be fixed with automated builds following commit 936669b

chri2, 2017-02-24:
Great :) !

closed 2017-02-24 by chri2