john30/ebusd #70: Install target vanished?

andig, 2017-02-26:
I'm building current git version according to wiki:

````
pi@heizung:~/ebusd $ make
Scanning dependencies of target ebuscontrib
[  3%] Building CXX object src/lib/ebus/contrib/CMakeFiles/ebuscontrib.dir/tem.cpp.o
[  7%] Linking CXX static library libebuscontrib.a
[ 11%] Built target ebuscontrib
Scanning dependencies of target utils
[ 14%] Building CXX object src/lib/utils/CMakeFiles/utils.dir/thread.cpp.o
[ 18%] Linking CXX static library libutils.a
[ 33%] Built target utils
Scanning dependencies of target ebus
[ 37%] Building CXX object src/lib/ebus/CMakeFiles/ebus.dir/datatype.cpp.o
[ 40%] Building CXX object src/lib/ebus/CMakeFiles/ebus.dir/data.cpp.o
[ 44%] Building CXX object src/lib/ebus/CMakeFiles/ebus.dir/device.cpp.o
[ 48%] Linking CXX static library libebus.a
[ 59%] Built target ebus
Scanning dependencies of target ebusd
[ 62%] Building CXX object src/ebusd/CMakeFiles/ebusd.dir/bushandler.cpp.o
[ 66%] Building CXX object src/ebusd/CMakeFiles/ebusd.dir/datahandler.cpp.o
[ 70%] Building CXX object src/ebusd/CMakeFiles/ebusd.dir/network.cpp.o
[ 74%] Building CXX object src/ebusd/CMakeFiles/ebusd.dir/mainloop.cpp.o
[ 77%] Building CXX object src/ebusd/CMakeFiles/ebusd.dir/main.cpp.o
[ 81%] Building CXX object src/ebusd/CMakeFiles/ebusd.dir/mqtthandler.cpp.o
[ 85%] Linking CXX executable ebusd
[ 85%] Built target ebusd
Scanning dependencies of target ebusfeed
[ 88%] Building CXX object src/tools/CMakeFiles/ebusfeed.dir/ebusfeed.cpp.o
[ 92%] Linking CXX executable ebusfeed
[ 92%] Built target ebusfeed
Scanning dependencies of target ebusctl
[ 96%] Building CXX object src/tools/CMakeFiles/ebusctl.dir/ebusctl.cpp.o
[100%] Linking CXX executable ebusctl
[100%] Built target ebusctl

pi@heizung:~/ebusd $ make install
make: *** No rule to make target 'install'.  Stop.

andig, 2017-02-26:
Also the `-V` version info seems to be gone:

````
pi@heizung:~/ebusd $ ./src/ebusd/ebusd -V
./src/ebusd/ebusd: invalid option -- 'V'
Try `ebusd --help' or `ebusd --usage' for more information.

pi@heizung:~/ebusd $ ebusd -V
ebusd 3.0pre.e256a74
````

So- its probably some kind of issue on my side but I'm stumped as to what might be going wrong here?

john30, 2017-02-26:
you seem to be using cmake. for that there is no install target.
thanks, the -V options seems to have gone with the intoduction of ebusd namespace. should be fixed with d3d4533

andig, 2017-02-26:
Confirmed. Version is back. If install is gone for good a wiki update might be helpful ;)

john30, 2017-02-26:
added hint in wiki

closed 2017-02-26 by john30