john30/ebusd #71: additional parameter "text + numeric value" for command-interface

J0EK3R, 2017-03-04:
It would be very useful to get both informations - text and numeric value - when reading messages and single fields with **TCP client command** or **ebusctl** for storing the data and plotting diagrams.

Till now you can either get the **text value** like this
```
ebusctl r -c hc1 Set Status
hotwaterinheating
```

or the **numeric value** by adding option "-n" to the command like this
```
ebusctl r -n -c hc1 Set Status
187
```

If we had anoher option - for example option "-t" ("t" for text value like "n" for numeric value) - we could get both information:
```
ebusctl r -t -n -c hc1 Set Status
hotwaterinheating 187
```

Another possibility could be to use option "-tn" for text-numeric and option "-nt" for numeric-text order.

J0EK3R, 2017-03-04:
Yet another idea is to use placeholders and formatproviders:
If there were a commandline option to pass a formatstring you could define your own output-format!

Example for placeholders {} for the internals fields like
{0} - Name of the field: Status
{1:t} - Value of the field as Text: hotwaterinheating
{1:n} - Value of the field as Number: 187
{2} - defined unit of the field: Unit
`
ebusctl r -t"mytext {0}: {1:t} {1:n} - {2}" -c hc1 Set Status
mytext Status: hotwaterinheating 187 - Unit
`

john30, 2017-04-22:
solved with new argument "-N" to read command in 4fdc9b2

closed 2017-04-22 by john30

J0EK3R, 2017-04-22:
Hello John,
I testet the new option "-N" and it seams to work proper!

Thank you!

```
pi@rasp2:~ $ ebusctl r -c hc1 ProgramChooseSwitch
Programm 2

pi@rasp2:~ $ ebusctl r -n -c hc1 ProgramChooseSwitch
12

pi@rasp2:~ $ ebusctl r -N -c hc1 ProgramChooseSwitch
12=Programm 2
```

But did you think about my other suggestion with the format-provider?
It would be the most flexible! :-)

john30, 2017-04-22:
yes I did and since this available in JSON from HTTP port that should be enough