john30/ebusd #74: Vaillant USB to ebus converter

rgillam100, 2017-03-15:
I have a genuine Vaillant USB to ebus converter, (Product: Vaillant VRD-USB PC-Interface)
which i am trying to get to work with Pi v3
I have the ebusd compiled and working on the PI.
The issues have been around getting the USB to mount as ttyUSB0
Using the following I have it mounted using the Vendor ID
modprobe ftdi_sio
echo "1482 1005" > /sys/bus/usb-serial/drivers/ftdi_sio/new_id
 
The issue now is when i run ebus it keeps reopening 

pi@raspberrypi:~/ebusd $ ebusd -f
2017-03-15 13:13:14.482 [main notice] ebusd 3.0pre.daeea20 started
2017-03-15 13:13:14.492 [bus error] device /dev/ttyUSB0 not available
2017-03-15 13:13:14.499 [main notice] found messages: 10 (0 conditional on 0 conditions, 0 poll, 7 update)
2017-03-15 13:13:24.495 [bus notice] re-opened /dev/ttyUSB0
2017-03-15 13:13:34.499 [bus notice] re-opened /dev/ttyUSB0

Is there any logging i can enable to see what is happening?

Thanks
Rob

john30, 2017-03-19:
this interface can't be used with ebusd as it does not provide serial connection IMO

rgillam100, 2017-03-21:
Thanks for answer, I will build a serial  interface

closed 2017-03-21 by rgillam100