john30/ebusd #75: EHP invalid position ERR

leschNUX, 2017-03-21:
Hi,
In my ebusd log, are always these messages in it. Do you have an idea, if this is missing CSV Infos. All other readings are recognized.

Heating:
geotherm vwl 61/3s

`2017-03-21 22:05:13.659 [update error] unable to parse ehp Mode from 1008b51009000100000000000002 / 00: ERR: invalid position
2017-03-21 22:05:22.509 [update error] unable to parse ehp Mode from 1008b51009000100000000000002 / 00: ERR: invalid position
2017-03-21 22:05:32.572 [update error] unable to parse ehp Mode from 1008b51009000100000000000002 / 00: ERR: invalid position
2017-03-21 22:05:42.643 [update error] unable to parse ehp Mode from 1008b51009000100000000000002 / 00: ERR: invalid position
2017-03-21 22:05:52.700 [update error] unable to parse ehp Mode from 1008b51009000100000000000002 / 00: ERR: invalid position
2017-03-21 22:06:02.775 [update error] unable to parse ehp Mode from 1008b51009000100000000000002 / 00: ERR: invalid position`

br

john30, 2017-03-25:
already answered [here](https://forum.fhem.de/index.php/topic,29737.msg611277.html#msg611277)

leschNUX, 2017-03-30:
Hi John,

i attached some file

[ebusdvaillant.txt](https://github.com/john30/ebusd/files/883500/ebusdvaillant.txt)

john30, 2017-04-08:
as already answered in the forum, your ehp device does not transmit messages in the correct format. So the CSVs would have to be adapted to your specified HW/SW version. Please use [ebusd-configuration issue tracking](https://github.com/john30/ebusd-configuration/issues) for reporting an issue regarding the configs.

closed 2017-04-08 by john30