john30/ebusd #76: ebusd isn't answering to scan telegrams any more

J0EK3R, 2017-03-30:
The current committed version [Mar 26, 2017 2b8b0860c18a5710f1f89bd723a952efc12b1fe7] of ebusd isn't answering to **scan telegrams** any more.

A previous version was working proper.

I'm using two ebus-adaptors connected to two different raspberries each running an instance of ebusd (default-address 0x31 and address 0xFF). Both instances of ebusd were set to answer.

Info of ebusd with address 0x31:
```
version: ebusd 3.0pre.2b8b086
signal: acquired
symbol rate: 22
max symbol rate: 88
reconnects: 0
masters: 6
messages: 552
conditional: 0
poll: 0
update: 11
address 03: master #11
address 08: slave #11, scanned "MF=Kromschroeder;ID=W ;SW=0216;HW=0101", loaded "kromschroeder/08..sc.csv"
address 30: master #3
address 31: master #8, ebusd (answering)
address 35: slave #3, scanned "MF=Kromschroeder;ID=;SW=2633;HW=0000", loaded "kromschroeder/35..hc1.csv"
address 36: slave #8, ebusd (answering)
address 51: slave, scanned "MF=Kromschroeder;ID=;SW=3233;HW=0001", loaded "kromschroeder/51..hc2.csv"
address 70: master #4
address 75: slave #4, scanned "MF=Kromschroeder;ID=;SW=2633;HW=0000", loaded "kromschroeder/75..hc2.csv"
address f1: master #10
address f6: slave #10, scanned "MF=Kromschroeder;ID=WWST?;SW=0216;HW=0101", loaded "kromschroeder/f6..sc.csv"
address ff: master #25
```

Info of ebusd with address 0xFF:
```
version: ebusd 3.0pre.2b8b086
signal: acquired
symbol rate: 23
max symbol rate: 90
reconnects: 0
masters: 6
messages: 549
conditional: 0
poll: 0
update: 8
address 03: master #11
address 04: slave #25, ebusd (answering)
address 08: slave #11, scanned "MF=Kromschroeder;ID=W ;SW=0216;HW=0101", loaded "kromschroeder/08..sc.csv"
address 30: master #3
address 31: master #8
address 35: slave #3, scanned "MF=Kromschroeder;ID=;SW=2633;HW=0000", loaded "kromschroeder/35..hc1.csv"
address 51: slave, scanned "MF=Kromschroeder;ID=;SW=3233;HW=0001", loaded "kromschroeder/51..hc2.csv"
address 70: master #4
address 75: slave #4, scanned "MF=Kromschroeder;ID=;SW=2633;HW=0000", loaded "kromschroeder/75..hc2.csv"
address f1: master #10
address f6: slave #10, scanned "MF=Kromschroeder;ID=WWST?;SW=0216;HW=0101", loaded "kromschroeder/f6..sc.csv"
address ff: master #25, ebusd (answering)
```

I'm watching **ebusd.log** an both raspies:
`pi@rasp2:~ $ tail -f /var/log/ebusd.log
`

And with this current builds of ebusd I can see the same **[main error]** in both logfiles:

Logfile of ebusd with address 0x31
```
2017-03-30 17:45:25.130 [main error] scan config 04: ERR: SYN received
2017-03-30 17:45:30.192 [update notice] update sc Act QQ=f1: 1;BrennerAus;1;1;1;0;0;0;1;0;0;Winter;0;0;0;0;0;Heating;0;33.0;-;47.0;0.0;21;16.047;39
2017-03-30 17:45:33.164 [update notice] update hc2 Set: hotwaterinheating;-;36.00;-;-;-;-
2017-03-30 17:45:33.371 [update notice] update hc2 Act QQ=70: 24.500;2;128;36.5;31.5;30.5;21;16;heating;0;0;0
2017-03-30 17:45:35.146 [main error] scan config 04: ERR: SYN received
2017-03-30 17:45:42.185 [update notice] update hc1 Set: hotwaterinheating;startconsumer;39.12;-;-;50.5;-
2017-03-30 17:45:43.442 [update notice] update hc2 Set: hotwaterinheating;-;36.00;-;-;-;-
2017-03-30 17:45:43.632 [update notice] update hc2 Act QQ=70: 24.500;2;128;36.5;31.5;30.5;21;16;heating;0;0;0
2017-03-30 17:45:45.138 [main error] scan config 04: ERR: SYN received
2017-03-30 17:45:52.156 [update notice] update hc1 Set: hotwaterinheating;startconsumer;39.12;-;-;50.5;-
```

Logfile of ebusd with address 0xFF
```
2017-03-30 17:44:53.961 [update notice] update sc FlameFailureDuringOperation QQ=31: 2723
2017-03-30 17:44:56.745 [main error] scan config 36: ERR: read timeout
2017-03-30 17:44:58.523 [update notice] update sc Act QQ=f1: 1;BrennerAus;1;1;1;0;0;0;1;0;0;Winter;0;0;0;0;0;Heating;0;33.0;-;47.0;0.0;21;16.047;39
2017-03-30 17:45:00.633 [update notice] update hc1 Set: hotwaterinheating;startconsumer;39.12;-;-;50.5;-
2017-03-30 17:45:01.922 [update notice] update broadcast datetime QQ=30: 21.000;17:45:00;30.03.2017
2017-03-30 17:45:03.200 [update notice] update hc2 Set: hotwaterinheating;-;36.00;-;-;-;-
2017-03-30 17:45:06.729 [main error] scan config 36: ERR: read timeout
2017-03-30 17:45:10.660 [update notice] update hc1 Set: hotwaterinheating;startconsumer;39.12;-;-;50.5;-
2017-03-30 17:45:11.922 [update notice] update hc2 Set: hotwaterinheating;-;36.00;-;-;-;-
2017-03-30 17:45:16.722 [main error] scan config 36: ERR: read timeout
2017-03-30 17:45:20.649 [update notice] update hc1 Set: hotwaterinheating;startconsumer;39.12;-;-;50.5;-
2017-03-30 17:45:21.923 [update notice] update hc2 Set: hotwaterinheating;-;36.00;-;-;-;-
2017-03-30 17:45:26.733 [main error] scan config 36: ERR: read timeout
```

An attempt to scan manually ends like this
Raspi with ebusd 0x31
```
pi@fhem:/etc/ebusd $ ebusctl scan 36
empty

pi@fhem:/etc/ebusd $ ebusctl scan 4
ERR: SYN received
```

john30, 2017-04-05:
should be fixed with edc1505

J0EK3R, 2017-04-08:
Sorry, isn't fixed...

Sadly the same behavior as before:

Raspi with ebusd 0x31
```
pi@fhem:~/src/ebusd $ ebusctl i
version: ebusd 3.0pre.6cfa74a

pi@fhem:~/src/ebusd $ ebusctl scan 36
ERR: read timeout

pi@fhem:~/src/ebusd $ ebusctl scan 4
ERR: read timeout

pi@fhem:~ $ tail -f /var/log/ebusd.log
2017-04-08 13:23:08.940 [main error] scan config 04: ERR: read timeout
```

Raspi with ebusd 0xFF
```
pi@rasp2:~/src $ ebusctl i
version: ebusd 3.0pre.6cfa74a

pi@rasp2:~/src $ ebusctl scan 4
ERR: read timeout

pi@rasp2:~/src $ ebusctl scan 36
ERR: read timeout

pi@rasp2:~ $ tail -f /var/log/ebusd.log
2017-04-08 13:23:34.982 [main error] scan config 36: ERR: read timeout
```

kkloesener, 2017-04-09:
Got same Problem here - with last change from #edc1505 it got better, but still a huge amount of errors

<pre>
2017-04-09 09:25:14.777 [bus error] send to 15: ERR: read timeout, retry
2017-04-09 09:25:15.349 [bus error] send to 15: ERR: read timeout, retry
2017-04-09 09:25:16.011 [bus error] send to 15: ERR: read timeout
2017-04-09 09:25:16.012 [bus error] send message part 0: ERR: read timeout
2017-04-09 09:25:16.568 [main error] scan config 26: ERR: CRC error
2017-04-09 09:25:16.580 [bus error] prepare message part 0: ERR: end of input reached
2017-04-09 09:25:16.597 [bus error] prepare message part 0: ERR: end of input reached
2017-04-09 09:25:18.768 [main error] scan config 36: ERR: read timeout
2017-04-09 09:25:20.937 [main error] scan config 50: ERR: read timeout
2017-04-09 09:25:23.109 [main error] scan config ec: ERR: read timeout
2017-04-09 09:25:35.246 [main error] scan config 23: ERR: read timeout
2017-04-09 09:25:37.395 [main error] scan config 26: ERR: CRC error
2017-04-09 09:25:39.551 [main error] scan config 36: ERR: read timeout
2017-04-09 09:25:41.915 [main error] scan config 50: ERR: read timeout
2017-04-09 09:25:44.123 [main error] scan config ec: ERR: read timeout
2017-04-09 09:25:56.294 [main error] scan config 23: ERR: read timeout
2017-04-09 09:25:57.507 [bus error] prepare message part 0: ERR: end of input reached
2017-04-09 09:25:57.525 [bus error] prepare message part 0: ERR: end of input reached
2017-04-09 09:25:57.561 [bus error] prepare message part 0: ERR: end of input reached
2017-04-09 09:25:57.579 [bus error] prepare message part 0: ERR: end of input reached
2017-04-09 09:25:57.807 [bus error] send to 15: ERR: read timeout, retry
2017-04-09 09:25:58.378 [bus error] send to 15: ERR: read timeout, retry
</pre>

john30, 2017-04-10:
@kkloesener that's a different issue. this one is about ebusd not answering to 0704 requests from other participants when in answering mode.
@J0EK3R can you check it again with commit 80e6f48?

J0EK3R, 2017-04-10:
Hello John,

I tested the current Version [80e6f48](https://github.com/john30/ebusd/commit/80e6f48cf1ec09bfdcc3fcc68c6f71faf3505d22) but it still isn't working:
(Btw. there are some compiler warnings in message.h)

Raspi with ebusd 0x31
```
pi@fhem:~/src/ebusd $ ebusctl i
version: ebusd 3.0pre.80e6f48

pi@fhem:~/src/ebusd $ ebusctl scan 36
ERR: read timeout

pi@fhem:~/src/ebusd $ ebusctl scan 4
ERR: read timeout

pi@fhem:~ $ tail -f /var/log/ebusd.log
2017-04-10 15:45:39.132 [main error] scan config 04: ERR: read timeout
```
 The same with my other raspi with ebusd address 0xFF...

john30, 2017-04-17:
should finally be solved with 038bb0e

J0EK3R, 2017-04-17:
Hi John,
looks better with the lates commit! aaea8f7 :-)

But take a closer look at the entries of ebusd (adr.31/36 and adr.04/ff): are there really some conflicts!?

Raspi with ebusd 0x31
```
pi@fhem:~/src/ebusd $ ebusctl i
version: ebusd 3.0pre.aaea8f7
signal: acquired
symbol rate: 22
max symbol rate: 81
reconnects: 0
masters: 6
messages: 552
conditional: 0
poll: 0
update: 11
address 03: master #11
address 04: slave #25, scanned "MF=ebusd.eu;ID=ebusd;SW=0300;HW=0100"
address 08: slave #11, scanned "MF=Kromschroeder;ID=W ;SW=0216;HW=0101", loaded "kromschroeder/08..sc.csv"
address 30: master #3
address 31: master #8, ebusd (answering), conflict
address 35: slave #3, scanned "MF=Kromschroeder;ID=;SW=2633;HW=0000", loaded "kromschroeder/35..hc1.csv"
address 36: slave #8, ebusd (answering), conflict
address 51: slave, scanned "MF=Kromschroeder;ID=;SW=3233;HW=0001", loaded "kromschroeder/51..hc2.csv"
address 70: master #4
address 75: slave #4, scanned "MF=Kromschroeder;ID=;SW=2633;HW=0000", loaded "kromschroeder/75..hc2.csv"
address f1: master #10
address f6: slave #10, scanned "MF=Kromschroeder;ID=WWST?;SW=0216;HW=0101", loaded "kromschroeder/f6..sc.csv"
address ff: master #25
```
```
pi@fhem:~/src/ebusd $ ebusctl scan 36
ERR: read timeout

pi@fhem:~/src/ebusd $ ebusctl scan 04
04;ebusd.eu;ebusd;0300;0100
```

Raspi with ebusd 0xFF
```
pi@rasp2:~/src/ebusd $ ebusctl i
version: ebusd 3.0pre.aaea8f7
signal: acquired
symbol rate: 37
max symbol rate: 57
reconnects: 0
masters: 6
messages: 549
conditional: 0
poll: 0
update: 8
address 03: master #11
address 04: slave #25, ebusd (answering), conflict
address 08: slave #11, scanned "MF=Kromschroeder;ID=W ;SW=0216;HW=0101", loaded "kromschroeder/08..sc.csv"
address 30: master #3
address 31: master #8
address 35: slave #3, scanned "MF=Kromschroeder;ID=;SW=2633;HW=0000", loaded "kromschroeder/35..hc1.csv"
address 36: slave #8, scanned "MF=ebusd.eu;ID=ebusd;SW=0300;HW=0100"
address 51: slave, scanned "MF=Kromschroeder;ID=;SW=3233;HW=0001", loaded "kromschroeder/51..hc2.csv"
address 70: master #4
address 75: slave #4, scanned "MF=Kromschroeder;ID=;SW=2633;HW=0000", loaded "kromschroeder/75..hc2.csv"
address f1: master #10
address f6: slave #10, scanned "MF=Kromschroeder;ID=WWST?;SW=0216;HW=0101", loaded "kromschroeder/f6..sc.csv"
address ff: master #25, ebusd (answering), conflict
```
```
pi@rasp2:~/src/ebusd $ ebusctl scan 04
ERR: read timeout

pi@rasp2:~/src/ebusd $ ebusctl scan 36
36;ebusd.eu;ebusd;0300;0100
```

john30, 2017-04-17:
can't reproduce this conflict issue... are you doing anything special?

john30, 2017-04-17:
conflict issue should be fixed with commit f252b07

J0EK3R, 2017-04-18:
Hi John,
I just did a quick check with 27f28eb.
The conflict issue seems to be fixed! :-)

closed 2017-04-19 by john30

Dominic151617, 2023-12-20:
@J0EK3R: a littel bit off Topic, but where can I Find the kromschroeder/f6.csv

J0EK3R, 2023-12-20:
...I'm completely out with the materia but perhaps you are searching for my [Repo](https://github.com/J0EK3R/ebusd-configuration-weishaupt/tree/master)?

Dominic151617, 2023-12-20:
@J0EK3R Thanks!

papax333, 2024-01-28:
is ist possible that the problem is on again?

```
version: ebusd 23.3.23.3-29-g2ea20ed8
update check: revision 23.3 available, broadcast.csv: different version available, memory.csv: different version available
device: 192.168.1.150:9999, TCP, enhanced, firmware 1.1[4106].1[4106]
signal: acquired
symbol rate: 22
max symbol rate: 67
min arbitration micros: 7
max arbitration micros: 17
min symbol latency: 11
max symbol latency: 29
scan: finished
reconnects: 0
masters: 6
messages: 101
conditional: 0
poll: 0
update: 21
address 03: master #11
address 08: slave #11, scanned "MF=Kromschroeder;ID=  3B ;SW=0777;HW=5130", loaded "kromschroeder/08.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Kromschroeder;ID=  ;SW=0204;HW=-"
address 30: master #3
address 31: master #8, ebusd (answering)
address 35: slave #3, scanned "MF=Kromschroeder;ID=  ;SW=0204;HW=-", loaded "kromschroeder/35.csv"
address 36: slave #8, ebusd (answering), scanning
address 70: master #4
address 75: slave #4, scanned "MF=Kromschroeder;ID=  ;SW=0204;HW=-"
```

```
ebusctl scan 36
ERR: read timeout
```