john30/ebusd #77: Listen doesn't appear to pick up message changes in the Master message

brucemiranda, 2017-03-30:
If I keep listen running, it picks up all the changes where the changes are to the Right of the / but any changes to the Left of the / are not displayed even if they are changing. 

e.g

7f08b51009000042ffff6400ff00 / 0101 changed to 7f08b51009000014ffff6401ff00 / 0101 but nothing was displayed on the listen output.

john30, 2017-04-05:
please provide the message definition (i.e. result of find -f with the right message name and circuit)

brucemiranda, 2017-04-05:
r,bai,ModeDesired,Betriebsart,,08,b510,00,hcmode,m,UCH,0=auto;1=off;2=water,,heating circuit mode,flowtempdesired,m,D1C,,°C,temperature,hwctempdesired,m,D1C,,°C,temperature,hwcflowtempdesired,m,UCH,,°C,temperature,,m,IGN:1,,,,disablehc,m,BI0:1,,,,disablehwctapping,m,BI1:1,,,,disablehwcload,m,BI2:1,,,,,m,IGN:1,,,,remoteControlHcPump,m,BI0:1,,,,releaseBackup,m,BI1:1,,,,releaseCooling,m,BI2:1,,,,currentmode,s,UCH,0=off;1=standby;2=heat;3=water,,

john30, 2017-04-09:
should be done with a19a382, please check

brucemiranda, 2017-04-10:
Yes it works now. Thanks a lot.

closed 2017-04-14 by john30