john30/ebusd #79: Arch Linux: cannot query ebusd anymore after sys upgrade

milankni, 2017-04-09:
Hi,

I use Arch Linux on RPi 2 and after update of many packages (see below), I cannot query ebusd anymore. I use Ebusd Koppler by eservice-online.de (USB version).

What still works
Ebusd can 'see' the messages broadcasted by the HVAC (well, the 'read timeout' for ACK is a new message):
```
2017-04-09 17:20:26.934 [bus debug] ERR: read timeout during receive command ACK, switching to skip                     
2017-04-09 17:20:30.635 [update info] update MS cmd: 1008b5110101 / 0928296011ffff0000ff                                
2017-04-09 17:20:30.636 [update notice] update MS StatusTHER: 20.0;20.5;17.375;-;-;0                                    
2017-04-09 17:20:32.290 [bus debug] ERR: read timeout during receive command ACK, switching to skip                     
2017-04-09 17:20:32.871 [update info] update MS cmd: 1008b5040100 / 0a00ffffffffffffff6011                              
2017-04-09 17:20:32.872 [update notice] update MS OutsideTemp: 17.375                                                   
2017-04-09 17:20:33.127 [update info] update MS cmd: 1008b5110102 / 050014965a78                                        
2017-04-09 17:20:33.127 [update notice] update MS PumpStatus: off                                                       
2017-04-09 17:20:34.723 [main debug] performing regular tasks
```
However, any command to actively query ebus results in errors, e.g. 'ebusctl read RoomTemp':
```
2017-04-09 09:46:08.259 [main debug] >>> read RoomTemp                                                                  
2017-04-09 09:46:08.259 [bus info] send message: 3115b509030d3a00                                                       
2017-04-09 09:46:08.301 [bus debug] notify request: ERR: read timeout                                                   
2017-04-09 09:46:08.302 [bus error] send to 15: ERR: read timeout, retry                                                
2017-04-09 09:46:08.302 [bus debug] ERR: read timeout during ready, switching to skip                                   
2017-04-09 09:46:08.364 [bus debug] notify request: ERR: read timeout                                                   
2017-04-09 09:46:08.365 [bus error] send to 15: ERR: read timeout, retry                                                
2017-04-09 09:46:08.365 [bus debug] ERR: read timeout during ready, switching to skip                                   
2017-04-09 09:46:08.429 [bus debug] notify request: ERR: read timeout                                                   
2017-04-09 09:46:08.429 [bus error] send to 15: ERR: read timeout, retry                                                
2017-04-09 09:46:08.429 [bus debug] ERR: read timeout during ready, switching to skip                                   
2017-04-09 09:46:08.493 [bus debug] notify request: ERR: read timeout                                                   
2017-04-09 09:46:08.493 [bus error] send to 15: ERR: read timeout                                                       
2017-04-09 09:46:08.493 [bus error] send message part 0: ERR: read timeout                                              
2017-04-09 09:46:08.493 [main debug] <<< ERR: read timeout                                                              
2017-04-09 09:46:08.493 [bus debug] ERR: read timeout during ready, switching to skip                                   
2017-04-09 09:46:08.494 [main debug] >>>                                                                                
                                                                                                                        
 ... lot of empty lines removed ...                                                                                     
                                                                                                                        
2017-04-09 09:46:08.494 [main debug] <<< ERR: command not found                                                         
2017-04-09 09:46:08.494 [network debug] [00007] wait for result                                                         
2017-04-09 09:46:08.494 [network info] [00007] connection closed
```
I tried various versions of ebusd (2.1 - master), but no go. Interestingly, in raspbian jessy ebusd 2.4 (binary downloaded from github) works just fine.

Shortly before the failure, I upgraded many packages (list is attached), although I assume the best candidate might be kernel or rpi firmware or possibly the new boot configuration (there is a new directory /boot/overlays, whose README says that RPi now boots with device tree enabled - I do not understand the architecture much to be able to comment on that). 

[pacman.log.txt](https://github.com/john30/ebusd/files/908375/pacman.log.txt)

Any idea?

john30, 2017-04-10:
is it stilla jessy distribution or something different?
check the read/write attributes on the device node.

milankni, 2017-04-10:
It is not a permissions problem on /dev/ttyUSB0 (rw), it more appears to be somewhat similar to this report (cannot write to serial device): https://openenergymonitor.org/forum-archive/node/12360.html

For the sake of completness, the upgrade on Arch Linux included:
```
upgraded linux-raspberrypi (4.4.41-1 -> 4.9.16-1)
upgraded linux-firmware (20161005.9c71af9-1 -> 20170309.695f2d6-1)
upgraded raspberrypi-firmware (20161211-2 -> 20170319-1)
upgraded raspberrypi-bootloader-x (20170112-1 -> 20170319-1)
upgraded raspberrypi-bootloader (20170112-1 -> 20170319-1)
```
I found prior binary packages in the package manager's cache, so I will try to downgrade the kernel and/or firmware and check if it helps.

P.S. I have two RPis: the main one (RPi 2 B) runs Arch Linux, the spare one (RPi B) runs Debian Jessie with ebusd as a temporary workaround to the above issue. Jessie is still on 4.4 kernel series.

john30, 2017-04-10:
as written in the forum:
try with bigger latency values to ebusd or check how to adjust the kernel to not buffer that much and transfer each single byte as fast as possible from/to user level.

milankni, 2017-04-10:
Increasing the latencies did not help. Although I did not check the buffers, CPU load is like 0.05 to 0.40 or so.
As a workaround, I downgraded linux kernel to 4.4.41 (it seems to be compatible with other up-to-date firmware packages).

I am not really skilled in finding out the source files, which take care about FTDI USB converter (used in the device). It should probably be reported upstream, but where to?

P.S. I can test patches (kernel compilation on Arch should be a straighforward issue, hopefully also on ARM).

john30, 2017-04-14:
there were some changes made to the ftdi_sio module at least in kernel 4.9.13, maybe that's the reason for the problems. could you try with latest 4.9.22 again?

milankni, 2017-04-19:
Nope. 4.9.22-0 does still not work.

john30, 2017-04-22:
too bad, maybe I'll find the time to test it on an updated arch linux

john30, 2017-04-22:
I'm afraid I can't reproduce this issue with kernel 4.9.23-1-ARCH. Everything works fine.

milankni, 2017-04-22:
I tried again with 4.9.23-1-ARCH but it does not work for me. Did you try ARM or x86?

john30, 2017-04-23:
on an RPi 2 B, i.e. ARM

milankni, 2017-04-26:
Maybe it is due to FTDI hw used in the coupler.
When I have time (might take a few weeks), I try to copy a fresh Arch Linux install on a separate SD card to see if it makes any difference.

Crazyachmed, 2017-05-01:
I have the same issue.

Working: 4.9.12-1
Broken (at least) since 4.9.14-2
Still broken: 4.9.25-1

So the issue has to be in somewhere between the first to versions which points to 4.9.13 as you already mentioned.

Is there anything I can run with debugs flags or do you want me to test 4.9.13 specifically?

john30, 2017-05-01:
you could try to directly read from the device (without starting ebusd) in order to see whether it works at all

doofy, 2017-05-12:
I can reproduce with arch linux. Does throw ERR with 4.9.27-1. Works fine with 4.4.39-1. I can provide any data/test you need @john30.

john30, 2017-05-13:
as said, try to directly read from the device without using ebusd to see whether reading from serial works at all

pbystrov, 2017-05-14:
I also can reproduce on Raspbian 4.9.27+.  Direct reading from the device works and running
`ebusd -f -c /tmp --logareas bus --loglevel info --lograwdata=bytes -d /dev/ttyUSB0`
reliably returns `<aa` 

I'm using Raspberry Pi 1 Model B and eBus Coupler USB from E-Service Online.

john30, 2017-05-15:
@pbystrov ok, and can you write on the device as well?

pbystrov, 2017-05-15:
@john30 Yes,
`` # cat < /dev/ttyUSB0 &``
and
`` # echo "AT" > /dev/ttyUSB0``
completed successfully

john30, 2017-05-25:
okay, but the question is, whether sending e.g. some data will also echo that data back, so when you send "AT" to the device then the same sequence should appear also when reading from the device simultaneously.

pbystrov, 2017-05-25:
@john30 yes, AT clearly echoed back. I tried this several times and it is always returned
```
# systemctl stop ebusd.service
# cat < /dev/ttyUSB0 &
[1] 12852
# ▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒echo "AT" > /dev/ttyUSB0▒▒▒▒▒
# ▒AT
▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒fg▒▒▒
cat < /dev/ttyUSB0
▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒^C
# uname -a
Linux loco 4.9.28+ #999 Tue May 16 13:59:16 BST 2017 armv6l GNU/Linux
```

john30, 2017-05-25:
okay, then please start ebusd with raw logging (--lograwdata) and post some of the log around a read timeout

pbystrov, 2017-05-25:
EBUSD_OPTS="--scanconfig --logareas=all --loglevel=debug --lograwdata"
```
2017-05-25 11:33:38.452 [main notice] ebusd 3.0pre.bf0fcd0 started
2017-05-25 11:33:38.453 [main info] loading configuration files from /etc/ebusd
2017-05-25 11:33:38.511 [main info] read templates in /etc/ebusd
2017-05-25 11:33:38.511 [main info] reading file /etc/ebusd/memory.csv
2017-05-25 11:33:38.517 [main info] reading file /etc/ebusd/broadcast.csv
2017-05-25 11:33:38.548 [main info] read config files
2017-05-25 11:33:38.563 [bus notice] bus started with own address 31/36
2017-05-25 11:33:38.574 [bus debug] ERR: SYN received during no signal, switching to ready
2017-05-25 11:33:38.574 [bus notice] signal acquired
2017-05-25 11:33:48.565 [main debug] performing regular tasks
2017-05-25 11:33:48.565 [main notice] starting initial broadcast scan
2017-05-25 11:33:48.566 [bus info] send message: 31fe070400
2017-05-25 11:33:48.587 [bus debug] notify request: ERR: read timeout
2017-05-25 11:33:48.588 [bus error] send to fe: ERR: read timeout, retry
2017-05-25 11:33:48.624 [bus notice] >31
2017-05-25 11:33:48.635 [bus debug] notify request: ERR: read timeout
2017-05-25 11:33:48.636 [bus error] send to fe: ERR: read timeout, retry
2017-05-25 11:33:48.672 [bus notice] >31
2017-05-25 11:33:48.683 [bus debug] notify request: ERR: read timeout
2017-05-25 11:33:48.683 [bus error] send to fe: ERR: read timeout, retry
2017-05-25 11:33:48.703 [bus notice] >31
2017-05-25 11:33:48.715 [bus debug] notify request: ERR: read timeout
2017-05-25 11:33:48.715 [bus error] send to fe: ERR: read timeout
2017-05-25 11:33:48.715 [main error] initial scan failed: ERR: read timeout
2017-05-25 11:33:48.751 [bus notice] >31
2017-05-25 11:33:58.716 [main debug] performing regular tasks
2017-05-25 11:34:08.716 [main debug] performing regular tasks

john30, 2017-05-25:
that's weird as not a single byte is received at all.
please repeat with --lograwdata=bytes

pbystrov, 2017-05-25:
Skipped some `<aa` lines for brewity
```2017-05-25 11:43:50.776 [main info] read templates in /etc/ebusd
2017-05-25 11:43:50.776 [main info] reading file /etc/ebusd/memory.csv
2017-05-25 11:43:50.791 [main info] reading file /etc/ebusd/broadcast.csv
2017-05-25 11:43:50.805 [main info] read config files
2017-05-25 11:43:50.812 [bus notice] bus started with own address 31/36
2017-05-25 11:43:50.822 [bus notice] <aa
2017-05-25 11:43:50.823 [bus debug] ERR: SYN received during no signal, switching to ready
2017-05-25 11:43:50.823 [bus notice] signal acquired
2017-05-25 11:43:50.824 [bus notice] <aa
**(230 [bus notice] lines skipped)**
2017-05-25 11:44:00.792 [bus notice] <aa
2017-05-25 11:44:00.813 [main debug] performing regular tasks
2017-05-25 11:44:00.813 [main notice] starting initial broadcast scan
2017-05-25 11:44:00.814 [bus info] send message: 31fe070400
2017-05-25 11:44:00.839 [bus notice] <aa
2017-05-25 11:44:00.841 [bus notice] >31
2017-05-25 11:44:00.851 [bus debug] notify request: ERR: read timeout
2017-05-25 11:44:00.851 [bus error] send to fe: ERR: read timeout, retry
2017-05-25 11:44:00.856 [bus notice] <31
2017-05-25 11:44:00.887 [bus notice] <aa
2017-05-25 11:44:00.889 [bus notice] >31
2017-05-25 11:44:00.899 [bus debug] notify request: ERR: read timeout
2017-05-25 11:44:00.900 [bus error] send to fe: ERR: read timeout, retry
2017-05-25 11:44:00.904 [bus notice] <31
2017-05-25 11:44:00.920 [bus notice] <aa
2017-05-25 11:44:00.921 [bus notice] >31
2017-05-25 11:44:00.932 [bus debug] notify request: ERR: read timeout
2017-05-25 11:44:00.932 [bus error] send to fe: ERR: read timeout, retry
2017-05-25 11:44:00.936 [bus notice] <31
2017-05-25 11:44:00.968 [bus notice] <aa
2017-05-25 11:44:00.969 [bus notice] >31
2017-05-25 11:44:00.980 [bus debug] notify request: ERR: read timeout
2017-05-25 11:44:00.980 [bus error] send to fe: ERR: read timeout
2017-05-25 11:44:00.980 [main error] initial scan failed: ERR: read timeout
2017-05-25 11:44:00.984 [bus notice] <31
2017-05-25 11:44:01.016 [bus notice] <aa
**(255 [bus notice] lines skipped)**
2017-05-25 11:44:10.970 [bus notice] <aa
2017-05-25 11:44:10.981 [main debug] performing regular tasks
2017-05-25 11:44:11.018 [bus notice] <aa
2017-05-25 11:44:11.066 [bus notice] <aa
...

john30, 2017-05-25:
okay, it seems it takes 15 ms until a sent byte is received back from the serial connection. please try again with --latency=20000

pbystrov, 2017-05-25:
Didn't help
```
2017-05-25 17:28:57.181 [main notice] ebusd 3.0pre.bf0fcd0 started
2017-05-25 17:28:57.182 [main info] loading configuration files from /etc/ebusd
2017-05-25 17:28:57.230 [main info] read templates in /etc/ebusd
2017-05-25 17:28:57.230 [main info] reading file /etc/ebusd/memory.csv
2017-05-25 17:28:57.241 [main info] reading file /etc/ebusd/broadcast.csv
2017-05-25 17:28:57.255 [main info] read config files
2017-05-25 17:28:57.262 [bus notice] bus started with own address 31/36
2017-05-25 17:28:57.273 [bus notice] <aa
2017-05-25 17:28:57.273 [bus debug] ERR: SYN received during no signal, switching to ready
2017-05-25 17:28:57.273 [bus notice] signal acquired
2017-05-25 17:28:57.274 [bus notice] <aa
2017-05-25 17:28:57.304 [bus notice] <aa

*** skipped bus <aa notices ***

2017-05-25 17:29:07.259 [bus notice] <aa
2017-05-25 17:29:07.263 [main debug] performing regular tasks
2017-05-25 17:29:07.263 [main notice] starting initial broadcast scan
2017-05-25 17:29:07.264 [bus info] send message: 31fe070400
2017-05-25 17:29:07.307 [bus notice] <aa
2017-05-25 17:29:07.308 [bus notice] >31
2017-05-25 17:29:07.322 [bus notice] <31
2017-05-25 17:29:07.322 [bus debug] switching from ready to send command
2017-05-25 17:29:07.323 [bus notice] >fe
2017-05-25 17:29:07.338 [bus notice] <fe
2017-05-25 17:29:07.340 [bus notice] >07
2017-05-25 17:29:07.346 [bus notice] <07
2017-05-25 17:29:07.347 [bus notice] >04
2017-05-25 17:29:07.361 [bus notice] <3d
2017-05-25 17:29:07.361 [bus debug] notify request: ERR: wrong symbol received
2017-05-25 17:29:07.362 [bus error] send to fe: ERR: wrong symbol received, retry
2017-05-25 17:29:07.362 [bus debug] ERR: wrong symbol received during send command, switching to skip
2017-05-25 17:29:07.363 [bus notice] <c1
2017-05-25 17:29:07.953 [bus notice] <aa
2017-05-25 17:29:07.955 [bus notice] >31
2017-05-25 17:29:07.968 [bus notice] <31
2017-05-25 17:29:07.969 [bus debug] switching from ready to send command
2017-05-25 17:29:07.970 [bus notice] >fe
2017-05-25 17:29:07.985 [bus notice] <fe
2017-05-25 17:29:07.986 [bus notice] >07
2017-05-25 17:29:07.987 [bus notice] <aa
2017-05-25 17:29:07.987 [bus debug] notify request: ERR: SYN received
2017-05-25 17:29:07.988 [bus error] send to fe: ERR: SYN received, retry
2017-05-25 17:29:07.988 [bus debug] ERR: SYN received during send command, switching to ready
2017-05-25 17:29:07.989 [bus notice] >31
2017-05-25 17:29:08.001 [bus notice] <07
2017-05-25 17:29:08.001 [bus debug] ERR: arbitration lost during ready, retry
2017-05-25 17:29:08.003 [bus notice] <31
2017-05-25 17:29:08.033 [bus notice] <aa
2017-05-25 17:29:08.033 [bus debug] ERR: SYN received during receive command, switching to ready
2017-05-25 17:29:08.081 [bus notice] <aa
2017-05-25 17:29:08.128 [bus notice] <aa
2017-05-25 17:29:08.130 [bus notice] >31
2017-05-25 17:29:08.145 [bus notice] <31
2017-05-25 17:29:08.145 [bus debug] switching from ready to send command
2017-05-25 17:29:08.146 [bus notice] >fe
2017-05-25 17:29:08.160 [bus notice] <fe
2017-05-25 17:29:08.161 [bus notice] >07
2017-05-25 17:29:08.166 [bus notice] <08
2017-05-25 17:29:08.166 [bus debug] notify request: ERR: wrong symbol received
2017-05-25 17:29:08.167 [bus error] send to fe: ERR: wrong symbol received, retry
2017-05-25 17:29:08.167 [bus debug] ERR: wrong symbol received during send command, switching to skip
2017-05-25 17:29:08.181 [bus notice] <fe
2017-05-25 17:29:08.757 [bus notice] <aa
2017-05-25 17:29:08.758 [bus notice] >31
2017-05-25 17:29:08.772 [bus notice] <31
2017-05-25 17:29:08.772 [bus debug] switching from ready to send command
2017-05-25 17:29:08.774 [bus notice] >fe
[... 59 lines cut ...]
2017-05-25 17:29:18.902 [bus debug] ERR: SYN received during receive command, switching to ready
2017-05-25 17:29:18.950 [bus notice] <aa

*** skipped bus <aa notices ***

2017-05-25 17:29:20.851 [bus notice] <aa
2017-05-25 17:29:20.858 [main debug] performing regular tasks
2017-05-25 17:29:20.899 [bus notice] <aa
2017-05-25 17:29:20.948 [bus notice] <aa
2017-05-25 17:29:20.982 [bus notice] <aa
2017-05-25 17:29:21.028 [bus notice] <aa
2017-05-25 17:29:21.075 [bus notice] <aa
2017-05-25 17:29:21.123 [bus notice] <aa
2017-05-25 17:29:21.155 [bus notice] <aa
2017-05-25 17:29:21.203 [bus notice] <aa
2017-05-25 17:29:21.219 [bus notice] <03
2017-05-25 17:29:21.220 [bus notice] <64
2017-05-25 17:29:21.222 [bus notice] <b5
2017-05-25 17:29:21.223 [bus notice] <12
2017-05-25 17:29:21.235 [bus notice] <02
2017-05-25 17:29:21.237 [bus notice] <02
2017-05-25 17:29:21.238 [bus notice] <00
2017-05-25 17:29:21.239 [bus notice] <66
2017-05-25 17:29:21.286 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2017-05-25 17:29:21.747 [bus notice] <aa

john30, 2017-05-28:
too bad: the SYN generator in your configuration creates a SYN symbol 16 ms after ebusd has sent the last symbol and thus interrupts ebusd's sending on the bus. This is less than the latency and as a consequence, ebusd will never be able to communicate thoroughly.
The only solution right now is to find and install a kernel version that does not impose this latency.

Furthermore, different bytes are received back from the bus sometimes, so I guess your interface also has a little trimming issue.

john30, 2017-05-28:
The latency issue is known to be present on the following kernel versions:
4.9.23-1-ARCH
4.9.30-1-ARCH

john30, 2017-05-28:
found a potential solution:
By running `/sbin/setserial /dev/ttyUSB0 low_latency` the poblem can be fixed, at least on my hardware

john30, 2017-05-28:
succesful writing on the bus after issuing `/sbin/setserial /dev/ttyUSB0` low_latency was reported by another user as well, so I guess this issue can be closed soon

Crazyachmed, 2017-05-28:
Hmm, could you please implement whatever syscall setserial uses into ebusd? Maybe with a simple command line parameter instead of fancy auto detection. If we don't have that ebusd's reconnection feature would be useless

john30, 2017-05-28:
@Crazyachmed setserial will change the setting on the device until reboot, so this does not influence reconnects at all. I'm adding it to the init script right now.

milankni, 2017-06-02:
The 'setserial $dev low_latency' works for me, too. I created patches for contrib/archlinux to reflect the changes done in init script of contrib/debian.

john30, 2017-06-03:
closed with 79c7383

closed 2017-06-03 by john30