john30/ebusd #80: add "listen unknown" command

john30, 2017-04-15:


brucemiranda, 2017-07-05:
Also maybe adding a TimeStamp will be especially helpful with the "listen unknown" command.

john30, 2017-07-06:
what would be the use case for having a timestamp there? When interpreting the output line programmatically, the time of reception is exactly the timestamp, so I don't see any benefit here.

brucemiranda, 2017-07-06:
This would be useful if we kept listen running for a long time, say overnight. You can then see what time a particular message was seen on the bus. Ofcourse, one could always direct the telnet output to a file and use a program that adds timestamp to every output line. (e.g. RealTerm)

john30, 2017-07-09:
all this information is available in ebusd log, so I don't see the benefit

brucemiranda, 2017-11-06:
Another use case for a listen unknown is that listen will only list out known and configured messages. Listen is often used to look at the messages in real time and you would want to know if any un-configured messages were seen on the bus.

commit bbfe10b9 (2019-10-27): add option to listen to unknown messages (fixes #80)
closed 2019-10-27 by john30