john30/ebusd #81: Connection iterator error in Network::cleanConnection

odewdney, 2017-04-19:
The iterator is incremented after erasing entry - fix:

void Network::cleanConnections() {
	list<Connection*>::iterator c_it = m_connections.begin();
  while (c_it != m_connections.end()) {
	  if (!(*c_it)->isRunning()) {
		  Connection* connection = *c_it;
		  c_it = m_connections.erase(c_it);
		  delete connection;
		  logDebug(lf_network, "dead connection removed - %d", m_connections.size());
	  }
	  else
		  c_it++;
  }
}

odewdney, 2017-04-20:
Also i think there are a number of ".erase(it--)" which are working more by luck - i understand that the order of the erase and the decrement is undefined - but typically the compiler will do the "--" first.

john30, 2017-04-22:
its not a real problem since cleanConnections is called frequently, so an entry would potentially be skipped on one call but removed afterwards.
your second comment is wrong since "it--" is a postfix operator. the order is indeed well defined.

john30, 2017-04-22:
fixed with 62bab7d

closed 2017-04-22 by john30

odewdney, 2017-04-22:
The question was if the -- happens before the erase, or after, if after, then it is pointing to a deleted entry. Also if its pointing to a beginning entry, then you cant --. I was getting a ASSERT thrown in MSVC libraries so i had to change:

diff --git a/src/lib/ebus/message.cpp b/src/lib/ebus/message.cpp
index 8f6a842..cc7f54d 100644
--- a/src/lib/ebus/message.cpp
+++ b/src/lib/ebus/message.cpp
@@ -2562,10 +2573,13 @@ void MessageMap::clear() {
         if (keyIt != m_messagesByKey.end()) {
           vector<Message*>* keyMessages = &keyIt->second;
           if (!keyMessages->empty()) {
-            for (vector<Message*>::iterator kit = keyMessages->begin(); kit != keyMessages->end(); kit++) {
-              if (*kit == message) {
-                keyMessages->erase(kit--);
-              }
+                         vector<Message*>::iterator kit = keyMessages->begin();
+            while ( kit != keyMessages->end()) {
+                               if (*kit == message) {
+                                       kit = keyMessages->erase(kit);
+                               }
+                               else
+                                       kit++;
             }
           }
         }

john30, 2017-04-23:
okay, got your point now. looked through all iterator usages and should be fixed with 43042ee