john30/ebusd #82: docker build

kobza70, 2017-04-19:
I am getting error starting the ebusd docker (debian, jessie 64bit)

ebusd: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by ebusd)

Any idea about the reason?

Thanks
Kobza70

chri2, 2017-04-19:
I had the same problem (trying to start on a synology 64-bit system) and could solve the issue by installing gcc from testing inside the docker image.

kobza70, 2017-04-19:
thanks, but I I cannot log-in into the docker image since the error ouccurs before the docker is started...

john30, 2017-04-22:
probably the wrong libraries due to updates in the docker hub.
to avoid this, use the devel version of the image instead.
anyway, the images have just been updated with 753ed44

closed 2017-09-02 by john30