john30/ebusd #84: ebusd does not load csv files - scan emtpy

kkloesener, 2017-05-10:
After update to latest version yesterday ebusd does no longer load csv files after scan.
Scan result show only one known device but info does show all devices.

<pre>
localhost: info
version: ebusd 3.0pre.340f862
signal: acquired
symbol rate: 22
max symbol rate: 140
reconnects: 0
masters: 5
messages: 24
conditional: 0
poll: 0
update: 7
address 01: master #6
address 03: master #11
address 06: slave #6
address 08: slave #11
address 0a: slave
address 10: master #2
address 12: slave
address 26: slave
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 50: slave
address ec: slave
address ed: slave
address f7: master #20
address fc: slave #20, scanned "MF=Vaillant;ID=PMW01;SW=0205;HW=8302"
</pre>

<pre>
localhost: scan result
fc;Vaillant;PMW01;0205;8302
</pre>

In attached logfile you can see different error messages about timeouts or "element not found". 
[ebusd.log.txt](https://github.com/john30/ebusd/files/989397/ebusd.log.txt)

I dont know what this could be. RPi 1 with raspbian kernel 4.9.26+

john30, 2017-05-12:
note to myself: seems that not finding the potential additional "scan" messages is the problem, as those are available only after the vendor ID was extracted from the first ID message

john30, 2017-05-13:
problem should be fixed with 92f575a.
@kkloesener could you check again please?

closed 2017-05-13 by john30

kkloesener, 2017-05-14:
Scan now shows all devices, but ebusd does not load the corresponding csv files.
Do you need logfiles?

john30, 2017-05-15:
yes, the ebusd log would be great

kkloesener, 2017-05-16:
find attached the logfile

this is the status at the end of the logfile:
<pre>
localhost: info
version: ebusd 3.0pre.1697c93
update check: revision dbe70f0 available, broadcast.csv: different version available
signal: acquired
symbol rate: 41
max symbol rate: 94
reconnects: 0
masters: 4
messages: 20
conditional: 0
poll: 0
update: 7
address 01: master #6
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0703;HW=7401"
address 0a: slave, scanned "MF=Vaillant;ID=PMW01;SW=0205;HW=8302"
address 10: master #2
address 12: slave, scanned "MF=Vaillant;ID=PMW01;SW=0205;HW=8302"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address ed: slave, scanned "MF=Vaillant;ID=PMS02;SW=0209;HW=8402"

localhost: scan result
08;Vaillant;BAI00;0703;7401
0a;Vaillant;PMW01;0205;8302
12;Vaillant;PMW01;0205;8302
ed;Vaillant;PMS02;0209;8402
</pre>

[ebusd.txt](https://github.com/john30/ebusd/files/1005282/ebusd.txt)

john30, 2017-05-25:
is your device able to actually write on the bus? or was it able to do so before?

kkloesener, 2017-05-26:
i am currently not able to test since i'm not at home till next week.

previously i was able to write to the bus.

I will respond back asap

kkloesener, 2017-05-28:
ok, the workaround itself worked, but i checked the write behavior and it seems that i'm also got the problem as in #79 
After executing <code> /sbin/setserial /dev/ttyUSB0 low_latency </code> evereything works as expected.
Im now on raspbian 4.9.30+ with newest ebusd

commit 92f575a0 (2017-05-13): store read scan message event if a defined message also exists (should fix #84