john30/ebusd #85: MQTT stops on invalid value

russss, 2017-05-11:
If ebusd gets an "invalid position" error on reading a value, it causes an error which stops MQTT messages for that circuit being published until ebusd is restarted. The error raised is:

```
2017-05-11 13:08:27.841 [mqtt error] decode bai HwcTempDesired: ERR: invalid position
```

russss, 2017-05-11:
Actually I don't think this is the bug now - even though I have disabled polling for invalid values, the MQTT messages for that circuit still stop (but I still get `ebusd/global/uptime` messages).

It also seems like the listen command in the TCP interface stops reporting messages.

I'm using ebusd 3.0pre.340f862.

john30, 2017-05-13:
I'm afraid I can't reproduce this behaviour. The decode error is correctly logged but the next valid decodable message gets posted regularly on MQTT.

russss, 2017-05-14:
Yes, I can't reproduce this now. I think it might have been a bug in the code I was using to receive the MQTT messages, sorry!

closed 2017-05-14 by russss