john30/ebusd #87: BAI etc. not recognized with Version 3

texel007, 2017-05-21:
Hi, 
I just upgraded from an old ebusd version (2.1.0) to v3.

However, it seems as most of the devices are not recognized.

I start ebusd with:
ebusd -s

In ebusctl I receive with "i" the following:

localhost: i
version: ebusd 3.0pre.dbe70f0
update check: OK, 4 newer configuration files available
signal: acquired
symbol rate: 23
max symbol rate: 98
reconnects: 0
masters: 4
messages: 122
conditional: 4
poll: 0
update: 8
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401"
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 50: slave, scanned "MF=Vaillant;ID=V6100;SW=0418;HW=1902", loaded "vaillant/50.v61.mc.csv"
address 70: master #4
address 75: slave #4, scanned "MF=Vaillant;ID=V8100;SW=0202;HW=2502"

However with find -v
only the commands of the V61 are shown (mc...)
Entire commands of BAI, VRC470 etc. are missing.

Any idea what went wrong?

Kind regards,

Texel

find -v result:

localhost: find -v
broadcast datetime = no data stored
broadcast error = no data stored
broadcast hwcStatus = no data stored
broadcast ident = no data stored
broadcast ident = no data stored
broadcast load = no data stored
broadcast outsidetemp = temp2=17.812
broadcast signoflife = no data stored
broadcast vdatetime = time=13:48:57;date=21.05.2017
mc ActualTempDesired = no data stored
mc AutoOffMode = no data stored
mc BoilerCircuitPump = no data stored
mc BoilerMaxPreShutdown = no data stored
mc CirPump = no data stored
mc currenterror = error=-;error=-;error=-;error=-;error=-
mc Date = no data stored
mc DayNightState = no data stored
mc DisplayedRoomTempDesired = no data stored
mc errorhistory = no data stored
mc FloorPavingDryingDay = no data stored
mc FloorPavingDryingTemp = no data stored
mc Flow1Sensor = no data stored
mc FlowTempDesired = no data stored
mc FlowTempMax = no data stored
mc FlowTempMin = no data stored
mc FrostProtConfig = no data stored
mc FrostProtectDelay = no data stored
mc FrostProtState = no data stored
mc Hc1PumpKol1PP1Port = no data stored
mc Hc2aLegPumpP2Port = no data stored
mc Hc2CircuitType = no data stored
mc Hc2ContinuosHeating = no data stored
mc Hc2ManualOPRoomTempDesired = no data stored
mc Hc2pDemand = no data stored
mc Hc2PumpAlPort = no data stored
mc Hc2QuickVetoActive = no data stored
mc Hc2QuickVetoTemp = no data stored
mc Hc2RoomControlMode = no data stored
mc Hc2SummerOffset = no data stored
mc Hc2zPort = no data stored
mc HcFlowIncrease = no data stored
mc HcParallelStorageFillingEnabled = no data stored
mc HcPreHeatingActive = no data stored
mc HcPreShutdownActive = no data stored
mc HcPumpIdlePeriod = no data stored
mc HcPumpIdlePeriodActive = no data stored
mc hcTimer.Friday = no data stored
mc hcTimer.Monday = no data stored
mc hcTimer.Saturday = no data stored
mc hcTimer.Sunday = no data stored
mc hcTimer.Thursday = no data stored
mc hcTimer.Tuesday = no data stored
mc hcTimer.Wednesday = no data stored
mc HeatingCurve = no data stored
mc HolidayTempDesired = no data stored
mc InternalOperatingMode470 = no data stored
mc IsInBankHoliday = no data stored
mc IsInHoliday = no data stored
mc IsInImpactVentilation = no data stored
mc IsInOneDayBankHoliday = no data stored
mc IsInOneDayHoliday = no data stored
mc IsInParty = no data stored
mc IsInSystemOff = no data stored
mc Mixer1DutyCycle = no data stored
mc Mixer1State = no data stored
mc MonitorEEpromInkonsiNumber = no data stored
mc OffConfig = no data stored
mc PreheatingMinutesMax = no data stored
mc PsTempDesired = no data stored
mc PumpSavingState = no data stored
mc Setpoints.Friday = no data stored
mc Setpoints.Monday = temp1=0.5;temp1=20.5;temp1=20.5
mc Setpoints.Saturday = no data stored
mc Setpoints.Sunday = no data stored
mc Setpoints.Thursday = no data stored
mc Setpoints.Tuesday = no data stored
mc Setpoints.Wednesday = no data stored
mc SfState = no data stored
mc SunMoonInfo = no data stored
mc TempDesiredLow = no data stored
mc ThermostatState = no data stored
mc Time = no data stored
mc Weekday = no data stored
memory eeprom = no data stored
memory ram = no data stored
scan id = no data stored
scan.08  = MF=Vaillant;ID=BAI00;SW=0518;HW=7401
scan.15  = no data stored
scan.50  = MF=Vaillant;ID=V6100;SW=0418;HW=1902
scan.75  = MF=Vaillant;ID=V8100;SW=0202;HW=2502

texel007, 2017-05-21:
The following log is also strange - what means [main error] scan config 08: ERR: element not found ?

2017-05-21 14:26:59.838 [update notice] unknown MS cmd: 7050b505053c3e013e01 / 00
2017-05-21 14:26:59.950 [update notice] unknown MS cmd: 1050b51303041f00 / 0a1f000200000002000200
2017-05-21 14:27:00.957 [update notice] unknown MS cmd: 1008b5110101 / 09666200123624000032
2017-05-21 14:27:01.227 [update notice] unknown MS cmd: 1050b504021901 / 0401292929
2017-05-21 14:27:01.883 [main notice] starting initial broadcast scan
2017-05-21 14:27:01.909 [bus error] send to fe: ERR: read timeout, retry
2017-05-21 14:27:01.957 [bus error] send to fe: ERR: read timeout, retry
2017-05-21 14:27:02.005 [bus error] send to fe: ERR: read timeout, retry
2017-05-21 14:27:02.053 [bus error] send to fe: ERR: read timeout
2017-05-21 14:27:02.055 [main error] initial scan failed: ERR: read timeout
2017-05-21 14:27:02.717 [update notice] unknown MS cmd: 7050b5040126 / 0705000009003e01
2017-05-21 14:27:02.969 [update notice] unknown MS cmd: 1008b5040100 / 0a03042714210507170012
2017-05-21 14:27:03.240 [update notice] unknown MS cmd: 1008b5110102 / 05033c96466e
2017-05-21 14:27:05.032 [update notice] unknown MS cmd: 1050b5040137 / 0200ff
2017-05-21 14:27:05.320 [update notice] unknown MS cmd: 1008b510090000006effff010000 / 0101
2017-05-21 14:27:07.015 [update notice] unknown MS cmd: 1050b505021800 / 00
2017-05-21 14:27:07.575 [update notice] unknown MS cmd: 7008b5110101 / 09666200123624000032
2017-05-21 14:27:08.911 [main error] scan config 08: ERR: element not found
2017-05-21 14:27:09.093 [update notice] unknown MS cmd: 1050b505023001 / 0101
2017-05-21 14:27:11.021 [main error] scan config 15: ERR: element not found
2017-05-21 14:27:11.237 [update notice] unknown MS cmd: 1008b5110101 / 09666200123624000032
2017-05-21 14:27:12.419 [update notice] unknown MS cmd: 7050b5040126 / 0705000009003e01
2017-05-21 14:27:13.104 [update notice] unknown MS cmd: 1050b5040126 / 0705000009003e01
2017-05-21 14:27:13.965 [main error] scan config 50: ERR: element not found
2017-05-21 14:27:14.243 [update notice] unknown MS cmd: 1008b5110100 / 08310312001f100000
2017-05-21 14:27:14.467 [update notice] unknown BC cmd: 10feb51603010012
2017-05-21 14:27:15.075 [update notice] unknown MS cmd: 1050b5040137 / 0200ff
2017-05-21 14:27:15.376 [update notice] unknown MS cmd: 1008b510090000006effff010000 / 0101
2017-05-21 14:27:15.787 [main error] scan config 75: ERR: element not found
2017-05-21 14:27:17.198 [update notice] unknown MS cmd: 1050b505021800 / 00
2017-05-21 14:27:17.296 [update notice] unknown MS cmd: 7008b5110101 / 09666200123624000032
2017-05-21 14:27:19.182 [update notice] unknown MS cmd: 1050b505023001 / 0101
2017-05-21 14:27:21.242 [update notice] unknown MS cmd: 1008b5110101 / 09666200123624000032
2017-05-21 14:27:22.093 [update notice] unknown MS cmd: 7050b5040126 / 0705000009003e01
2017-05-21 14:27:23.353 [update notice] unknown MS cmd: 1050b503020001 / 0affffffffffffffffffff
2017-05-21 14:27:23.608 [update notice] unknown MS cmd: 1050b509040e010000 / 00
2017-05-21 14:27:23.864 [update notice] unknown MS cmd: 1050b509040e200000 / 00
2017-05-21 14:27:24.123 [update notice] unknown MS cmd: 1008b5110102 / 05033c96466e
2017-05-21 14:27:25.286 [update notice] unknown MS cmd: 1050b5040137 / 0200ff

texel007, 2017-05-21:
Sorry to post again, but maybe the start of the log also helps (it seems the initial scan fails ..)

root@ebusd:/etc/ebusd/vaillant# ebusd -s -f --configpath=/etc/ebusd
2017-05-21 14:36:21.243 [main notice] ebusd 3.0pre.dbe70f0 started
2017-05-21 14:36:21.280 [bus notice] bus started with own address 31/36
2017-05-21 14:36:21.535 [bus notice] signal acquired
2017-05-21 14:36:21.625 [bus notice] new master 10, master count 2
2017-05-21 14:36:21.691 [update notice] unknown MS cmd: 1050b503020001 / 0affffffffffffffffffff
2017-05-21 14:36:21.944 [update notice] unknown MS cmd: 1050b509040e010000 / 00
2017-05-21 14:36:22.187 [update notice] unknown MS cmd: 1050b509040e200000 / 00
2017-05-21 14:36:22.456 [bus notice] new master 03, master count 3
2017-05-21 14:36:22.457 [update notice] unknown MS cmd: 1008b5110102 / 05033c96466e
2017-05-21 14:36:23.656 [update notice] unknown MS cmd: 1050b5040137 / 0200ff
2017-05-21 14:36:24.456 [bus notice] new master 70, master count 4
2017-05-21 14:36:24.505 [update notice] unknown MS cmd: 7050b5040126 / 0705000009003e01
2017-05-21 14:36:25.798 [update notice] unknown MS cmd: 1008b510090000006effff010000 / 0101
2017-05-21 14:36:26.038 [update notice] unknown MS cmd: 1050b505021800 / 00
2017-05-21 14:36:27.733 [update notice] unknown MS cmd: 1050b505023001 / 0101
2017-05-21 14:36:29.202 [update notice] unknown MS cmd: 1050b51303041f00 / 0a1f000200000002000200
2017-05-21 14:36:29.348 [update notice] unknown MS cmd: 7008b5110101 / 09625e30123624000032
2017-05-21 14:36:29.523 [update notice] unknown MS cmd: 1050b504021901 / 0401292929
2017-05-21 14:36:29.811 [update notice] unknown MS cmd: 1008b5110101 / 09625e30123624000032
2017-05-21 14:36:31.280 [main notice] starting initial broadcast scan
2017-05-21 14:36:31.337 [bus error] send to fe: ERR: read timeout, retry
2017-05-21 14:36:31.385 [bus error] send to fe: ERR: read timeout, retry
2017-05-21 14:36:31.433 [bus error] send to fe: ERR: read timeout, retry
2017-05-21 14:36:31.481 [bus error] send to fe: ERR: read timeout
2017-05-21 14:36:31.483 [main error] initial scan failed: ERR: read timeout
2017-05-21 14:36:31.886 [update notice] unknown MS cmd: 1050b503020003 / 0affffffffffffffffffff
2017-05-21 14:36:32.145 [update notice] unknown MS cmd: 1050b5040126 / 0705000009003e01
2017-05-21 14:36:32.445 [update notice] unknown MS cmd: 1008b5040100 / 0a03343614210507173012
2017-05-21 14:36:33.854 [update notice] unknown MS cmd: 1050b5040137 / 0200ff
2017-05-21 14:36:34.220 [update notice] unknown MS cmd: 7050b5040126 / 0705000009003e01
2017-05-21 14:36:35.866 [update notice] unknown MS cmd: 1008b510090000006effff010000 / 0101
2017-05-21 14:36:36.107 [update notice] unknown MS cmd: 1050b505021800 / 00
2017-05-21 14:36:37.945 [update notice] unknown MS cmd: 1050b505023001 / 0101
2017-05-21 14:36:39.048 [update notice] unknown MS cmd: 7008b5110101 / 09625e30123624000032
2017-05-21 14:36:39.976 [update notice] unknown MS cmd: 1008b5110101 / 09625e30123624000032
2017-05-21 14:36:40.999 [update notice] unknown MS cmd: 7050b505053c3e013e01 / 00
2017-05-21 14:36:41.519 [main error] scan config 08: ERR: element not found
2017-05-21 14:36:41.942 [update notice] unknown MS cmd: 1008b5100305ff01 / 00
2017-05-21 14:36:43.565 [main error] scan config 15: ERR: element not found
2017-05-21 14:36:43.894 [update notice] unknown MS cmd: 7050b5040126 / 0705000009003e01

texel007, 2017-05-21:
seems I can't write anymore on the ebus. There is always a read timeout.
When I downgraded to my old ebusd2.1 version it is a similar behavior. Maybe something with the Raspi1 went wrong (I updated Linux before changing to Ebusd v3). Maybe with the update the USB timing has changed? I saw in the FHEM forum a user reporting similar issues 

... any advice?

john30, 2017-05-25:
yes I guess the root cause is the linux update. can you go back to the previous version?

texel007, 2017-05-25:
unfortunately not - I don't have a backup of the old version and even don't know exactly the old version. 

New version is: 
Linux ebusd 4.9.24+ #993 Wed Apr 26 17:56:54 BST 2017 armv6l GNU/Linux

Do you have any hint where I search/debug? Unfortunately I don't have access to the hardware at the moment (it´s at a different location) so I can't plug it to another Raspi.

Do you have any experience with HW changes? Raspi1,2,3 , Odroid .. I thought it should not really matter which HW or Linux Version I use?

john30, 2017-05-28:
this is a duplicate of #79, please continue discussion there

closed 2017-05-28 by john30