john30/ebusd #92: Missing "update" entry in logfile when reading with ebusctl

J0EK3R, 2017-06-11:
I'm using two Raspies with current ebusd (Version 7a569e2) running on each, using the same physical ebus.

Two console-windows on my PC are showing the latest entries from both log-files:
`tail -f /var/log/ebusd.log
`
The ebusd-config files on both systems are equal, the ebusd-instances habe different ebus-adresses.

If I'm reading parameters like 
`ebusctl r -f NoFlameFormation
`
on one of the Raspies I'm getting the expected log-entry only in the logfile of the other Raspi!
`2017-06-11 10:43:46.617 [update notice] update sc NoFlameFormation QQ=ff: 9590
`
If I'm reading the same Parameter on the other Raspi, I'm getting the expected log-entry on the first one.

I've expected to see the same log-entries on both Raspies?!

Here is the log-setting:
`pi@fhem:~ $ ebusctl log
main: notice
network: notice
bus: notice
update: notice
other: notice
`

And this is the definition of the used telegram:
`r;b,,NoFlameFormation,Keine Flammbildung ,,,,"DD12D5",,s,_8_Skip,,, ,,s,_16_Counter,,,
`

 Any ideas?

john30, 2017-06-12:
why would you want an actively initiated bus request to appear just like one passively listened to appear in the log file?

closed 2017-06-12 by john30

J0EK3R, 2017-06-12:
Just for debugging purposes... :-)