john30/ebusd #93: Please disable buffering of raw-logging

J0EK3R, 2017-06-12:
It would be very helpful for debugging purposes to be able to watch the raw-logging file just-in-time.

I've patched rotatefile.cpp to flush each written line:
```
57: void RotateFile::write(const unsigned char* value, const size_t size, const bool received, const bool bytes) {
...
83: //  if ((m_fileSize%1024) == 0) {
84:     fflush(m_stream);
85: //  }

```

Now I'm able to watch each detected ebus-message just-in-time just by using the tail command:

`pi@rasp2:~/src/ebusd $ sudo tail -f /var/log/ebusrawdata.log
`

```
2017-06-12 17:13:47.864 <ff350902031201025700020c00f400
2017-06-12 17:13:48.222 <ff08500003db12d3bd000300a30bb000
2017-06-12 17:13:48.357 <ff08500003dd12d5cf00030076253500
2017-06-12 17:13:53.350 <30f1050709550490010000ff65ffdf00
2017-06-12 17:13:54.645 <30fe070009001b00171712060717d3
2017-06-12 17:13:55.809 <70f105070955ff90010000ffffffb300
2017-06-12 17:13:55.946 <70515010041a19028005
2017-06-12 17:13:56.142 <70515010041a1902800500093228311b1920200000fd00
2017-06-12 17:13:58.307 <f1fe500a0d010007000056ff60001b0019083e
2017-06-12 17:14:03.351 <30f1050709550490010080ff65ffdf00
2017-06-12 17:14:04.649 <70f105070955ff90010000ffffffb300
2017-06-12 17:14:04.929 <70515010041a1902800500093228311b1920200000fd00
2017-06-12 17:14:07.909 <ff350902031601020f000296007900
2017-06-12 17:14:13.359 <30f1050709550490010000ff65ffdf00
2017-06-12 17:14:14.620 <70f105070955ff90010000ffffffb300
2017-06-12 17:14:14.871 <70515010041a1902800500093228311b1920200000fd00
```

If you switch raw-logging to bytes you are able to adjust the potentiometer, too
`pi@rasp2:~/src/ebusd $ ebusctl raw bytes
`

```
2017-06-12 17:22:23.053 <aa
2017-06-12 17:22:23.098 <aa
2017-06-12 17:22:23.143 <aa
2017-06-12 17:22:23.188 <aa
2017-06-12 17:22:23.234 <aa
2017-06-12 17:22:23.237 <30
2017-06-12 17:22:23.246 <f1
2017-06-12 17:22:23.259 <05
2017-06-12 17:22:23.263 <07
2017-06-12 17:22:23.268 <09
2017-06-12 17:22:23.272 <55
2017-06-12 17:22:23.276 <04
2017-06-12 17:22:23.281 <90
2017-06-12 17:22:23.285 <01
2017-06-12 17:22:23.289 <00
2017-06-12 17:22:23.294 <00
2017-06-12 17:22:23.298 <ff
2017-06-12 17:22:23.301 <65
2017-06-12 17:22:23.307 <ff
2017-06-12 17:22:23.310 <df
2017-06-12 17:22:23.321 <00
2017-06-12 17:22:23.324 <aa
2017-06-12 17:22:23.370 <aa
2017-06-12 17:22:23.413 <aa
```

john30, 2017-06-13:
that's easily achievable by specifying no dedicated raw log file so that the data runs into the regular ebusd log file

closed 2017-06-13 by john30

J0EK3R, 2017-06-15:
OK, you're right :-)

I'm using the tail-command and pipe the output to grep to filter raw-messages:

```
tail -f /var/log/ebusd.log | grep "\[bus notice\]"
2017-06-15 12:15:09.761 [bus notice] <7051501004e619028021
2017-06-15 12:15:09.968 [bus notice] <7051501004e61902802100093228321c1720200000da00
2017-06-15 12:15:16.776 [bus notice] <f1fe500a0d01000700004cff58001cb0160810
2017-06-15 12:15:18.348 [bus notice] <30f1050709550390010080ff50ff5c00
2017-06-15 12:15:19.724 [bus notice] <70f105070955ff90010080ffffffb3
2017-06-15 12:15:19.850 [bus notice] <70f105070955ff90010080ffffffb300
2017-06-15 12:15:20.138 [bus notice] <7051501004e61902802100093228321c1720200000da00
2017-06-15 12:15:27.740 [bus notice] >ff08500003dd12d5cf<000300762535>00
2017-06-15 12:15:28.372 [bus notice] <30f1050709550390010080ff50ff5c00
```

But I was thinking about the if-condition for flushing to the log-file:

https://github.com/john30/ebusd/blob/master/src/lib/utils/rotatefile.cpp#L83

```
83 if ((m_fileSize%1024) == 0) {
84    fflush(m_stream);
85 }
```

Am I wrong or must _m_fileSize_ exactly be a **multiple of 1024** (modulo-result == 0) to match the condition?!
Is this correct or a bug?

john30, 2017-06-17:
that's right, should be done in a better way :-), see #95