john30/ebusd #96: ebusd3 build from source : ERR: element not found

rvdm82, 2017-06-22:
After gettiing 2.1.4 to work without issue, I built v3 from master in order to get MQTT support. The version starts but doesnt seem to load the config files for the relevant devices upon scanning. ( log below ).
I suspect my 2.1.4 ebusd-configuration is somehow incompatible, but have no clue as how to solve this.
Running with --checkconfig also produces errors. Any ideas on how to solve this ?

Log output
```2017-06-22 23:33:35.097 [main error] scan config 08: ERR: element not found
2017-06-22 23:33:35.772 [update notice] unknown MS cmd: 1050b505023001 / 0101
2017-06-22 23:33:37.101 [main error] scan config 15: ERR: element not found
2017-06-22 23:33:37.859 [update notice] unknown MS cmd: 1050b503020001 / 0affffffffffffffffffff
2017-06-22 23:33:38.130 [update notice] unknown MS cmd: 1008b5110101 / 094041d012ff650000ff
2017-06-22 23:33:38.489 [update notice] unknown MS cmd: 7008b5110101 / 094041d012ff650000ff
2017-06-22 23:33:39.105 [main error] scan config 50: ERR: element not found
2017-06-22 23:33:39.859 [update notice] unknown MS cmd: 1008b5100305ff01 / 0101
2017-06-22 23:33:41.109 [main error] scan config 75: ERR: element not found```

CHeckconfig output
```ebusd --loglevel debug --checkconfig
2017-06-22 23:42:58.892 [main notice] ebusd 3.0pre.e451e96 performing configuration check...
2017-06-22 23:42:58.895 [main info] loading configuration files from /etc/ebusd
2017-06-22 23:42:58.910 [main info] read templates in /etc/ebusd
2017-06-22 23:42:58.913 [main info] reading file /etc/ebusd/memory.csv
2017-06-22 23:42:58.922 [main info] reading file /etc/ebusd/broadcast.csv
2017-06-22 23:42:58.936 [main info] reading dir  /etc/ebusd/vaillant
2017-06-22 23:42:58.969 [main info] read templates in /etc/ebusd/vaillant
2017-06-22 23:42:58.973 [main info] reading file /etc/ebusd/vaillant/50.ehp.mc.csv
2017-06-22 23:42:59.030 [main info] reading file /etc/ebusd/vaillant/52.mc2.mc.4.csv
2017-06-22 23:42:59.092 [main info] reading file /etc/ebusd/vaillant/15.36p.csv
2017-06-22 23:42:59.244 [main info] reading file /etc/ebusd/vaillant/0a.pmw.hwc.csv
2017-06-22 23:42:59.307 [main info] reading file /etc/ebusd/vaillant/25.vr630.hwc.csv
/etc/ebusd/vaillant/25.vr630.hwc.csv:19: ERR: duplicate name, invalid name
2017-06-22 23:42:59.326 [main error] error reading config files: ERR: duplicate name, last error: /etc/ebusd/vaillant/25.vr630.hwc.csv:19: ERR: duplicate name, invalid name
2017-06-22 23:42:59.438 [main info] included "/etc/ebusd/vaillant/hwcmode.inc" for "0a", included "/etc/ebusd/vaillant/timer.inc" for "0a", included "/etc/ebusd/vaillant/errors.inc" for "0a", included "/etc/ebusd/vaillant/mcmode.inc" for "50", included "/etc/ebusd/vaillant/timer.inc" for "50", included "/etc/ebusd/vaillant/errors.inc" for "50", included "/etc/ebusd/vaillant/roomtempoffset.inc" for "50", included "/etc/ebusd/vaillant/quick.inc" for "50", included "/etc/ebusd/vaillant/timercool.inc" for "52.*.4", included "/etc/ebusd/vaillant/mcmode.inc" for "52.*.4", included "/etc/ebusd/vaillant/timer.inc" for "52.*.4", included "/etc/ebusd/vaillant/errors.inc" for "52.*.4", included "/etc/ebusd/vaillant/roomtempoffset.inc" for "52.*.4", included "/etc/ebusd/vaillant/quick.inc" for "52.*.4"
2017-06-22 23:42:59.441 [main notice] found messages: 501 (64 conditional on 19 conditions, 1 poll, 5 update)
2017-06-22 23:42:59.461 [main notice] ebusd stopped```

rvdm82, 2017-06-23:
Fixed - element not found errors where apparantly caused by starting ebusd with -r ( older versions failed on -r togheter with --scanconfig with an error message explaining parameters are incompatible , latest master doesn't )

john30, 2017-06-24:
why do you use 2.1.4 anyway? at least 2.4 would be the latest recommended & released version

closed 2017-06-24 by john30