john30/ebusd #99: ebusd 3 error with last ebusd-configuration

modenet, 2017-07-12:


modenet, 2017-07-12:
2017-07-12 18:16:00.981 [main error] unable to load scan config 26: no file from /etc/ebusd/vaillant with prefix 26. matches ID "vr_71", SW0104, HW0503
2017-07-12 18:16:00.982 [main error] scan config 26: ERR: element not found
2017-07-12 18:16:03.754 [main error] unable to load scan config 76: no file from /etc/ebusd/vaillant with prefix 76. found
2017-07-12 18:16:03.754 [main error] scan config 76: ERR: element not found
2017-07-12 18:16:07.901 [main error] unable to load scan config e8: no file from /etc/ebusd/vaillant with prefix e8. found
2017-07-12 18:16:07.901 [main error] scan config e8: ERR: element not found
2017-07-12 18:16:20.557 [main error] unable to load scan config 26: no file from /etc/ebusd/vaillant with prefix 26. matches ID "vr_71", SW0104, HW0503
2017-07-12 18:16:20.557 [main error] scan config 26: ERR: element not found
2017-07-12 18:16:23.189 [main error] unable to load scan config 76: no file from /etc/ebusd/vaillant with prefix 76. found
2017-07-12 18:16:23.190 [main error] scan config 76: ERR: element not found
2017-07-12 18:16:25.780 [main error] unable to load scan config e8: no file from /etc/ebusd/vaillant with prefix e8. found
2017-07-12 18:16:25.780 [main error] scan config e8: ERR: element not found
2017-07-12 18:16:38.465 [main error] unable to load scan config 26: no file from /etc/ebusd/vaillant with prefix 26. matches ID "vr_71", SW0104, HW0503
2017-07-12 18:16:38.465 [main error] scan config 26: ERR: element not found
2017-07-12 18:16:41.389 [main error] unable to load scan config 35: no file from /etc/ebusd/vaillant with prefix 35. matches ID "vr_91", SW0200, HW1903

modenet, 2017-07-12:
scan id = ERR: invalid address
scan.08 ident = Vaillant;HMU01;0300;8802
scan.15 ident = Vaillant;70000;0419;4603
scan.26 ident = Vaillant;VR_71;0104;0503
scan.76 ident = Vaillant;VWZIO;0106;0103
scan.e8 ident = Vaillant;FMU00;0202;6502

john30, 2017-07-13:
this is a configuration issue, please post it in [ebusd-configuration](https://github.com/john30/ebusd-configuration/issues) to keep track of it and please post the result of `ebusctl scan` together with information of which configuration file set you are using

closed 2017-07-13 by john30