john30/ebusd #100: MQTT in pre build release

biohazardxxx, 2017-08-01:
HI,

using the prebuild for rpi I get this error when adding mqtt params:

> Starting ebusd: ebusd/usr/bin/ebusd: unrecognized option '--mqtthost=192.168.0.1'

I've added the params in  /etc/default/ebusd:
at 
`EBUSD_OPTS="--scanconfig --httpport=8889 --mqtthost=192.168.0.1 --mqtttopic=ebusd"`

Is MQTT support missing in the prebuild package or am I doing something wrong?
If it is missing any chance to get it included? (did not manage to build it on my own)

BTW: Thank you for the great work!

john30, 2017-08-03:
you need to install a release with "mqtt" in the file name

closed 2017-08-03 by john30

biohazardxxx, 2017-08-03:
Where can I find that?
Not listed here: https://github.com/john30/ebusd/releases/tag/v2.4

john30, 2017-08-03:
this is a 3.0 feature (not available in 2.*)