john30/ebusd #101: Build warnings

andig, 2017-08-20:
Current git () shows the following build warnings on raspbian stretch:

	/home/pi/ebusd/src/lib/ebus/data.cpp: In member function ‘virtual ebusd::result_t ebusd::DataFieldTemplates::getFieldMap(const string&, std::vector<std::__cxx11::basic_string<char> >*, std::__cxx11::string*) const’:
	/home/pi/ebusd/src/lib/ebus/data.cpp:1213:37: warning: variable ‘previous’ set but not used [-Wunused-but-set-variable]
	       map<string, size_t>::iterator previous = seen.find(lowerName);
	                                     ^~~~~~~~

	/home/pi/ebusd/src/lib/ebus/message.cpp: In member function ‘virtual ebusd::result_t ebusd::MessageMap::getFieldMap(const string&, std::vector<std::__cxx11::basic_string<char> >*, std::__cxx11::string*) const’:
	/home/pi/ebusd/src/lib/ebus/message.cpp:1863:37: warning: variable ‘previous’ set but not used [-Wunused-but-set-variable]
	       map<string, size_t>::iterator previous = seen.find(lowerName);
	                                     ^~~~~~~~

	In file included from /usr/include/c++/6/map:61:0,
	                 from /home/pi/ebusd/src/lib/ebus/message.h:26,
	                 from /home/pi/ebusd/src/lib/ebus/message.cpp:19:
	/usr/include/c++/6/bits/stl_map.h: In member function ‘std::map<_Key, _Tp, _Compare, _Alloc>::mapped_type& std::map<_Key, _Tp, _Compare, _Alloc>::operator[](const key_type&) [with _Key = long long unsigned int; _Tp = std::vector<ebusd::Message*>; _Compare = std::less<long long unsigned int>; _Alloc = std::allocator<std::pair<const long long unsigned int, std::vector<ebusd::Message*> > >]’:
	/usr/include/c++/6/bits/stl_map.h:483:4: note: parameter passing for argument of type ‘std::_Rb_tree<long long unsigned int, std::pair<const long long unsigned int, std::vector<ebusd::Message*> >, std::_Select1st<std::pair<const long long unsigned int, std::vector<ebusd::Message*> > >, std::less<long long unsigned int>, std::allocator<std::pair<const long long unsigned int, std::vector<ebusd::Message*> > > >::const_iterator {aka std::_Rb_tree_const_iterator<std::pair<const long long unsigned int, std::vector<ebusd::Message*> > >}’ will change in GCC 7.1
	    __i = _M_t._M_emplace_hint_unique(__i, std::piecewise_construct,
	    ^~~
	In file included from /usr/include/c++/6/map:60:0,
	                 from /home/pi/ebusd/src/lib/ebus/message.h:26,
	                 from /home/pi/ebusd/src/lib/ebus/message.cpp:19:
	/usr/include/c++/6/bits/stl_tree.h: In member function ‘std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::iterator std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_M_emplace_hint_unique(std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::const_iterator, _Args&& ...) [with _Args = {const std::piecewise_construct_t&, std::tuple<const long long unsigned int&>, std::tuple<>}; _Key = long long unsigned int; _Val = std::pair<const long long unsigned int, std::vector<ebusd::Message*> >; _KeyOfValue = std::_Select1st<std::pair<const long long unsigned int, std::vector<ebusd::Message*> > >; _Compare = std::less<long long unsigned int>; _Alloc = std::allocator<std::pair<const long long unsigned int, std::vector<ebusd::Message*> > >]’:
	/usr/include/c++/6/bits/stl_tree.h:2193:7: note: parameter passing for argument of type ‘std::_Rb_tree<long long unsigned int, std::pair<const long long unsigned int, std::vector<ebusd::Message*> >, std::_Select1st<std::pair<const long long unsigned int, std::vector<ebusd::Message*> > >, std::less<long long unsigned int>, std::allocator<std::pair<const long long unsigned int, std::vector<ebusd::Message
	*> > > >::const_iterator {aka std::_Rb_tree_const_iterator<std::pair<const long long unsigned int, std::vector<ebusd::Message*> > >}’ will change in GCC 7.1
	       _Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::
	       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
	/usr/include/c++/6/bits/stl_tree.h: In member function ‘std::pair<std::_Rb_tree_node_base*, std::_Rb_tree_node_base*> std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_M_get_insert_hint_unique_pos(std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::const_iterator, const key_type&) [with _Key = long long unsigned int; _Val = std::pair<const long long unsigned int, std::vector<ebusd::Message*> >; _KeyOfValue = std::_Select1st<std::pair<const long long unsigned int, std::vector<ebusd::Message*> > >; _Compare = std::less<long long unsigned int>; _Alloc = std::allocator<std::pair<const long long unsigned int, std::vector<ebusd::Message*> > >]’:
	/usr/include/c++/6/bits/stl_tree.h:1928:5: note: parameter passing for argument of type ‘std::_Rb_tree<long long unsigned int, std::pair<const long long unsigned int, std::vector<ebusd::Message*> >, std::_Select1st<std::pair<const long long unsigned int, std::vector<ebusd::Message*> > >, std::less<long long unsigned int>, std::allocator<std::pair<const long long unsigned int, std::vector<ebusd::Message*> > > >::const_iterator {aka std::_Rb_tree_const_iterator<std::pair<const long long unsigned int, std::vector<ebusd::Message*> > >}’ will change in GCC 7.1
	     _Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::
	     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

john30, 2017-08-22:
should work better with 6be124f, can you verify please?

andig, 2017-08-22:
The first two are gone, `message.cpp` and missing install make target (mentioned in wiki) are still there.

john30, 2017-08-27:
the warnings are for ABI changes as stated in https://gcc.gnu.org/gcc-7/changes.html, so it seems nothing to worry about right now.
for the missing install target: this is due to compile support only for CMake right now. Subject to be enhanced some time in future.

john30, 2018-05-10:
mini install added with 3af05a2

john30, 2022-09-17:
install target should be fine by now, e.g. with commit 13c121704b539873ca78c7fd9a0dcb75191d4144

closed 2022-09-17 by john30