john30/ebusd #102: EnvironmentFile not set correctly in ebusd.service for Raspbian Jessie

guillaumezin, 2017-09-02:
Hello,

In Jackfruit release, in ebusd-3.0_armhf_mqtt1.deb (I didn't check in other deb files), EnvironmentFile should be set to "-/etc/default/ebusd" instead of "-/etc/conf-d/ebusd", this way the daemon is started with the "--scanconfig" parameter.

john30, 2017-09-09:
sorry for the inconvenience.
Jessie packages with correct EnvironmentFile are now available here:
https://github.com/john30/ebusd/releases/download/v3.0/ebusd-3.0_armhf.deb
https://github.com/john30/ebusd/releases/download/v3.0/ebusd-3.0_armhf_mqtt1.deb
and the problematic ones have been renamed in the release

mike-bike, 2017-09-20:
Hi John, firstly many thanks for your effort and the great program you offer. I was trying to upgrade to v3.0 but found that https://github.com/john30/ebusd/releases/download/v3.0/ebusd-3.0_armhf_mqtt1.deb still seems to contain the pre release 
pi@pi1:~/tmp $ ebusd --version
ebusd 3.0.595c7c0

Same apply to the other armhf.deb. The wheezy package seems to have v3.0 packaged.

Also where could I find most recent config files. The config package 2.1.1 seems to be unchanged.

version: ebusd 3.0.595c7c0
update check: revision v3.0 available, broadcast.csv: different version available, vaillant/15.ui.csv: different version available, vaillant/25.solsy.hwc.csv: different version available, vaillant/26.solsy.hc.csv: different version available, vaillant/bai.0010004121.inc: different version available, vaillant/broadcast.csv: different version available, vailla 

Many thanks & Kind regards
Michael

brucemiranda, 2017-10-17:
The same bug in the amd64.deb file too.

john30, 2017-10-18:
reopened due to last coment

john30, 2017-12-17:
fixed in next release automatically due to commit d57c625b5c0645c7b5a79600fe77e81a6bffa43a

closed 2017-12-17 by john30