john30/ebusd #105: cmake error on raspberry

modenet, 2017-09-13:
when I launch cmake . I obtained:

root@raspdenis:~/ebusd# cmake .
-- The C compiler identification is GNU 4.6.3
-- The CXX compiler identification is GNU 4.6.3
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
CMake Error at CMakeLists.txt:38 (add_compile_options):
  Unknown CMake command "add_compile_options".

modenet, 2017-09-13:
cmake version 2.8.9

john30, 2017-09-14:
could you give it a try after replacing add_compile_options with add_definitions in all CMakeLists.txt files please?
Or alternatively upgrade to cmake >= 3.0

modenet, 2017-09-14:
hi, I can't use cmake 3.0 on my  my raspbian wheezy.
after replacing  add_compile_options with add_definitions in all CMakeLists.txt files cmake returns no errors.
thanks

closed 2017-09-14 by modenet

john30, 2017-09-16:
thanks for reporting this, fixed with 52c7372

commit 52c73723 (2017-09-16): compatibility to cmake < 3.0 (fixes #105)