john30/ebusd #106: repeating error in log:

modenet, 2017-09-14:
after upgrade from 2.0 to 3.0 (ebusd and configuration) I notice a repeating error:
...
2017-09-15 00:22:59.745 [bus error] poll ui YieldThisYear failed: ERR: read timeout
2017-09-15 00:23:05.042 [bus error] poll ui YieldThisYear failed: ERR: read timeout
2017-09-15 00:23:14.594 [bus error] poll ui YieldThisYear failed: ERR: arbitration lost
2017-09-15 00:23:26.824 [bus error] poll ui YieldThisYear failed: ERR: read timeout
2017-09-15 00:23:41.056 [bus error] poll ui YieldThisYear failed: ERR: wrong symbol received
2017-09-15 00:23:54.334 [bus error] poll ui YieldThisYear failed: ERR: read timeout
2017-09-15 00:24:01.672 [bus error] poll ui YieldThisYear failed: ERR: read timeout
...

my info:
version: ebusd 3.0.v3.0-6-gb174352
update check: revision v3.0-2-g29cee10 available, different configuration files available
signal: acquired
symbol rate: 63
max symbol rate: 168
reconnects: 0
masters: 8
messages: 1002
conditional: 12
poll: 1
update: 8
address 00: master #1
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0703;HW=7401", loaded "vaillant/bai.308523.inc" ([HW=7401]), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=UI   ;SW=0501;HW=6201", loaded "vaillant/15.ui.csv"
address 17: master #17
address 1c: slave #17, scanned "MF=Vaillant;ID=RC C ;SW=0501;HW=6201", loaded "vaillant/1c.rcc.4.csv"
address 23: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/23.solsy.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/25.solsy.hwc.csv"
address 26: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/26.solsy.hc.csv"
address 30: master #3
address 31: master #8, ebusd
address 35: slave #3, scanned "MF=Vaillant;ID=RC C ;SW=0501;HW=6201", loaded "vaillant/35.rcc.1.csv"
address 36: slave #8, ebusd
address 37: master #18
address 3c: slave #18, scanned "MF=Vaillant;ID=RC C ;SW=0507;HW=6201", loaded "vaillant/3c.rcc.5.csv"
address 50: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/50.solsy.mc.csv"
address 52: slave, scanned "MF=Vaillant;ID=MC2  ;SW=0500;HW=6301", loaded "vaillant/52.mc2.mc.4.csv"
address 53: slave, scanned "MF=Vaillant;ID=MC2  ;SW=0500;HW=6301", loaded "vaillant/53.mc2.mc.5.csv"
address 70: master #4
address 75: slave #4, scanned "MF=Vaillant;ID=RC C ;SW=0501;HW=6201", loaded "vaillant/75.rcc.csv"
address ec: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/ec.solsy.sc.csv"

scan result:
1 scan(s) still running
08;Vaillant;BAI00;0703;7401;21;11;14;0010004134;0006;005031;N8
15;Vaillant;UI   ;0501;6201;21;11;16;0020080465;0907;006841;N8
1c;Vaillant;RC C ;0501;6201;21;11;12;0020040079;0907;005381;N6
23;Vaillant;SOLSY;0500;6301
25;Vaillant;SOLSY;0500;6301;21;11;16;0020080463;0907;006536;N9
26;Vaillant;SOLSY;0500;6301;21;11;16;0020080463;0907;006536;N9
35;Vaillant;RC C ;0501;6201;21;11;29;0020040079;0907;005645;N9
3c;Vaillant;RC C ;0507;6201
50;Vaillant;SOLSY;0500;6301;21;11;16;0020080463;0907;006536;N9

john30, 2017-09-16:
did you also change the CSVs?
anyway, the YieldThisYear message is a complex one, as it consists of several sub-messages. could you increase the logging to check which of the 12 messages is the problematic one?

modenet, 2017-09-18:
yes, I updated also CSV (english version).

have I to enable debug log mode? to which area?

john30, 2017-09-21:
for a start, add "--lograwdata" to the start parameters of ebusd, restart it, initiate the read of the yield message again, then stop ebusd again and post the log file.

modenet, 2017-09-23:
[ebusd.log](https://github.com/john30/ebusd/files/1326936/ebusd.log)

excuse me for delay

john30, 2017-09-26:
well this seems to happen from time to time, especially during the scan process. but that's normal IMO, as other devices will always want to send something in between. So nothing to worry about from my point of view

modenet, 2017-09-26:
I see this repeating error once a second, and this happens only to my ebusd installation; in my father's house (same Vaillant equipment) this does not appear.

john30, 2017-09-28:
do you have a different USB to serial adapter in this installation? or a different OS version/ebusd version?

modenet, 2017-09-28:
the problem appeared after upgrade to ebusd 3.0 and to lastest ebus-configuration files.

john30, 2017-10-04:
you could try using the old version of ebusd again and see if you can reproduce that the problem does not exist with that version and that no OS upgrade or any other modification is the reason for this problem.

modenet, 2017-10-04:
yes I can, but the raspberry is only for ebusd and I have only upgraded
ebusd and ebusd-configuration from github

john30, 2017-10-05:
and did you?

john30, 2018-02-27:
closed due to inactivity

closed 2018-02-27 by john30